espruino/Espruino #1201: Adding SPI.Tranfer(Buffer, size) equivalent

Sparow199, 2017-07-21:
Hello,
I would send commands over SPI and receive the response.Actually, i can just send integer and get integer etc...
example: i wanna send commande : 1 (integer) and get a string as response.

gfwilliams, 2017-07-21:
Why not:

`var response = SPI1.send(my_command);`

http://www.espruino.com/Reference#l_SPI_send

If you want to convert the response to a string there are a few methods, but `E.toString` is nice and easy.

Sparow199, 2017-07-21:
this is an example of command:
{"command":"GAS","sensor":"MQ135","gas":"CO2"}  with 46 caracters but the response will be bigger than 46 and the lenght is dynamique in my case.

With actual implementation of SP1.send(data), i can send and receive the same number of caracters.
If the response size is less than send size it will work, but if the size of response is bigger than sended, data will be lost.

gfwilliams, 2017-07-21:
You can just do:

```
SPI1.send([my_command, {data: 255, count:#}])
```

Which has the benefit that you can specify the value of the extra SPI characters that are transmitted (because characters do need transmitting, even if they are ignored).

If you want something that does exactly what you're suggesting, do:

```
SPI.prototype.transfer = function(data, size) {
  var d = E.toString(data);
  return this.send([d, {data: 255, count:size-d.length}]);
}
```

closed 2017-07-21 by gfwilliams