espruino/Espruino #1203: Serial1.on('data', function(data) { ... }) never called

Sparow199, 2017-07-24:
Hello,
I send command over SPI and get the response via Serial.

I work fine with console, but i can't get response data to be processed.

I tried to setup ( with disconnected usb, i tried to get the value of data with http server ) Serial.on ... but it's never called, i used examples and i respected instructions  in official documentation.

gfwilliams, 2017-07-24:
I guess this is on ESP8266? Did you do what I'd suggested in the last bug that you filed about this?

Use `LoopbackA.setConsole()` - you can't use `Serial1.on('data',` while the console is also on Serial1.

Sparow199, 2017-07-24:
When i use LoopbackA.setConsole()

var LED = whatever_a_led_is_connected_to;
Serial1.setup(115200);
Serial1.on('data', function (data) { LED.toggle(); });
Serial1.print("Hello World");
LoopbackA.setConsole();
 i got this error:
Flash map 4MB:512/512, manuf 0xe0 chip 0x4016
>Hello WorldERROR: Prompt not detected - upload failed. Trying to recover...

gfwilliams, 2017-07-24:
Yes. That's because - as I explained on the previous bug you filed, the console is now moved away, so the IDE cannot send or receive data to it: https://github.com/espruino/Espruino/issues/1199

However you can ignore that message, and everything should work.

Sparow199, 2017-07-24:
but i tried to get data from the handler and it's not work:
var dataFromHandler=""; //global var

dataFromHandler = data;//inside the handler

and i tried to display the content of the variable over http server, it always show "", never change.

If i change the variable in my code , it's displayed correctly, so the problem is not my http server but there is no data got from the handler.

gfwilliams, 2017-07-24:
How are you wiring everything up?

Because remember the RX pin is *already* connected to the USB adaptor - so if you stick something else on that pin then the signals will be 'fighting'.

Sparow199, 2017-07-24:
My wiring is little bit complicated:
i have ESP8266 and Arduino Uno.

SPI connection: to send command from ESP8266 to Arduino Uno
4 pins connected (SPI bus) GND;

UART connection: to receive response from Arduino Uno
1Pin connected Rx(ESP8266) to Tx(Arduino Uno)

both devices are powered by USB cables connected to my PC.

Thanks

gfwilliams, 2017-07-24:
But your ESP8266 has a USB connection as well? Which is also (perhaps internally) connected to the RX pin?

gfwilliams, 2017-07-24:
What if you disconnect the Arduino uno from TX, and then upload the same code you had before, randomly type in the IDE, and then check from the HTTP server? You'll probably see data

Sparow199, 2017-07-24:
if i add LoopbackA.setConsole(); my device is down

gfwilliams, 2017-07-24:
It should still be working - it's just that you can't communicate with it!

Just upload this code:

```
Serial1.on('data', function (data) { Serial1.print("Char\r\n"); });
LoopbackA.setConsole();
```

The Web IDE will complain, but try pressing some keys. 'Char' should be displayed every time you press a key.

Sparow199, 2017-07-24:
This code works

gfwilliams, 2017-07-24:
Maybe someone else can help with this? @MaBecker ?

I have no idea why wifi would stop working if you do `setConsole`.

Just one point though: If you're using SPI correctly you should be able to have two way communications and should have no need to use the UART as well. It might solve a lot of problems for you...

Sparow199, 2017-07-24:
I tried to use just SPI, but i have a big deal because Espruino don't supporte Slave mode, so i can't send data from Arduino.

This is the result what i expected ( work fine without setConsole but no way to get data  only display it)

`Flash map 4MB:512/512, manuf 0xe0 chip 0x4016
>
=undefined
Visit http://192.168.43.95 in your web browser.
>{"MQ131":["O3"],"MQ135":["NH3","CO2"],"MULTI":["NH3","CO","NO2","C3H8","C4H10","CH4","H2","C2H5OH"]}

MaBecker, 2017-07-24:
Hi @Sparow199,

not sure what your trying to build.

    Setup Wifi and connect to a AP, use WEBIDE form you PC to connect over TCP/IP
    Settings->Communication-> Connect over TCPIP

Now your Serials are free :)

- Serial1 can send and receive data,

- Serial2 can only send data after debug is disabled (check #665).

I guess you would reach much more people  if you post such things to the Espruino Forum.

Sparow199, 2017-07-24:
@MaBecker , what is the default IP address ? , should i upload the code [example](https://www.espruino.com/WiFi+Remote+Console) of it's built in ?

MaBecker, 2017-07-24:
For AccesPoint it is  192.168.4.1 and Station IP is set via dhcp.

Sparow199, 2017-07-24:
@MaBecker @gfwilliams thank you very much, i got what i need, using WiFi instead USB resolve Serial1.on handler probleme, i'm able to get data from it now.

Adding LoopbackA.setConsole(); crash the device, i don't know why ?

MaBecker, 2017-07-24:
> Adding LoopbackA.setConsole(); crash the device, i don't know why ?

not on my side - feel free to close this issue :)

Sparow199, 2017-07-24:
@MaBecker , now UART works fine but no SPI available , i hadn't change anything in my code.

closed 2017-07-24 by Sparow199

manavakela1996, 2018-09-11:
[# This](https://github.com/espruino/Espruino/issues/1203#issuecomment-317411798) code is not working for me. I tried below code...and tried many ways but not receiving on RX but TX works fine.
**on Espruino v1.99** @gfwilliams 

`var status=0;`
`function swap() {`
`  status = !status;`
`  digitalWrite(NodeMCU.D0, status);`
`}`
`function onInit()`
`{`
`  Serial1.setup(9600);`
`  Serial1.on('data', function (data) { swap(); });`
`  LoopbackA.setConsole();`
`}`
`//onInit();`

wilberforce, 2018-09-11:
@manavakela1996 

Sorry - github is not the place for these kind of questions. issues here's are for the C code that runs. Espruino.

Post your question here in the esp8266 section of the forum.

http://forum.espruino.com/microcosms/925/

Thanks

manavakela1996, 2018-09-11:
ok but if you or any other can solve, It should great