espruino/Espruino #1205: ESP8266: wifi module connected event not fired

fedy2, 2017-07-25:
I'm trying to listen to the `connected` event from the wifi module but the callback function don't get called.

Here the code:

```
var wifi = require("Wifi");

wifi.on('connected', function(details) { 
  console.log('connected to wifi', details);
});

wifi.connect("*****", {password:"****"}, function(err){
    console.log("connected? err=", err, "info=", wifi.getIP());
});
```

Here the output:

```
>
=undefined
connected? err= null info= {
  "ip": "192.168.1.105",
  "netmask": "255.255.255.0",
  "gw": "192.168.1.1",
  "mac": "5c:**:7f:**:7a:**"
 }
```

MaBecker, 2017-07-26:
could be that you are already connected :)

you might add a `wifi.disconnect()` because the connect event is only fired if you change from not connected to connected.

fedy2, 2017-07-26:
Nice :). Never thought about that. I will check it.

MaBecker, 2017-07-27:
Feel free to close if done :)

wilberforce, 2017-07-27:
There is also an associated event - listen for that one too and see if that fires on a restart...

fedy2, 2017-07-29:
The associated event is not fired.

fedy2, 2017-07-29:
@MaBecker you're right, it was already connected to the wifi so no event should be fired. I would use the `getDetails` method to understand if it is already connected and react consequently.

closed 2017-07-29 by fedy2

MaBecker, 2017-07-29:
👍