espruino/Espruino #1206: NodeMCU.D0 value is undefined on ESP8266

fedy2, 2017-07-26:
I'm using a NodeMCU Amica board with Espruino. I'm printing the D0 equivalent pin name:

`console.log(NodeMCU.D0);`

But I get undefined as value:

`>undefined`

MaBecker, 2017-07-26:
It is defined, but returns undefined 

https://github.com/espruino/Espruino/blob/master/targets/esp8266/jswrap_nodemcu.c#L31

board definition names that pin B16

https://github.com/espruino/Espruino/blob/master/boards/ESP8266_BOARD.py#L59

MaBecker, 2017-07-26:
GPIO 16 is used to  wakeup the ESP from deep sleep and doesn't have an interrupt associated with it, so  this is why this port is not available. 

https://github.com/espruino/Espruino/blob/master/boards/ESP8266_4MB.py#L114

Please share what you like to use NodeMCU.D0 : GPIO16 for  to figure out if it makes sense to enable GPIO16

fedy2, 2017-07-26:
I'm using the GPIO16 to connect a DHT22 sensor. Why disable it a priori?

MaBecker, 2017-07-26:
just tested 
```
>console.log(NodeMCU.D0);
D16
=undefined
```
so lets create a pull request

MaBecker, 2017-07-26:
Running some test - It does not work ......

GPIO16 can be INPUT, OUTPUT, or INPUT_PULLDOWN_16 but not input_pullup, like in DHT22 module

    setTimeout(function() {pinMode(ht.pin,'input_pullup');},1);

code:

```
var dht = require("DHT22").connect(NodeMCU.D0);

setInterval(function() { 
  dht.read( function(a){
    console.log("Temp is "+a.temp.toFixed(1).toString()+
                " and RH is "+a.rh.toFixed(1).toString());
  });},3000);

```

using NodeMCU.D0 or D16 produces a stack dump and reboot

```
ASSERT(channel>=EV_EXTI0 && channel<=EV_EXTI_MAX) FAILED AT src/jsdevices.c:626
  #1[r3,l2] Object {
    #2[r1,l2] Name String [1 blocks] "\xFF"      #3[r1,l2] Object {
        #6[r1,l2] Name String [2 blocks] "timers"          #8[r2,l2] Array(2) [
            #59[r1,l3] Name Integer 1              #47[r1,l2] Object {
                #50[r1,l2] Name String [1 blocks] "time"= int 83
                #52[r1,l2] Name String [2 blocks] "interval"                  #51[r1,l2] Integer 3000000
                #57[r1,l2] Name String [2 blocks] "callback"                  #61[r1,l3] Function {
                    #36[r1,l2] Name String [1 blocks] "\xFFcod"                      #79[r1,l3] FlatString [10 blocks] "dht.read( function(a){\n    console.log(\"Temp is \"+a.temp.toFixed(1).toString()+\n                \" and RH is \"+a.rh.toFixed(1).toString());\n  });"
                  }
              }
          ]
        #9[r1,l2] Name String [2 blocks] "watches"          #11[r2,l1] Array(0) [ ]
        #13[r1,l2] Name String [1 blocks] "net"          #14[r1,l2] String [1 blocks] "\x04\x00\x00\x00\x00\xFF\xFF\xFF"
        #16[r1,l2] Name String [2 blocks] "history"          #20[r1,l1] Array(6) [
            #21[r1,l2] Name Integer 0              #17[r1,l1] String [3 blocks] "setTime(1501103265.575);"
            #28[r1,l2] Name Integer 1              #24[r1,l1] String [3 blocks] "Modules.removeAllCached();"
            #102[r1,l2] Name Integer 2              #103[r1,l1] String [69 blocks] "Modules.addCached(\"DHT22\",\"function DHT22(a){this.pin=a;}\\nDHT22.prototype.read=function(d,c){c||(c=10);var a='';var b=this;pinMode(b.pin),digitalWrite(b.pin,0),this.watch=setWatch(function(b){a+=0|b.time-b.lastTime>0.00005;},b.pin,{edge:'falling',repeat:!0}),setTimeout(function(){pinMode(b.pin,'input_pullup');},1),setTimeout(function(){pinMode(D5,'output'),digitalPulse(D5,0,1),clearWatch(b.watch),delete b.watch;var e=parseInt(a.substr(2,8),2)+parseInt(a.substr(10,8),2)+parseInt(a.substr(18,8),2)+parseInt(a.substr(26,8),2);e&&(e&255)==parseInt(a.substr(34,8),2)?d({raw:a,rh:parseInt(a.substr(2,16),2)*0.1,temp:parseInt(a.substr(19,15),2)*0.2*(0.5-a[18])}):c>1?setTimeout(function(){b.read(d,--c);},500):d({err:!0,checksumError:e>0,raw:a,temp:-1,rh:-1});},6);},exports.connect=function(a){return new DHT22(a);};\");"
            #35[r1,l2] Name Integer 3              #100[r1,l1] String [2 blocks] "var pin = NodeMCU.D0;"
            #37[r1,l2] Name Integer 4              #39[r1,l1] String [4 blocks] "var dht = require(\"DHT22\").connect(pin);"
            #62[r1,l2] Name Integer 5              #63[r1,l1] String [16 blocks] "setInterval(function() {\n  dht.read( function(a){\n    console.log(\"Temp is \"+a.temp.toFixed(1).toString()+\n                \" and RH is \"+a.rh.toFixed(1).toString());\n  });},3000);"
          ]
        #32[r1,l2] Name String [2 blocks] "modules"          #34[r1,l1] Object {
            #54[r1,l2] Name String [1 blocks] "Wifi"              #55[r1,l1] NativeFunction 0x40216368 (0) { }
            #99[r1,l2] Name String [2 blocks] "DHT22"              #175[r2,l1] Object {
                #243[r1,l2] Name String [2 blocks] "connect"                  #245[r1,l1] Function { return
                    #246[r1,l2] Name Param "\xFFa"                       undefined
                    #249[r1,l2] Name String [1 blocks] "\xFFcod"                      #247[r1,l1] String [2 blocks] "new DHT22(a);"
                    #250[r1,l2] Name String [1 blocks] "\xFFsco"                      #174[r5,l2] ...
 
                  }
              }
          }
      }
    #22[r1,l2] Name String [2 blocks] "Modules"      #23[r1,l1] NativeFunction 0x40216278 (1) { }
    #101[r1,l2] Name String [1 blocks] "pin"      #31[r2,l2] Pin 16
    #172[r1,l2] Name String [2 blocks] "NodeMCU"      #173[r1,l1] NativeFunction 0x40216264 (1) { }
    #38[r1,l2] Name String [1 blocks] "dht"      #43[r7,l3] Object {
        #44[r1,l2] Name String [2 blocks] "__proto__"          #188[r2,l1] Object {
            #189[r1,l2] Name String [2 blocks] "constructor"              #182[r2,l1] Function {
                #183[r1,l2] Name Param "\xFFa"                   undefined
                #186[r1,l2] Name String [1 blocks] "\xFFcod"                  #184[r1,l1] String [2 blocks] "this.pin=a;"
                #187[r1,l2] Name String [1 blocks] "\xFFsco"                  #174[r5,l2] Object {
                    #176[r1,l2] Name String [2 blocks] "exports"                      #175[r2,l1] ...
 
                    #178[r1,l2] Name String [2 blocks] "module"                      #174[r5,l3] ...
 
                    #180[r1,l2] Name String [2 blocks] "DHT22"                      #182[r2,l2] ...
[... 17 lines cut ...]
              }
          }
        #49[r1,l2] Name String [1 blocks] "pin"          #31[r2,l2] ...
 
      }
  }
---console end---
 ets Jan  8 2013,rst cause:2, boot mode:(3,6)
load 0x40100000, len 2408, room 16
tail 8
chksum 0xe5
load 0x3ffe8000, len 776, room 0
tail 8
chksum 0x84
load 0x3ffe8310, len 632, room 0
tail 8
chksum 0xd8
csum 0xd8
2nd boot version : 1.6
  SPI Speed      : 80MHz
  SPI Mode       : QIO
  SPI Flash Size & Map: 32Mbit(1024KB+1024KB)
jump to run user1 @ 1000
âìnìròn|ìll`ârlþ
```

MaBecker, 2017-07-27:
Is there a smarter way then to add if to exclude GPIO16 for input_pullup in this section?

https://github.com/espruino/Espruino/blob/master/targets/esp8266/jshardware.c#L385-L406

MaBecker, 2017-07-27:
Hmm, there is much more to change to make GPIO16 work....

Bad news: The SDK does not support GPIO16 as a regular GPIO.

gfwilliams, 2017-07-27:
I imagine that EXTI assert is because the code is trying to add a watch to that pin? That's what EXTI is usually used for. That's *really* unlikely to work even if you can get the GPIO working. There should probably be a check for that?

Realistically I think the best bet would be to create a special-case for D16 handling where needed...

Honestly, the initial bug with `console.log(NodeMCU.D0);` printing `undefined` seems just to be what you pointed out and what's mentioned in the FAQ, which isn't a bug at all though?

MaBecker, 2017-08-04:
There is a sample implementation on githup for [gpios16.c](https://github.com/espressif/ESP8266_NONOS_SDK/blob/master/driver_lib/driver/gpio16.c) and [gpios16.h](https://github.com/espressif/ESP8266_NONOS_SDK/blob/master/driver_lib/include/driver/gpio16.h)

Lets keep this open for some weeks maybe someone likes to jump in and do the implementation, if not lets close it.

MaBecker, 2017-09-02:
Got a new [ESP8266 device](https://www.tindie.com/products/lspoplove/wifi-deauther-v-15pre-flashed-pocket8266/) with a 18650 that uses GPIO16 to indicate power status.

Check this [commit ](https://github.com/MaBecker/Espruino/commit/8f0e60e93665355e518a7ab0a4c477342a7fb400)to see  how far  I got  with the implementation of GPIO16.

tested:
```
pinMode(D16,"output");
getPinMode(D16);
D16.toggle();
NodeMCU.D0.toggle();
digitalRead(D16);
digitalWrite(D16,0);
pinMode(D16,"input");
getPinMode(D16);
```

open:
[ ] D16.write(1) does not work if D16 is defined as input.

Any hints or comments are welcome.

MaBecker, 2017-09-02:
[ ] reject setWatch() for D16

MaBecker, 2017-09-03:
@gfwilliams is there a way to define D16 as invalid watch pin?

gfwilliams, 2017-09-04:
Just return EV_NONE here I think: https://github.com/espruino/Espruino/blob/master/targets/nrf5x/jshardware.c#L780

MaBecker, 2017-09-04:
Thanks for that hint.

added another [commit ](https://github.com/MaBecker/Espruino/commit/1da4f5475899956dcea6238c5e40ed02e8bd34b5)

```
>setWatch(function() {console.log('Hi');}, D16, {repeat:true, edge:'falling'});
WARNING: Unable to set watch. You may already have a watch on a pin with the same number (eg. A0 and B0),
or it cannot be used with watch
```
[x] pinMode(D16,"output");
[x] pinMode(D16,"input");
[x] getPinMode(D16);
[x] D16.toggle();
[x] digitalRead(D16);
[x] digitalWrite(D16,0);
[x] digitalWrite(D16,1);
[x] reject setWatch() for D16
[ ] D16.write(1) or digitalWrite(D16,1) does switch to output and write

wilberforce, 2017-09-04:
@MaBecker 
Hope you don't mind - edited your comment by adding `-` to the front of `- [ ]` they now render as lists

MaBecker, 2017-09-05:
Nice !

MaBecker, 2017-09-14:
added pullrequest #1243

gfwilliams, 2017-09-15:
Fixed - thanks!

closed 2017-09-15 by gfwilliams