espruino/Espruino #1207: ESP8266 NONOS SDK  2.1

MaBecker, 2017-07-30:
As version 2.1 is now available since  May 2017 let give it a try :)

Overview:
* [release info](https://github.com/espressif/ESP8266_NONOS_SDK/releases/tag/v2.1.0)
* [download](https://github.com/espressif/ESP8266_NONOS_SDK/archive/v2.1.0.zip)

Code changes:
* [make/family/ESP8266.make](https://github.com/MaBecker/Espruino/blob/master/make/family/ESP8266.make#L76-L80) 
* [targets/esp8266/espmissingincludes.h](https://github.com/MaBecker/Espruino/blob/master/targets/esp8266/espmissingincludes.h#L61-L82)

Used libraries: NET, TELNET, GRAPHICS,  CRYPTO and NEOPIXEL

make statement

* RELEASE=1 ESPSDK_2_1=1 make

The result looks promising, freeHeap: 11264 

     _____                 _
    |   __|___ ___ ___ _ _|_|___ ___
    |   __|_ -| . |  _| | | |   | . |
    |_____|___|  _|_| |___|_|_|_|___|
              |_| http://espruino.com
     1v93.3119 Copyright 2016 G.Williams 
    Espruino is Open Source. Our work is supported
    only by sales of official boards and donations:
    http://espruino.com/Donate
    Flash map 4MB:1024/1024, manuf 0xe0 chip 0x4016

    >require('ESP8266').logDebug(false);
    =undefined
    >require("ESP8266").getState(); 
    ={
      "sdkVersion": "2.1.0(116b762)",
      "cpuFrequency": 160, "freeHeap": 11264, "maxCon": 10,
      "flashMap": "4MB:1024/1024",
      "flashKB": 4096,
      "flashChip": "0xe0 0x4016"
    }
    >process.env
    ={
      "VERSION": "1v93.3120",
      "BUILD_DATE": "Jul 30 2017",
      "BUILD_TIME": "11:04:34",
      "GIT_COMMIT": "ed27629121172c7c090cae3a6273feb6227d48ef",
      "BOARD": "ESP8266_4MB",
      "CHIP": "ESP8266",
      "CHIP_FAMILY": "ESP8266",
      "FLASH": 0, "RAM": 81920,
      "SERIAL": "5ccf7fc1-120b",
      "CONSOLE": "Serial1",
      "EXPORTS": { "jsvLock": 1075851304, "jsvLockAgainSafe": 1075851356, "jsvUnLock": 1075851728, "jsvSkipName": 1075860364,
        "jsvMathsOp": 1075865696, "jsvMathsOpSkipNames": 1075867412, "jsvNewFromFloat": 1075858076, "jsvNewFromInteger": 1075857976, "jsvNewFromString": 1075857620,
        "jsvNewFromBool": 1075858044, "jsvGetFloat": 1075863436, "jsvGetInteger": 1075859860, 
        "jsvGetBool": 1075863796, "jspeiFindInScopes": 1075879024,
        "jspReplaceWith": 1075878744, "jspeFunctionCall": 1075889112, "jspGetNamedVariable": 1075880868, "jspGetNamedField": 1075881312, "jspGetVarNamedField": 1075881444,
        "jsvNewWithFlags": 1075857344 }
     }
    >

MaBecker, 2017-08-03:
Code changes:

* [make/targets/ESP8266.make](https://github.com/MaBecker/Espruino/blob/master/make/targets/ESP8266.make#L36-L40)

MaBecker, 2017-08-15:
@gfwilliams, please let me know if this can go to Espruino master, so I can prepare the PRs.

wilberforce, 2018-04-09:
@MaBecker 

V2.2 is out now.

ESP8266 non-OS SDK V2.2.0 | NON-OS SDK | V2.2.0 | 2018.02.07 |  

https://www.espressif.com/en/support/download/sdks-demos?keys=&field_type_tid%5B%5D=14

wilberforce, 2018-04-09:
We are still getting the V2.0 tool chain from tve's server:

https://github.com/espruino/Espruino/blob/master/scripts/provision.sh#L60-L69

```
curl -Ls http://s3.voneicken.com/esp_iot_sdk_v2.0.0.p1.tgx | tar Jxf - --no-same-owner
curl -Ls http://s3.voneicken.com/xtensa-lx106-elf-20160330.tgx | tar Jxf - --no-same-owner
```
Once 2.2 is tested and working, this could be updated and moved to here:
https://github.com/espruino/EspruinoBuildTools

gfwilliams, 2018-04-09:
Eep. Yes, we should definitely be trying to drag stuff from EspruinoBuildTools now :)

MaBecker, 2018-04-09:
2.2 has some link issues

MaBecker, 2018-04-10:
like this:
```
/Volumes/Espruino/gcc/xtensa-lx106-elf-4.8.5/bin/../lib/gcc/xtensa-lx106-elf/4.8.5/../../../../xtensa-lx106-elf/bin/ld: espruino_esp8266_user2.elf section `.text' will not fit in region `iram1_0_seg'
collect2: error: ld returned 1 exit status
make: *** [espruino_esp8266_user2.elf] Error 1
make: *** Waiting for unfinished jobs....
/Volumes/Espruino/gcc/xtensa-lx106-elf-4.8.5/bin/../lib/gcc/xtensa-lx106-elf/4.8.5/../../../../xtensa-lx106-elf/bin/ld: espruino_esp8266_user1.elf section `.text' will not fit in region `iram1_0_seg'
collect2: error: ld returned 1 exit status
make: *** [espruino_esp8266_user1.elf] Error 1

```

MaBecker, 2018-04-10:
build chain plus SDK 2.2

https://github.com/espruino/Espruino/blob/master/README_Building.md#under-macos

MaBecker, 2018-04-10:
Ok, have to replace the LD file with the new version coming with der SDK 2.2.

MaBecker, 2018-04-10:
There are some  issue named for 2.2 at https://github.com/espressif/ESP8266_NONOS_SDK/issues

MaBecker, 2018-04-13:
SDK 2.2 sounds very promising ([release notes](https://github.com/espressif/ESP8266_NONOS_SDK/releases))

will use this [wifi test](https://github.com/espruino/Espruino/tree/master/libs/network/esp8266/tests) created by  tve to get a go or nogo for this SDK

MaBecker, 2018-04-15:
added ESP82666 SDK to Build 2.0, 2.1 and 2.2 to EspruinoBuildTools

MaBecker, 2018-04-15:
Build and change result for Espruino 1.96 with SDK 2.1  - Status: stable

````
console.log('\n',process.env,'\n',
            require("ESP8266").getState(),
            '\n',process.memory());

/*

Buid options: RELEASE=1 BOARD_4MB=1
SDK: 2.1 
Changes:
    targets/esp8266/espmissingincludes.h remove entries
    make/family/ESP8266.make: remove CFLAG mfore-l32, because of increase in size around 350K

To disassemble: xtensa-lx106-elf-objdump -d -l -x espruino_esp8266_user2.elf
Sections:
Idx Name          Size      VMA       LMA       File off  Algn
  3 .text         000076d4  40100000  40100000  00002d3c  2**2
                  CONTENTS, ALLOC, LOAD, READONLY, CODE
  4 .irom0.text   0006c1e4  40201010  40201010  0000a410  2**4
                  CONTENTS, ALLOC, LOAD, READONLY, CODE
To disassemble: xtensa-lx106-elf-objdump -d -l -x espruino_esp8266_user1.elf
xtensa-lx106-elf-objdump -d -l -x espruino_esp8266_user1.elf >espruino_esp8266_user1.lst
  8 -rwxr-xr-x  1 markbecker  staff    2188 15 Apr 07:20 eagle.app.v6.data.bin
872 -rwxr-xr-x  1 markbecker  staff  442852 15 Apr 07:20 eagle.app.v6.irom0text.bin
 24 -rwxr-xr-x  1 markbecker  staff    9164 15 Apr 07:20 eagle.app.v6.rodata.bin
 64 -rwxr-xr-x  1 markbecker  staff   30420 15 Apr 07:20 eagle.app.v6.text.bin
** user1.bin uses  484692 bytes of 962560 available

>
 {
  "VERSION": "1v96.3780",
  "GIT_COMMIT": "34d678da",
  "BOARD": "ESP8266_4MB",
  "FLASH": 0, "RAM": 81920,
  "SERIAL": "5ccf7f05-c100",
  "CONSOLE": "Serial1",
  "MODULES": "Flash,Storage,net,dgram,http,NetworkJS,Wifi,ESP8266,TelnetServer,crypto,neopixel",
  "EXPTR": 1073643636 }
 {
  "sdkVersion": "2.1.0(116b762)",
  "cpuFrequency": 160, "freeHeap": 12016, "maxCon": 10,
  "flashMap": "4MB:1024/1024",
  "flashKB": 4096,
  "flashChip": "0xe0 0x4016"
 }
 { "free": 1500, "usage": 100, "total": 1600, "history": 0,
  "gc": 0, "gctime": 1.324 }
=undefined
>

*/

````

MaBecker, 2018-04-15:
Build  result for Espruino 1.96 with SDK 2.2 - Status: infinitely reboot loop after flash

using the targets/esp8266/eagle.app.v6.new.2048.ld ld complains:

```
/Volumes/ESPTools/esp-open-sdk/xtensa-lx106-elf/bin/../lib/gcc/xtensa-lx106-elf/4.8.5/../../../../xtensa-lx106-elf/bin/ld: espruino_esp8266_user1.elf section `.text' will not fit in region `iram1_0_seg'
collect2: error: ld returned 1 exit status
make: *** [espruino_esp8266_user1.elf] Error 1
make: *** Waiting for unfinished jobs....
/Volumes/ESPTools/esp-open-sdk/xtensa-lx106-elf/bin/../lib/gcc/xtensa-lx106-elf/4.8.5/../../../../xtensa-lx106-elf/bin/ld: espruino_esp8266_user2.elf section `.text' will not fit in region `iram1_0_seg'
collect2: error: ld returned 1 exit status
make: *** [espruino_esp8266_user2.elf] Error 1
```

wilberforce, 2018-04-15:
@MaBecker 
`espruino_esp8266_user1.elf section `.text' will not fit in region `iram1_0_seg'`

This means that the RAM parts of firmware are now to big to fit within the RAM memory area.

Are there constants that are not declared as constants that could to ROM?

have you tried comping turning some features of such as Crypto?

MaBecker, 2018-04-15:
Looks like sdk 2.2 libs Need more space.

https://github.com/espressif/ESP8266_NONOS_SDK/blob/master/lib/readme.md

MaBecker, 2018-04-16:
Thanks @wilberforce for that hint.

Removing  CRYPTO and  GRAPHICS does not help....

Looks like Espressif did changes to their libs and now lots of stuff is puled into .irom0.text

increasing Memory

```
MEMORY
{
  dport0_0_seg :                        org = 0x3FF00000, len = 0x10
  dram0_0_seg :                         org = 0x3FFE8000, len = 0x14000
  iram1_0_seg :                         org = 0x40100000, len = 0x20000         <-  0x8000
  irom0_0_seg :                         org = 0x40201010, len = 0xE0000
}
```

result:
```
Idx Name          Size      VMA       LMA       File off  Algn
  3 .text         00016e28  40100000  40100000  00002d4c  2**2
                  CONTENTS, ALLOC, LOAD, READONLY, CODE
  4 .irom0.text   0005c8e4  40201010  40201010  00019b80  2**4
                  CONTENTS, ALLOC, LOAD, READONLY, CODE
```

Sorry no idea how to handle this.

MaBecker, 2018-04-17:
Could need some help.

wilberforce, 2018-04-17:
It is ram that is the issue.  You are constrained by the amount of physical ram -- so changing sizes won't help.

Try reducing the jsvars to see if for you can get a build.

I seem to recall the libc.a was added to the project at some point - you might want to link against the latest version.

MaBecker, 2018-04-17:
Well removing CRYPTO, GRAPHICS and shrink variable to 500 does not fix this.

MaBecker, 2018-04-17:
Change everything back to default and run a test with this ld file

https://github.com/nodemcu/nodemcu-firmware/blob/master/ld/nodemcu.ld

linking now works again

```
Sections:
Idx Name          Size      VMA       LMA       File off  Algn
  0 .text         00006464  40100000  40100000  00001b60  2**2
                  CONTENTS, ALLOC, LOAD, READONLY, CODE
  6 .irom0.text   0006d900  40210000  40210000  00009000  2**12
                  CONTENTS, ALLOC, LOAD, READONLY, CODE
To disassemble: xtensa-lx106-elf-objdump -d -l -x espruino_esp8266_user1.elf
xtensa-lx106-elf-objdump -d -l -x espruino_esp8266_user1.elf >espruino_esp8266_user1.lst
  8 -rwxr-xr-x  1 markbecker  staff    2188 17 Apr 16:47 eagle.app.v6.data.bin
880 -rwxr-xr-x  1 markbecker  staff  448768 17 Apr 16:47 eagle.app.v6.irom0text.bin
  8 -rwxr-xr-x  1 markbecker  staff       8 17 Apr 16:47 eagle.app.v6.rodata.bin
 56 -rwxr-xr-x  1 markbecker  staff   25700 17 Apr 16:47 eagle.app.v6.text.bin
** user1.bin uses  476724 bytes of 962560 available
```

MaBecker, 2018-04-17:
put this 2.2 on hold and run test for esp01 and esp12

MaBecker, 2018-08-02:
SDK after 2.2 do not work with because Espressif decided to compile  some libraries with -ffunction-sectons -fdata-sections and modify ld to put this.text.* into .irom0.text

Question is it possible to reverse this with ld?

gfwilliams, 2018-08-02:
Is it bad that they've put the text sections into irom0? I would have thought that'd be good?

MaBecker, 2018-08-02:
Me too, but now it does not build any more because text section is too big, will invest some time on weekend and hopefully come up with some details to discuss.

MaBecker, 2018-08-02:
Update:

changed targets/esp8266/espmissingincludes.h to work with SDK 2.2

and added Espruino stuff to  eagle.app.v6.new.2048.ld  from SDK 2.2

Now got  a stable Espruino version with SDK 2.2  for a ESP8266_4MB

    irom0_0_seg :                         org = 0x40281010, len = 0xE0000

    >require("ESP8266").getState();
    ={
      sdkVersion: "2.2.0(f28eaf2)",
      cpuFrequency: 160, freeHeap: 9416, maxCon: 10,
      flashMap: "4MB:1024/1024",
      flashKB: 4096,
      flashChip: "0xe0 0x4016"
     }

Bad news for ESP8266_BOARD:

    `.text' will not fit in region `iram1_0_seg'

      irom0_0_seg :                         org = 0x40281010, len = 0x7C000

wilberforce, 2018-08-02:
Any idea of much it is too big by?

What size is the 4Mb firmware

MaBecker, 2018-08-03:
> Any idea of much it is too big by?

No, but see attached file for all sections created with 
xtensa-lx106-elf-objdump -h espruino_esp8266_partial.o > espruino_esp8266_user1.lst.txt
[espruino_esp8266_user1.lst.txt](https://github.com/espruino/Espruino/files/2255539/espruino_esp8266_user1.lst.txt)

> What size is the 4Mb firmware?

    ** user1.bin uses  528644 bytes of 962560 available

MaBecker, 2018-08-07:
like to switch to SDK 2.2 or  2.2.1 for 4mb boards. 

Gordon, how do you think about it?

MaBecker, 2018-08-07:
Wait, missed to replace eagle.app.v6.new.1024 files, now the SSDK 2.2  works for all ESP8266 boards.

> What does it add?

If you mean features: To much to name it, so just added links below

* [release 2.1](https://github.com/espressif/ESP8266_NONOS_SDK/releases/tag/v2.1.0)
* [release 2.2](https://github.com/espressif/ESP8266_NONOS_SDK/releases/tag/v2.2.0)
* [release 2.2.1](https://github.com/espressif/ESP8266_NONOS_SDK/releases/tag/v2.2.1)

Now starting test for wifi, i2c, spi and neopixel.

Any thing else to test?

gfwilliams, 2018-08-08:
Then yes, I'd definitely be up for changing it.

It'd also be nice to get the relevant files into the EspruinoBuildTools repo so we're not thrashing @tve's S3 account with Travis :)

gfwilliams, 2018-08-14:
`./espruino --test-all` - there's always `espruino --help`.

There's no way to run tests on the device itself though.

MaBecker, 2018-08-15:
Well I was thinking of some code like this to do on device testing (odt)

```
/* onDeviceTest_v0.js

try to run espruino test on device

load a test.js file from webserver into a string

*/

// for now static
// maybe store all in flash and load from flash test by test
// or load from webserver test by test
// or ....
catalog = [
  ["test_number_comparison.js",
   `var a = 42; result = a==42;`],
  ["test_tostring2.js",
    `0x8000000000000000.toString(2) // could segfault :)
     result = (8446744073709551616).toString(16) == "7538dcfb76180000"`
  ],
  ["test_undefined_behaviour.js",
`var testUndefined;        // variable declared but not defined, set to value of undefined
var testObj = {};  

var a = 0;
if ((""+testUndefined) != "undefined") a = 1; // test variable exists but value not defined, displays undefined
if ((""+testObj.myProp) != "undefined") a = 2; // testObj exists, property does not, displays undefined
if (!(undefined == null)) a = 3;  // unenforced type during check, displays true
if (undefined === null) a = 4;// enforce type during check, displays false
if (null != undefined) a = 5;  // unenforced type during check, displays true
if (null === undefined) a = 6; // enforce type during check, displays false
if (undefined != undefined) a = 7;
if (!(undefined == undefined)) a = 8;
result = a==0;`]
];

function onDeviceTest(test){
  var result, error;
  try {
    result = eval(test[1]);
  } catch(e) {
    error = e;
    result = false;
  }
  console.log(test[0],'[',result,',',error,']');
}

for( i = 0; i< catalog.length;i++)
  onDeviceTest(catalog[i]);

```

MaBecker, 2018-08-15:
got Espruino Command-line Tool 0.1.12, but looks like this is not the one to be used for --test-all

> ./espruino --test-all 

Marks-MacBook-Pro:Espruino markbecker$ espruino --test-all --board ESP8266_4MB -p /dev/cu.usbserial -b 115200 
/usr/local/Cellar/node@8/8.9.4/lib/node_modules/espruino/bin/espruino-cli.js:139
   } else throw new Error("Unknown Argument '"+arg+"', try --help");
          ^

Error: Unknown Argument '--test-all', try --help
    at Object.<anonymous> (/usr/local/Cellar/node@8/8.9.4/lib/node_modules/espruino/bin/espruino-cli.js:139:17)

wilberforce, 2018-08-15:
the `espruino` in this case is the linux target.

`make clean && make`

This will make ./espruino in your build folder which you can then run.

gfwilliams, 2018-08-16:
I've been after doing actual on-device testing for some time - this issue really isn't the place to discuss that though. I started a new issue at https://github.com/espruino/Espruino/issues/1490

MaBecker, 2018-08-16:
Updated EspruinoBuildTools, now there are all available 2.x SDKS.

Decide which to choose after testing.

MaBecker, 2018-09-02:
will be updated and extended

- [ ] [wifi tests created by tve](https://github.com/espruino/Espruino/tree/master/libs/network/esp8266/tests) 
  ....
- [x] Wifi station
  - [x] wifi connect to AP
  - [x] wifi save / restore / reconnect after reboot
  - [x] simple http server
  - [x] mqtt send and receive 
- [ ] Wifi AP
  ....
- [ ] modules
  - [x] neopixel
  - [ ] neopixel (APA102) with SPI 
  - [x] i2c oled
  - [ ] spi oled
  - [ ] DHT22/AM230x/RHT0x Temperature and RH Sensor
  - [ ] DS18B20 Temperature Sensor
  ....

MaBecker, 2018-09-10:
Discovered that  ESP8266 AP is going down sometimes for some seconds even if stations are connected.

[image]
Monitored with Wifi Explorer for Mac OS X

MaBecker, 2018-10-15:
How to preapare Travis build stuff?

Wanted SDK is 2.2.1 and it is uploaded as zip file to Golfer https://github.com/espruino/EspruinoBuildTools/tree/master/esp8266

gfwilliams, 2018-10-16:
How ready are we to merge this? I'd like to get a new release of Espruino out this week (ideally tomorrow).

Sounds like it's unlikely we'd get all the tools put in and have it tested in the next 2 days?

MaBecker, 2018-10-16:
Thanks @wilberforce this was exactly what I was looking for.

Ok Gordon lets cross fingers an see how travis will handle this.

MaBecker, 2018-10-16:
```
 ____                 _
|  __|___ ___ ___ _ _|_|___ ___
|  __|_ -| . |  _| | | |   | . |
|____|___|  _|_| |___|_|_|_|___|
         |_| espruino.com
 1v99.4205 (c) 2018 G.Williams
Espruino is Open Source. Our work is supported
only by sales of official boards and donations:
http://espruino.com/Donate
Flash map 4MB:1024/1024, manuf 0xe0 chip 0x4016
>
 {
  "VERSION": "1v99.4205",
  "GIT_COMMIT": "066cd080",
  "BOARD": "ESP8266_4MB",
  "FLASH": 0, "RAM": 81920,
  "SERIAL": "5ccf7f05-c100",
  "CONSOLE": "Serial1",
  "MODULES": "Flash,Storage,heatshrink,net,dgram,http,NetworkJS,Wifi,ESP8266,TelnetServer,crypto,neopixel",
  "EXPTR": 1073643636 }
 {
  "sdkVersion": "2.2.1(6ab97e9)",
  "cpuFrequency": 160, "freeHeap": 11008, "maxCon": 10,
  "flashMap": "4MB:1024/1024",
  "flashKB": 4096,
  "flashChip": "0xe0 0x4016"
 }
 { "free": 1492, "usage": 108, "total": 1600, "history": 6,
  "gc": 0, "gctime": 1.812 }
[
  "Flash",
  "Storage",
  "heatshrink",
  "net",
  "dgram",
  "http",
  "NetworkJS",
  "Wifi",
  "ESP8266",
  "TelnetServer",
  "crypto",
  "neopixel"
 ]
> 
```

commit 38c84039 (2018-10-18): ESP8266: switch to SDK 2.2.1 (fix #1207)
commit 471eb287 (2018-10-18): Merge pull request #1528 from espruino/ESP8266_SDK_2.2.1

ESP8266: switch to SDK 2.2.1 (fix #1207)
closed 2018-10-18 by gfwilliams