espruino/Espruino #1208: ESP32: Addind Bluetooth support

Sparow199, 2017-08-01:


gfwilliams, 2017-08-01:
Ok, so you want someone else to add bluetooth for you - but you can't be bothered to write a description or even to spell the title correctly? Nice. :(

recidive, 2017-10-01:
Hello @gfwilliams. Could you tell me what's is involved in developing support for Bluetooth and BLE for the ESP32? Then I could try and do that myself and contribute back to the project?

Thanks!

jumjum123, 2017-10-01:
There are a lot problems, getting Bluetooth/BLE running for Espruino on
ESP32.
With Classic Bluetooth you cannot even start Espruino anymore, lack of heap
during start.
With BLE, size of binary becomes very big, first test was about 500kb more
and binary arrived at >1300kb

Actually I'm testing standard Espruino and add Gatts Server example.
Since this is my hobby, don't expect results soon.

gfwilliams, 2017-10-02:
Thanks @jumjum123 - I think you know more about this than me. The first step would definitely be getting EspressIF's BLE examples compiled in like you're doing.

After that, it'd be a job of trying to get `jswrap_bluetooth.c` to compile without being dependent on Nordic's libraries

jumjum123, 2017-10-24:
@gfwilliams - first of all I tried to get the ESP example for BLE server running in Espruino. This is working now with a lot of restrictions. Restrictions like reducing available jsvars down to 2000.
After taking a first look to jswrap_bluetooth.c, I decided to start testing with a different kind of wrapper.
Main reason is that my test is on a high level, and it is not nrf oriented.
Lot of functionality needs to be done. Anyway, feedback is welcome
Right now I have a wrapper like this:
- class BLE
- method BLE.init_bt() to initiate low level support for bluetooth
- method BLE.init_gatts() starts a gatts server with handles for GAP and GATTS, uses UUIDs from ESP32-example
- method BLE.setDeviceName("myName") set device name for scanning from clients
- method BLE.setServiceUUID(128 bit uuid) sets ServiceUUID, without causing cache refresh on client
- method BLE.setReadFunction(function()......) function is used on read from client and returns a string
- method BLE.notify("notify this") notifies client, sending a new value
- event connect fired if a client connects, right now without any parameter
- event disconnect fired if a client disconnects, without any parameter
- event data fired if a client sends data, with data as string parameter
- method BLE.scan() should emit events for each BLE device found, under heavy construction

gfwilliams, 2017-10-25:
Hi, it's great that you have it going - a bit sad that it takes so much RAM though :(

Do you have a link to your code available somewhere?

I'm really against coming up with a brand new API for BLE though -  I want to avoid the problem we have for internet stuff where we have a bunch of subtly different APIs for Ethernet, WiFi, WiFi on ESP8266, and GSM.

Very little in the existing NRF class is actually specifically for Nordic chips. I totally understand why you'd want your own API for testing to avoid having to `#ifdef` existing Nordic code, but in terms of getting something in mainline Espruino I really want to keep the same API. You'll also find that a huge amount of the work has already been done for you, so it'll make life easier in the long run.

(Renaming the existing NRF class to BLE is something we could easily do at any point though)

It feels like given what you have implemented you've proved that it'll work, and 2k vars is still pretty reasonable - so in a way it makes more sense to now try and work on code on top of the NRF API, as at some point soon you'll end up spending a lot of time reimplementing code that already exists and is tested in the nRF5x port.

diogoviannaaraujo, 2017-10-26:
Me, as @recidive, am too interested of contributing my own time in getting this to work.
Not a experienced C programmer here, but ill have lots of time this end of year.

if @jumjum123 can share his work maybe we could use it as a starting point

jumjum123, 2017-10-26:
Take a look to github targets/ESP32/docs in branch ESP32. There is a new folder with all changes created during my testing.
Its more or less a proof of concept, based on examples from ESP-IDF, Neil Kolban and pcbreflux.

jumjum123, 2017-10-30:
@gfwilliams Itook a closer look to Bluetooth classes in Espruino.
Based on my understanding they support Espruino in Clientmode
Looks like following description from  Web Bluetooth Community Group
In my approach I started from the other direction and use Espruino as Service

NRF class is different from that, but as name says, this is strongly nrf oriented
I wonder, how we could (or should ?) marry nrf and bluetooth components in esp-idf

Porting Bluetooth to Espruino is my way to get a better understanding of BLE.
As long as I don't know better, I will follow this, and cleanup the (IMHO cluttered) example from esp-idf.
My next step will be to add setting of characteristics and descriptors for service(s)
There is a nice implementation https://gist.github.com/heiko-r/f284d95141871e12ca0164d9070d61b4

gfwilliams, 2017-10-30:
Espruino supports both as a peripheral (NRF.setAdvertising/setServices) and as a central, in exactly the same form as the Web Bluetooth group (NRF.requestDevice/connect, then BluetoothDevice, and BluetoothRemote*).

It'd be worth looking at all the existing Puck.js tutorials to see what it's like using Bluetooth: http://www.espruino.com/Puck.js#tutorials

But using BLE as a service, literally all you need is to implement 3 functions - NRF.setServices, setAdvertising, and updateServices.

I understand that this is just a fun way to learn BLE, and that's great - obviously you're free to do whatever you want. However I'm not going to merge in any code that uses a totally different API for Bluetooth - it's just going to be a nightmare to support.

I mean, we have a whole bunch of tutorials, modules, and code in EspruinoDocs showing people how to use BLE. If you come up with a different API then that's all of no use - and worse, it'll actually confuse anyone trying to use BLE on Espruino.

jumjum123, 2017-10-30:
Fully understand, it would be a nightmare to support ESP32 specific Bluetooth and NRF specific Bluetooth and xyz specific ....

Naming my testing class BLE is nothing which should survive, its a working name nothing more.
Long time target is to match existing naming and functions whenever possible.
But thats 2nd step, 1st I have to understand whats going on behind the scenes.

On the other hand, its a bit confusing to have a wrapper named NRF.... for hardware from somewhere else.
IMHO it would be a better approach to move this function to a bluetooth class, e.g. Bluetooth.localGattServer. As far as I can see Web Bluetooth Group has no idea to this (yet ?).

To avoid problems with existing software/docu/etc may be, we could start with an additional wrapper. I will try to create an example for that, since an example often tells more than my wording ;-)

skeevestevens, 2018-02-26:
Hey guys.  If BLE is a challenge on ESP32, what other platforms does it work well on (other than the puck which is awesome).  I want to settle on Espruino as a platform and would like to have a cheap little companion device similar to an ESP32 or 8266 based thing.. what else is out there?

gfwilliams, 2018-02-26:
You want something that's more of a bare PCB? If so there'll be an Espruino device coming pretty soon like that

jumjum123, 2018-02-27:
For ESP32, we are (still) waiting for next release of ESP-IDF.
Hopefully I get more to know tomorrow on Embedded World 2018 in Nürnberg.
Bluetooth for server and client works, at least in my tests.
To put water into the wine, only some NRF-commands are supported.
There have been some people offering help for BLE, but at the end, ....

skeevestevens, 2018-02-27:
We're playing a lot with beacons at the moment... they actually have a lot of interesting and practical uses. We're testing a lot of simple devices with basic or minimal sensors, or just the BLE itself. 
Full-featured are the best obviously, but not every deployment needs all the sensors, and the cheaper the units the bigger the deployment or more potential solutions.

wilberforce, 2018-05-14:
Ble support on the esp32 has just been merged into the master branch!

Yay!

https://github.com/espruino/Espruino/commit/11ac45f88aac1ebe04319c587f6558e5f1000126

closed 2018-05-14 by wilberforce

BananaAcid, 2020-11-06:
Maybe the docs should be updated, to reflect this merge @ https://www.espruino.com/ESP32 -> Bluetooth :)

MaBecker, 2020-11-06:
Yes definitely, there is an open issue list about missing documentation, have a look #1770 and feel free to contribute.

nistvan86, 2020-11-24:
Can we use WiFi and Bluetooth at the same time? Eg. scan for beacons and send the result on WiFi to somewhere?

MaBecker, 2020-11-24:
Yes

commit 1d521eef (2023-02-27): Bangle.js2: Fix Compass Z heading (fix #1208)