espruino/Espruino #1210: make topstrings and topreadonly work on mac os x

MaBecker, 2017-08-03:
had to change the expr call used with topstrings and topreadonly to work on mac os x.

Can some please verify if this breaks other os - thanks.

https://github.com/MaBecker/Espruino/blob/master/make/targets/ESP8266.make#L121

https://github.com/MaBecker/Espruino/blob/master/make/targets/ESP8266.make#L135

MaBecker, 2017-08-04:
This needs to be verified on a other os than Mac OS X:

```
# original
expr match "$(awk --version)"  "GNU.*"

# changed to work on Mac OS X 
expr "$(awk --version)" : "GNU.*"
````

both expr calls should return same result

gfwilliams, 2017-08-04:
Just tried on linux and:

```
$ expr match "$(awk --version)"  "GNU.*"
720
$ expr "$(awk --version)" : "GNU.*"
720
```

So I guess that's what you'd expect. Both fail on Pi, but that's because of `awk --version` failing, not the change

MaBecker, 2017-08-04:
Yes, thanks. So let me create a PR and close this issue

closed 2017-08-04 by gfwilliams