espruino/Espruino #1219: ESP32 Travis build failing

gfwilliams, 2017-08-17:
Like this: https://travis-ci.org/espruino/Espruino/jobs/265082014

Seems that `esptool` is being called and it's complaining about `serial`, but I don't think anything in the recent commits that'd cause it: https://travis-ci.org/espruino/Espruino/builds

It could be related to a new Travis environment?

wilberforce, 2017-08-17:
```
python /home/travis/build/espruino/Espruino/esp-idf/components/esptool_py/esptool/esptool.py \
	--chip esp32 \
	elf2image \
	--flash_mode "dio" \
	--flash_freq "40m" \
	-o espruino_1v93_esp32.bin \
	espruino_1v93_esp32.elf
Traceback (most recent call last):
  File "/home/travis/build/espruino/Espruino/esp-idf/components/esptool_py/esptool/esptool.py", line 25, in <module>
    import serial
ImportError: No module named serial
```

This *might* be related to the update to esp-idf 2.1.

Do we know if the travis environment has changed?

This looks like a similar issue:
https://github.com/travis-ci/travis-ci/issues/5522

gfwilliams, 2017-08-18:
Could be - that bug was ages ago though.

I think things have changed recently. The places where is stopped were:

Last working: https://travis-ci.org/espruino/Espruino/jobs/265080715
First failed: https://travis-ci.org/espruino/Espruino/jobs/265082014

There were no changes in the Espruino commits that would have affected ESP8266 - however `gcc --version` reports two different values. So I guess something has changed.

I wonder whether we could submit a PR to the esptool folks that only imported 'serial' if it was actually needed?

wilberforce, 2017-08-19:
Sorry - that pull request was not successful - there is no way of testing the Travis stuff without actually doing a pull request. (:;

I'm wondering now if this a python 2 vs 3 thing, as the python bosrd generation now seems to be failing before the elf file step as when you raised this?

wilberforce, 2017-08-19:
New issue now - getting this in the travis build:

```
/home/travis/build/espruino/Espruino/gen/jspininfo.c:6:37: error: 'JSH_ANALOG2' undeclared here (not in a function)
 /* PD0  */ { JSH_PORTD, JSH_PIN0+0, JSH_ANALOG2|JSH_ANALOG_CH1, { 0 } },
``

Looks like related to:
save a few bytes of flash on chips with <=1 ADC
https://github.com/espruino/Espruino/commit/b0a282dbb441bd086284c6d8c32cd03009c2aee9

ADC count must be wrong.... in the board py file

wilberforce, 2017-08-19:
https://github.com/espruino/Espruino/pull/1223

closed 2017-08-24 by wilberforce