espruino/Espruino #1221: Implement  Captive Portal to enter AccessPoint credentials

MaBecker, 2017-08-18:
Started to work on Captive Portal to enter SSID and Password to easily connect a Wifi enable Espruino device to a local AP. All this is based on UPD  dgram from @opichals .

[-] implement a wrapper module for all wifi devices 
[x] setup dns server to  handle IP4 request
[x]  with own ip
[x] setup http server
[x] redirect to own http server 
[-] present page to enter SSID and Password
[-] create a CaptivePortal JS module

First version of [CaptivePortal.js](https://gist.github.com/MaBecker/ae9dade26b44524e076ca19f5fd72fab.js) is still ESP specific because the first topic is missing.

@gfwilliams What are your thoughts  about such a wifi device wrapper?

**Any hints, comments or wishes are welcome.**

gfwilliams, 2017-08-18:
Ideally, I'd like to make `Wifi` a built-in object for all boards... Maybe tweaking one or two of the commands so it was possible to implement them on more platforms (some are sync, when other platforms require async).

There's an `at` branch where I was looking at this for ESP8266. To be honest I think that would be a better use of time?

opichals, 2017-08-18:
Just an idea here:

It would be cool if the wifi capable Espruino firmware 'shipped' with such a thing built-in and the only thing the user would have to do to start working with such a board would just be to plug it in a USB and connect to it using his dev machine/phone. It could give the user a choice to connect to his wifi router and provide the Espruino IDE links (or even redirects) allowing for a smooth onboarding.

Once router-connected it could advertise itself as an Espruino capable via mDNS (just like the esp8226 version does at the moment) and allow for local discoverability.

gfwilliams, 2017-08-18:
Yes, that's what I'd been thinking as well. I've been experimenting with compiling in JS code, and it works pretty well - so that could be a good option for this I think.

MaBecker, 2017-08-18:
Great, so this will be a cool firmware enhancement!

MaBecker, 2018-02-17:
I think a more useful variant of having a captive portal is to have a captiveForm

  * open a captive portal (dns and http server)
  * present given form or default form (sid &  pass)
  * collect postData
  * close captive portal (dns and http server)
  * return postData via callback   

```
 captiveForm.open({'callback':callback,'FormName':'captiveForm'});
```
This allows user to setup their on captive form or use the default form

gfwilliams, 2018-02-17:
Personally I think this is more of a getting started thing that we'd want pre-installed - so customisation wouldn't be a big deal as long as it was easy for people to change the code to roll their own?

Once configured, potentially we could also look at using this to serve up an iframe to espruino.com/ide with a websocket server for the IDE? It shouldn't actually require much code.

wilberforce, 2018-04-10:
@MaBecker 

I've extended your starting code:

https://gist.github.com/wilberforce/cc6025a535b8a4c7e2910d4ba7845f11

It scans for wifi ssid, adds that to the drop list, and allows you to enter a password and connect to the network.

[image]
[image]

Still very much proof of concept....

gfwilliams, 2018-04-10:
Very nice. If this went into flash (which it would) then it could hopefully be tweaked such that it didn't use *any* RAM once WiFi had been connected?

wilberforce, 2018-04-10:
Yes. I would like to use the Storage module to save a list of known access points and passwords.

Then on start up:

Test for conection. If ok done.
Try to connect to last connection. If ok done.
Scan the network, try to connect to the strongest with a known saved password... work the through the list.
If this fails then start as an AP and fire up the active portal... on connect, stop AP.

I might be over complicating things here!

So extending the concept of wifi.save to work with known existing networks, and only on failure do the captive portal.

MaBecker, 2018-04-10:
Wow - great work

MaBecker, 2020-07-10:
time to close it ;-)

closed 2020-07-10 by MaBecker