espruino/Espruino #1227: WebSockets on('open') event is not fired.

buglinjo, 2017-08-24:
I'm trying to catch web socket client connection event.

`var server = require('ws').createServer();`
`server.listen(81);`
`server.on("websocket", function(ws) {`
`    ws.on('open', function() {`
`      console.log("Connected to server");`
`    });`
`    ws.on('message', function(msg) {`
`      console.log(msg);`
`    });`
`    ws.on('close', function() {
`      console.log("Connection closed");`
`    });`
`});`

When I send message it prints in console what I send, but when I connect does not do anything. Also `ws.on('close')` event listener is working perfectly.

gfwilliams, 2017-08-29:
Hi - I'm pretty sure this is just a documentation issue.

You're creating a server, so when you get the `on("websocket"` event the connection is already open - hence there's no `open` event.

The `open` event is only for the websocket client - you've copy/pasted code from the client example in to the server example.

closed 2017-08-29 by gfwilliams

wilberforce, 2017-08-29:
@gfwilliams 

The confusion is here:

http://www.espruino.com/ws

Available callbacks
```
ws.on('open', function() {
  console.log("Connected to server");
});

ws.on('message', function(msg) {
  console.log("MSG: " + msg);
});
```
Needs a comment?

// open is Client only 

Same for ping and pong?

gfwilliams, 2017-08-29:
Great minds and all that: https://github.com/espruino/EspruinoDocs/commit/2dd802085ee5cd21b5d04fcf4426646f97e8f11d

I'd just forgotten to push my changes :)