espruino/Espruino #1228: ESP32: DigitalPulse causes crashes

chevdor, 2017-08-25:
DigitalPulse was initially not working for the ESP32. It has been fixed in #1175 
I am using the cuttingEdge from 2017-08-24 on a ESP32-WROOM32 Devkit.

Issue: Calling digitalPulse several times causes a crash.

The following sample does 'count' as expected, way above 200.
```
var count=0;

function blip() {
  //digitalPulse(18, 1, 15);
  console.log(count++);
}

function start() {
  setInterval(blip, 30);
}

E.on('init', start);
```

The following code however, never reaches 200, and crashes after an apparently random number of crashes. During the test, until it crashes, the pulses are correct:
[image]
 
```
var count=0;

function blip() {
  digitalPulse(18, 1, 15);
  console.log(count++);
}

function start() {
  setInterval(blip, 30);
}

E.on('init', start);
```

The crash occurs usually between count ~=30 to 120.
I have seen few examples where it runs without problem above 300 so you may have to run the  code a few times before catching the issue but the failure rate is high. If you wait 10s, you will see the issue a few times as shown below.

[image]

Here is a dump:
```
...
8
9
10
11
>abort() was called at PC 0x400827ef on core 0
Backtrace: 0x400876ac:0x3ffb0470 0x400877ab:0x3ffb0490 0x400827ef:0x3ffb04b0 0x400828dd:0x3ffb04e0 0x400d323a:0x3ffb0500 0x400829b8:0x3ffb0520 0x40115059:0x3ffb0540 0x40100e41:0x3ffb0560 0x400e68e0:0x3ffb0590 0x40082b1f:0x3ffb05e0 0x400816f5:0x3ffb0600
================= CORE DUMP START =================
nBwAAAkAAAB0AQAA
GP36P4DS/D/k0/w/
gNL8P4DT/D9aK2utlCf7P5Qn+z8Y/fo/jCf7PxkAAAAXNnv9ECYqCxj9+j8AAAAA
AAAAAOjP/D9JRExFAGxHvAFslchBuikAAAAAAOTT/D8AAAAAIQAGAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAFjj+j/A4/o/KOT6PwAAAAAAAAAA
AQAAAAAAAAB4KUA/AAAAAEgdAEAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAdE8
uBQIQPRYEkAwCAYAOFwIgEDT/D8IAAAAAQAAACMABgAY/fo/AQAAAAAAAADYM/s/
INP8PwAAAAAgCAYAIQAGALD6+j8kCPs/IAcGAAAAAACA0/w/AAAAAAAAAAAAAAAA
AAAAALUWCECw+vo/gGgIQAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAYNP8PwAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAACA0/w/AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAjNP8PwAAAAAAAAAAAAAAAAAAAAAAAAAA
[... 132 lines cut ...]
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA==
GOr6P4Do+j8M6vo/
gOj6P6Dp+j9FKTpXWLz9PyQn+z8Y6vo/HCf7PwEAAADo5fo/6OX6Pxjq+j/g5fo/
GAAAABDm+j9pcGMwAL4BrFyEbfYoYwAAAAAAAAzq+j8AAAAAIQAGABgAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAFjj+j/A4/o/KOT6PwAAAAAAAAAA
AQAAAAAAAAB4KUA/AAAAAEgdAEAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAu01
uBQIQMVPCEAwAAYAzxEIgEDp+j+85fo/AAAAAATm+j8AAAAAAQAAAAAAAACQUAiA
IOn6P9wA8D8BAAAAgCn7P2A7/j8MC/s/AgAAAAAAAAClpaWlpaWlpQAAAAAAAAAA
AAAAALUWCEBgO/4/gGgIQAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAgOn6PwAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAP////8AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAKDp+j8AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAABaDgiAEDz+Pwwn+z+c+/o/
AAAAAAAAAACs6fo/AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAA
================= CORE DUMP END =================
E (18359) esp_core_dump: Skipped 1 tasks with bad TCB!
E (18404) esp_core_dump: Crashed task has been skipped!
Rebooting...
```

jumjum123, 2017-08-25:
Hmm, checked and it crashes. Will try to find the reason.

wilberforce, 2017-08-28:
@jumjum123 
Im wondering if this due to interrupts calling a function that needs a prefix to keep it in memory?

jumjum123, 2017-08-28:
hmm, good point. I will check.
But this will take some time, just working on something else.

wilberforce, 2017-08-29:
Here is the core decoded:

```
 source scripts/provision.sh ESP32
export IDF_PATH=$ESP_IDF_PATH
ls../esp-idf/components/espcoredump/espcoredump.py  info_corefile -t b64 -c coredump.dat  espruino_1v93.227_esp32.elf
```

```
==================== ESP32 CORE DUMP START ====================

================== CURRENT THREAD REGISTERS ===================
pc             0x40127ea4       0x40127ea4 <esp_vApplicationIdleHook+40>
lbeg           0x0      0
lend           0x0      0
lcount         0x0      0
sar            0x0      0
ps             0x60820  395296
threadptr      <unavailable>
br             <unavailable>
scompare1      <unavailable>
acclo          <unavailable>
acchi          <unavailable>
m0             <unavailable>
m1             <unavailable>
m2             <unavailable>
m3             <unavailable>
expstate       <unavailable>
f64r_lo        <unavailable>
f64r_hi        <unavailable>
f64s           <unavailable>
fcr            <unavailable>
fsr            <unavailable>
a0             0x40085c38       1074289720
a1             0x3ffcd340       1073533760
a2             0x8      8
a3             0x1      1
a4             0x60023  393251
a5             0x3ffafd18       1073413400
a6             0x1      1
a7             0x0      0
a8             0x3ffb33d8       1073427416
a9             0x3ffcd320       1073533728
a10            0x0      0
a11            0x60820  395296
a12            0x60021  393249
a13            0x3ffafab0       1073412784
a14            0x3ffb0824       1073416228
a15            0x60720  395040

==================== CURRENT THREAD STACK =====================
#0  0x40127ea4 in esp_vApplicationIdleHook () at /mnt/c/Users/rhys/esp32/edge/EspruinoBuildTools/esp32/build/esp-idf/components/esp32/./freertos_hooks.c:52
#1  0x40085c38 in prvIdleTask (pvParameters=0x0) at /mnt/c/Users/rhys/esp32/edge/EspruinoBuildTools/esp32/build/esp-idf/components/freertos/./tasks.c:3322

======================== THREADS INFO =========================
  Id   Target Id         Frame
  9    process 8         xQueueGenericReceive (xQueue=0x3ffae5bc, pvBuffer=0x0, xTicksToWait=4294967295, xJustPeeking=0) at /mnt/c/Users/rhys/esp32/edge/EspruinoBuildTools/esp32/build/esp-idf/components/freertos/./queue.c:1452
  8    process 7         xQueueGenericReceive (xQueue=0x3ffd5150, pvBuffer=0x3ffd6750, xTicksToWait=4294967295, xJustPeeking=0) at /mnt/c/Users/rhys/esp32/edge/EspruinoBuildTools/esp32/build/esp-idf/components/freertos/./queue.c:1452
  7    process 6         xQueueGenericReceive (xQueue=0x3ffdbdc4, pvBuffer=0x3ffdcbd0, xTicksToWait=4294967295, xJustPeeking=0) at /mnt/c/Users/rhys/esp32/edge/EspruinoBuildTools/esp32/build/esp-idf/components/freertos/./queue.c:1452
  6    process 5         xQueueGenericReceive (xQueue=0x3ffd40a0, pvBuffer=0x3ffd4e40, xTicksToWait=4294967295, xJustPeeking=0) at /mnt/c/Users/rhys/esp32/edge/EspruinoBuildTools/esp32/build/esp-idf/components/freertos/./queue.c:1452
  5    process 4         xQueueGenericReceive (xQueue=0x3ffdb0b8, pvBuffer=0x3ffdbb20, xTicksToWait=4294967295, xJustPeeking=0) at /mnt/c/Users/rhys/esp32/edge/EspruinoBuildTools/esp32/build/esp-idf/components/freertos/./queue.c:1452
  4    process 3         prvTimerTask (pvParameters=0x0) at /mnt/c/Users/rhys/esp32/edge/EspruinoBuildTools/esp32/build/esp-idf/components/freertos/./timers.c:445
  3    process 2         xQueueGenericReceive (xQueue=0x3ffce3b0, pvBuffer=0x3ffcee40, xTicksToWait=85, xJustPeeking=0) at /mnt/c/Users/rhys/esp32/edge/EspruinoBuildTools/esp32/build/esp-idf/components/freertos/./queue.c:1452
  2    process 1         xQueueGenericReceive (xQueue=0x3ffdef74, pvBuffer=0x0, xTicksToWait=100, xJustPeeking=0) at /mnt/c/Users/rhys/esp32/edge/EspruinoBuildTools/esp32/build/esp-idf/components/freertos/./queue.c:1452
* 1    <main task>       0x40127ea4 in esp_vApplicationIdleHook () at /mnt/c/Users/rhys/esp32/edge/EspruinoBuildTools/esp32/build/esp-idf/components/esp32/./freertos_hooks.c:52

======================= ALL MEMORY REGIONS ========================
Name   Address   Size   Attrs
.rtc.text 0x400c0000 0x0 RW
.iram0.vectors 0x40080000 0x400 R XA
.iram0.text 0x40080400 0x14014 R XA
.dram0.data 0x3ffb0000 0x2614 RW A
.flash.rodata 0x3f400010 0x1e7e8 RW A
.flash.text 0x400d0018 0x95494 R XA
.coredump.tasks 0x3ffafd18 0x174 RW
.coredump.tasks 0x3ffcd280 0x164 RW
.coredump.tasks 0x3ffde724 0x174 RW
.coredump.tasks 0x3ffde4d0 0x248 RW
.coredump.tasks 0x3ffceee0 0x174 RW
.coredump.tasks 0x3ffcecf0 0x1e4 RW
.coredump.tasks 0x3ffcdbf8 0x174 RW
.coredump.tasks 0x3ffcda90 0x15c RW
.coredump.tasks 0x3ffdbbc8 0x174 RW
.coredump.tasks 0x3ffdba20 0x19c RW
.coredump.tasks 0x3ffd4efc 0x174 RW
.coredump.tasks 0x3ffd4d40 0x1b0 RW
.coredump.tasks 0x3ffdcc70 0x174 RW
.coredump.tasks 0x3ffdcad0 0x194 RW
.coredump.tasks 0x3ffd67ec 0x174 RW
.coredump.tasks 0x3ffd6640 0x1a0 RW
.coredump.tasks 0x3ffaea18 0x174 RW
.coredump.tasks 0x3ffae880 0x18c RW

===================== ESP32 CORE DUMP END =====================
```

wilberforce, 2017-08-29:
possibly related?

https://www.bountysource.com/issues/40870147-tw-12230-gettimeofday-not-reentrant-when-used-with-rtc-registers-only-waiti-called-and-triggers-watchdog

wilberforce, 2017-08-29:
Same abort() code dump as https://github.com/espruino/Espruino/issues/1176

Possibily related to the stack check watchdog code?

wilberforce, 2017-08-29:
@jumjum123 

Searching for 

esp_vApplicationIdleHook
In the esp32 forum
https://www.esp32.com/viewtopic.php?f=13&t=1566&p=7318&hilit=esp_vApplicationIdleHook#p7318

Last comment is:

==============
problem solved!
increased task stack size in xTaskCreate

thanks!!!

Danny
===========

So in this pull request when the timer was moved we need to increase the stack size from 20000 to st least 22000

https://github.com/espruino/Espruino/commit/1b3f15a1951f5bc3fcb59cfa768abdb8840abb36#diff-da4c9462e3dc0765f7a8d798ee5a44c8R83

wilberforce, 2017-12-14:
```
var count=0;

function blip() {
  digitalPulse(D21, 1, 15);
  console.log(count++);
}

function start() {
  setInterval(blip, 30);
}

start();
```

Fixed with https://github.com/espruino/Espruino/pull/1285

closed 2017-12-14 by wilberforce