espruino/Espruino #1230: Update formatFs docs

wilberforce, 2017-08-28:
Update this 

http://www.espruino.com/Reference#l_E_flashFatFS

Update docs to use try/catch as per http://www.espruino.com/ESP32

As now error is thrown on Unformatted filesystem.

wilberforce, 2018-04-09:
https://github.com/espruino/Espruino/blob/8aae321cd880fa98d862416915a6e1c9c9dea458/libs/filesystem/jswrap_file.c#L606

```
fs=require("fs");
if ( typeof(fs.readdirSync())==="undefined" ) {
  console.log("Formatting FS");
  E.flashFatFS({format:true});
}
```

To become

```
try {
  fs.readdirSync();
 } catch (e) { //'Uncaught Error: Unable to mount media : NO_FILESYSTEM'
  console.log('Formatting FS - only need to do once');
  E.flashFatFS({ format: true });
}
```

gfwilliams, 2018-04-09:
Do you want me to do that and close this issue then?

wilberforce, 2018-04-09:
Thanks! - If you can be bothered!

I documented what needs to be done on the ipad, as it's hard to edit code on that, let a alone check it compiles!

Otherwise I'll do it sometime.

commit d71c82c0 (2018-04-10): fix #1230
closed 2018-04-10 by gfwilliams