espruino/Espruino #1240: E.setBootCode not working for ESP8266_4MB

MaBecker, 2017-09-07:
looks like code cannot be found during dump() and ESP8266.reboot()

````
>E.setBootCode("console.log('Hi.....');");
Erasing Flash..................
Writing...
Compressed 25600 bytes to 32
Checking...
Done!
=undefined
>dump();
pinMode(D0, "input_pullup", true);
pinMode(D4, "input_pullup", true);
pinMode(D5, "input_pullup", true);
pinMode(D12, "input_pullup", true);
pinMode(D13, "input_pullup", true);
pinMode(D14, "input_pullup", true);
pinMode(D15, "input_pullup", true);
// Code saved with E.setBootCode
=undefined
>require('ESP8266').reboot()
=undefined
>
 ets Jan  8 2013,rst cause:2, boot mode:(3,6)
load 0x40100000, len 2592, room 16
tail 0
chksum 0xf3
load 0x3ffe8000, len 764, room 8
tail 4
chksum 0x92
load 0x3ffe82fc, len 676, room 4
tail 0
chksum 0x22
csum 0x22
2nd boot version : 1.7(5d6f877)
SPI Speed : 80MHz
SPI Mode : QIO
SPI Flash Size & Map: 32Mbit(1024KB+1024KB)
jump to run user1 @ 1000
âìnìròn|ìll`b|rlnànâl`ârllþLoading 32 bytes from flash...
> 

````
maybe this is not working for a Flash map 4MB:1024/1024?

found this: https://github.com/espruino/Espruino/blob/master/src/jswrap_flash.c#L568-L572

@tve  do you remember if this value only works for ES8266-01?

MaBecker, 2017-09-07:
looks fine in flash....

````
// Automatically generated header file for ESP8266_4MB
// Generated by scripts/build_platform_config.py
>FLASH_SAVED_CODE_START = 4108288;
=4108288
>FLASH_SAVED_CODE_LENGTH = 65536;
=65536
>FLASH_MAGIC_LOCATION = (FLASH_SAVED_CODE_START + FLASH_SAVED_CODE_LENGTH - 4);
=4173820
>FLASH_MAGIC = 0xDEADBEEF;
=3735928559

>E.setBootCode("console.log('Hi.....');");
Erasing Flash..................
Writing...
Compressed 25600 bytes to 32
Checking...
Done!
=undefined
>E.toString(f.read(4,FLASH_MAGIC_LOCATION));
="\xEF\xBE\xAD\xDE"
>E.toString(f.read(32,FLASH_SAVED_CODE_START));
="\x17\x00\x00\x00 \xB0>\x00console.log('Hi.....');\x00"
> 
````

gfwilliams, 2017-09-07:
What about `load()`? And is it still in flash after you reboot? Maybe the bootloader messes with it?

MaBecker, 2017-09-07:
results for load() ,reboot and flash read, so still in flash after reboots

````
>load();
=undefined
Loading 32 bytes from flash...
>dump()
pinMode(D0, "input_pullup", true);
pinMode(D4, "input_pullup", true);
pinMode(D5, "input_pullup", true);
pinMode(D12, "input_pullup", true);
pinMode(D13, "input_pullup", true);
pinMode(D14, "input_pullup", true);
pinMode(D15, "input_pullup", true);
// Code saved with E.setBootCode
=undefined

>require('ESP8266').reboot();
=undefined
>
 ets Jan  8 2013,rst cause:2, boot mode:(3,6)
load 0x40100000, len 2592, room 16
tail 0
chksum 0xf3
load 0x3ffe8000, len 764, room 8
tail 4
chksum 0x92
load 0x3ffe82fc, len 676, room 4
tail 0
chksum 0x22
csum 0x22
2nd boot version : 1.7(5d6f877)
SPI Speed : 80MHz
SPI Mode : QIO
SPI Flash Size & Map: 32Mbit(1024KB+1024KB)
jump to run user1 @ 1000
âìnìròn|ìll`b|rlnànâl`ârllþLoading 32 bytes from flash...
>FLASH_SAVED_CODE_START = 4108288;
=4108288
>FLASH_SAVED_CODE_LENGTH = 65536;
=65536
>FLASH_MAGIC_LOCATION = (FLASH_SAVED_CODE_START + FLASH_SAVED_CODE_LENGTH - 4);
=4173820
>FLASH_MAGIC = 0xDEADBEEF;
=3735928559
>FLASH_MAGIC_SIZE = 4;
=4
=undefined
>f = require("Flash");
=function () { [native code] }
>E.toString(f.read(FLASH_MAGIC_SIZE,FLASH_MAGIC_LOCATION));
="\xEF\xBE\xAD\xDE"
=undefined
>E.toString(f.read(32,FLASH_SAVED_CODE_START));
="\x17\x00\x00\x00 \xB0>\x00console.log('Hi.....');\x00"
> 
````

gfwilliams, 2017-09-07:
Wow, strange. I'd have said it might have been jsiInit() getting called with `false` (as happens normally when BTN1 is pressed at boot) but it's hard-coded to `true` on ESP8266. Only thing left is to add some prints I think.

MaBecker, 2017-09-07:
this output looks strange, side effect console.log is no longer defined....

````
>this
={
  "console": function () { [native code] },
  "JSON": function () { [native code] },
  "process": function () { [native code] }
 }
>
````

gfwilliams, 2017-09-07:
> this output looks strange, side effect console.log is no longer defined....

You get that if you save code with one version of Espruino and load it with another?

MaBecker, 2017-09-07:
Can not reproduce it, so lets forget that!

MaBecker, 2017-09-09:
This code for reading boot code will only work if the save area is in the first partition, because only this partition is mapped to memory starting at 0x40200000.

local commit: https://github.com/MaBecker/Espruino/commit/3145260e02c84d5b23aca6364fc57fb7e18ff45f

adding -mforce-l32 will dump boot code correct:

````
>dump()
......
// Code saved with E.setBootCode
console.log('Hi.....');
=undefined
> 
````
local commit: https://github.com/MaBecker/Espruino/commit/144c3c6a724dacf4a469249daef2f29c24de9f57

commit 0211008c (2017-09-15): ESP8266: Enable unaligned reads for ESP8266_4MB (fix #1240,#837)
commit 94031dbc (2017-09-15): ESP8266: move code save section to fist partition for memory mapping  (fix #1240)

only fist partition is memory mapped
commit f9079cec (2017-09-15): Merge pull request #1244 from MaBecker/master

ESP8266: Enable unaligned reads for ESP8266_4MB (fix #1240,#837)
closed 2017-09-15 by gfwilliams