espruino/Espruino #1241: ESP8266: Remove osprintf for RELEASE builds

gfwilliams, 2017-09-12:
* With osprintf: 465764
* Without: 461556

By ditching it on the 512k builds you'd free up a lot more flash, which would make it sensible to include crypto while also having the 32 bit loads.

Tested by replacing the define in `osapi.h` with:

```
#define os_printf(...) /* removed */
```

But we could presumably come up with a better way.

@MaBecker? I don't think anyone actually uses the debugging prints any more do they? And if they do, they can probably compile their own firmware with them in.

MaBecker, 2017-09-12:
Good point, so let me continue with this and come back with the new size

MaBecker, 2017-09-12:
start with this:

````
#ifdef RELEASE
#define DBG(format, ...) os_printf(format, ## __VA_ARGS__)
#else
#define DBG(format, ...) do { } while(0)
#endif
````

and change 

````
os_printf();
//to 
DBG();
````

MaBecker, 2017-09-12:
No, lets comment them

gfwilliams, 2017-09-13:
Sure? 

Another option is that if we always include `osapi.h` via another include file, we can just `#undef` and redefine `os_printf.

MaBecker, 2017-09-13:
Wow, yes this is a much easier way,  13 files with  a single line to change

MaBecker, 2017-09-13:
osapi_release.h

````
#include <osapi.h>

#ifdef RELEASE
#undef os_printf
#define os_printf(format, ...) do { } while(0)
#endif
````

gfwilliams, 2017-09-13:
Looks great :)

MaBecker, 2017-09-13:
Starting with 

    ** user1.bin uses  459732 bytes of 479232 available

using osapi_release.h frees 3696 bytes

    ** user1.bin uses  456036 bytes of 479232 available

plus -mforce-l32 adds 33280 bytes 

    ** user1.bin uses  489316 bytes of 479232 available

So this a big win for a board with 512k but not enough to use -mforce-l32. 

Lets try this for branch RODATA

gfwilliams, 2017-09-13:
Wow, that's odd. I think I may have hacked something into the RODATA branch by changing `osapi.h` because somehow I've managed to get it building with Crypto.

Interesting the SHA1 function is huge - it seems to be one of the biggest crypto functions (perhaps not including lookup tables) but it feels like there's so much repetition (but just done with macros) there must be a less resource-intensive version available.

MaBecker, 2017-09-13:
No, branch RODATA with osapi_release.h and CRYPTO is to big too

    ** user1.bin uses  485828 bytes of 479232 available

MaBecker, 2017-09-13:
So lets put osapi_release.h in folder targets/esp8266/

gfwilliams, 2017-09-13:
Yes, sounds good - thanks!

Very strange about RODATA though. I just compiled with `RELEASE=1` and got `** user1.bin uses  478788 bytes of 479232 available` - with crypto

MaBecker, 2017-09-13:
this is the best I can get:

    ** user1.bin uses  481604 bytes of 479232 available

which branch are you testing?

gfwilliams, 2017-09-13:
```
git checkout ESP8266_RODATA
make clean;BOARD=ESP8266_BOARD RELEASE=1 V=1 make
```

I'm not sure why I need `V=1` (verbose) at the moment but for some reason just today it's complaining if I don't.

MaBecker, 2017-09-13:
which version of SDK are you using?

MaBecker, 2017-09-13:
I guess crypto is excluded on your side

gfwilliams, 2017-09-13:
`v2.0.0.p1`, and `xtensa-lx106-elf-gcc (crosstool-NG 1.20.0) 4.8.2`.

Travis seems to get the same result as you - worse actually: https://travis-ci.org/espruino/Espruino/jobs/274953710

So I don't know why mine's different/

gfwilliams, 2017-09-13:
Nope, it's in there. Without it I get `461700`

gfwilliams, 2017-09-15:
Fixed with https://github.com/espruino/Espruino/pull/1248

closed 2017-09-15 by gfwilliams