espruino/Espruino #1242: git and branches - how to contribute and stay up to date

MaBecker, 2017-09-14:
There are two things I like:

- stay up to date 

- work parallel on bug fixes or enhancements for ESP8266

If creating a pull request it contains all my changes plus the merges which is confusing, so this ends in something like git cherry picking. 

Question: How to handle this?

gfwilliams, 2017-09-14:
I think this is a pretty normal Git thing - you might find some advice online.

But what I do is:

* `git checkout branch_i_want`
* `git checkout -b my_branch`
* make my changes

Then when something changes:

* `git pull`
* `git merge branch_i_want`

That does create a bunch of merge commits, but it's just normal Git and nothing to worry about. When it comes to create a PR from that branch all I really end up seeing is your differences, not any of the merges.

What is does mean is that if there was a merge conflict and something wasn't merged right, I can track it down to the merge and see what happened - which I guess is useful.

gfwilliams, 2017-09-14:
Also, just to add that having a separate branch means that when you issue a PR, you can then keep working on another branch based off your first branch - because if you keep doing stuff on the original branch it'll keep updating the PR.

opichals, 2017-09-14:
I do a lot of rebasing of the branch I work in as follows:

    git pull --rebase origin/master

That way I always end up _on top of_ what is in the remote branch. It requires doing force push but as long as it is in my private branch not shared with anyone else it is fine and it keeps the history clean & tidy.

gfwilliams, 2017-09-14:
^^^ nice. I'll have to try that :)

MaBecker, 2017-09-14:
Great - Thanks!

So I just have to clean up my mess and start working with branches.

MaBecker, 2017-09-14:
OK, created a branch ESP8266_ADD_GPIO16 reset and added changes.

Now pushed the changes created a pull request  https://github.com/MaBecker/Espruino/pull/1

How to request this to be merged into espruino/Espruino master?

opichals, 2017-09-14:
You need to create the PR against espruino/Espruino:master repo (see the repo/branch dropdowns in the PR creation)

MaBecker, 2017-09-14:
Yep, but this will include all other changes from the master too. This what I want to avoid.

opichals, 2017-09-14:
You either need the master [merged into your branch or your changes rebased](https://www.atlassian.com/git/tutorials/merging-vs-rebasing) on top of master before creating the PR.

MaBecker, 2017-09-14:
OK, got it  - Thanks.

Is it necessary  to  keep  a brach after pull request is merged into master?

opichals, 2017-09-14:
I usually keep them and clean them in bulk from time to time. But I guess there is no real reason to.

MaBecker, 2017-09-15:
Interesting fact for pull request titles:

Adding  `fix #<issue number>` to title  will close  issue when merged

gfwilliams, 2017-11-16:
@MaBecker  can we close this one now?

closed 2017-12-15 by gfwilliams

MaBecker, 2017-12-15:
Sorry - yes of course