espruino/Espruino #1249: CoAP - Constrained Application Protocol over UDP

MaBecker, 2017-09-21:
As UDP is available we can plan to implement [CoAP](https://en.wikipedia.org/wiki/Constrained_Application_Protocol) as another service layer for M2M communication.

Got touched by CoAP as watching video number one of Andreas Spiess video [144](https://www.youtube.com/watch?v=pfG8uEDZj5g)

First question:     

    Implementation in C or JS,  for both lots of sample are available on github

Any comment, suggestion or ideas are helpfull

gfwilliams, 2017-10-02:
Personally I'd say use JS. It'll be a lot faster to iterate/develop, and we've already used it for things like MQTT and Websockets - I'm not sure there are huge reasons why CoAP should be significantly different?

jaimejim, 2017-10-23:
Hi, I'd be interested on having CoAP on Espruino too. Has anyone moved forward on this already?
There are multiple CoAP implementations available, JS too.

http://coap.technology/impls.html

bartmichu, 2017-10-23:
There is a lot of talk about CoAP in context of Ikea Tradfri system. How cool would that be if we could talk with these devices via Espruino? One thing to keep in mind is that it would also require DTLS implementation.

MaBecker, 2019-11-12:
there is no interest - reopen it needed

closed 2019-11-12 by MaBecker