espruino/Espruino #1257: RegEx backreferences

gfwilliams, 2017-10-11:
With `\1`/etc in the search string

http://forum.espruino.com/conversations/311743/#comment13901937

gfwilliams, 2020-10-29:
We now error if backreferences are used

gfwilliams, 2020-10-29:
- [ ] `x{3}` numeric quantifier

gfwilliams, 2020-10-29:
- [ ] `\B` Assertion Matches a non-word boundary

gfwilliams, 2020-10-29:
- [ ] `x(?=y)` Lookahead assertion / `x(?!y)` Negative lookahead assertion

tanepiper, 2021-02-23:
@gfwilliams Here is a nice use case I can't use yet:

```
function splitString(string, size, multiline) {
    const matchAllToken = multiline ? '[^]' : '.';
    const re = new RegExp(matchAllToken + '{1,' + size + '}(?=\\s|$)', 'gm');
    return string.match(re);
}
```

This would take a line of text and split it into lines up to the passed size, up to the next space or end of the line:

[image]

Makes text formatting very easy :D

gfwilliams, 2021-02-24:
I'm guessing that uses numeric quantifiers and lookahead assertions? Honestly, I'm not sure if we'll get to the stage when enough is implemented to make that work soon (unless someone feels like making a PR!). Line splitting is something that probably makes sense to add to Espruino at some point though (ideally it could take char width into account).

MaBecker, 2021-02-24:
This is a snippet used in node js. Is this simpler to implement?

```js
text = "Mary had a little lamb. His fleece was black as coal, yeah. Everywhere the child went. That little lamb was sure to go now.";
s = 20;
const wrap = (s, w) => s.replace(
    new RegExp(`(?![^\\n]{1,${w}}$)([^\\n]{1,${w}})\\s`, 'g'), '$1\n'
);

console.log(wrap(text,20));
/* output when running in Emulator 
Mary had a little lamb. His fleece was black as coal, yeah. Everywhere the child went. That little lamb was sure to go now.
*/
/* output when running in node
Mary had a little
lamb. His fleece was
black as coal, yeah.
Everywhere the child
went. That little
lamb was sure to go
now.
*/
```

gfwilliams, 2021-02-24:
It still uses numeric quantifiers with min/max in them. Honestly, I'm not sure string splitting is a super useful thing to do with regex - because I reckon in a lot of cases you may want to use non-fixed-width strings (so splitting just on char count isn't what you want?).

MaBecker, 2021-02-24:
So what about a wrap function that is splitting at spaces and then looping to adding words up to a specific length into lines or a new string with '\n'?

gfwilliams, 2021-02-24:
Yes, that's what I had in mind. The word splitting functions exist in Espruino right now: https://github.com/espruino/BangleApps/blob/master/apps/notify/notify.js#L12

So all it'd need is for that code to be pulled out into a module.

MaBecker, 2021-02-24:
just added a [pr](https://github.com/espruino/EspruinoDocs/pull/594/commits/9b236d8965cfe677cdaad96a0479444995d85c98) for a new module to be the home for more of those string decoding tasks.

MaBecker, 2021-02-24:
@tanepiper  please check the new module name [StringDecoders](https://github.com/espruino/EspruinoDocs/blob/master/modules/StringDecoders.js)

tanepiper, 2021-02-27:
@MaBecker Looks like a good start (although it's not quite what I'm looking for since it's still treating the input and output as a string)

esdeboer, 2024-06-19:
@gfwilliams could the missing RegExp features missing be added to the reference documentation? http://www.espruino.com/Reference#RegExp (Or a link to this issue)

I was quite surprised that my simple regex wasn't working and it took me a while to find on the Espruino feature list that the numeric quantifier {} isn't supported.

esdeboer, 2024-06-20:
Copied the notes from the Feature List to the Class documentation: https://github.com/espruino/Espruino/pull/2518

I'll take a look if I can add the numeric quantifiers

atjn, 2024-08-16:
- [ ] `x?` zero or one of `x`
 
 I needed this for a project. I was able to substitute it with `x*` which is not ideal, but probably doesn't make a difference in practice for my use case.