espruino/Espruino #1261: esp8266 does not when disconnected from IDE

aminjoharinia, 2017-10-22:
Hi, i wrote this code and it is perfectly run on esp8266 ,but when i disconnect it from IDE , IT does not work at all , it evrn not connected to wifi
Am i doing sth wrong?
Help me pls
```javascript
var wifi = require("Wifi");
wifi.setHostname("SAJWWOODY");
wifi.connect("Ultra Speed", {password:"sajwoodind"}, function(err){
 console.log("connected? err=", err, "info=", wifi.getIP());
});
wifi.stopAP();
wifi.save();

var led = new Pin(NodeMCU.D7);

var http = require("http");
var server = http.createServer(function(request,response){
  if(request.url === '/on')
  {
    response.writeHead(200,{'Content-Type':'text/plain'});
    response.end("The Lamp Turned On !");
    digitalWrite(led, 1);
  }
  if(request.url === '/off')
  {
    response.writeHead(200,{'Content-Type':'text/plain'});
    response.end("The Lamp Turned OFF !");
    digitalWrite(led, 0);
  }
  if(request.url === '/'){
    response.writeHead(200,{'Content-Type':'text/plain'});
    response.end("Welcome To SAJ WOOD IND");
  }
});

server.listen(80);
```

gfwilliams, 2017-10-23:
Are you disconnecting it from power or resetting it between? If you haven't explicitly saved your code then that could be the problem?

aminjoharinia, 2017-10-24:
@gfwilliams 
i press "send to espruino" and then it works
but if i press reset button or unplug it from power , then plug it in ,  it does not work at all
like there is no program uploaded
even after i connected to ide, it does not work
i should  send my code to espruino  again to work
Is it because of firmware? i use version 1v94

gfwilliams, 2017-10-25:
Just type `save()` on the left hand side, or choose `save on send` when uploading. If you have any other questions please ask on the forum - that's what it's for: http://forum.espruino.com/

It'd also be worth looking at the Espruino Quick start and FAQ - they'd probably answer a bunch of your other questions.

closed 2017-10-25 by gfwilliams

aminjoharinia, 2017-10-25:
when i type ```save()``` or enable ```save on send```
i got this error
[image]

PrimeTime416, 2017-10-25:
I guest you put the save() on the right hand side of the webIDE, also make
sure you only have one save not multiple.

Anthony G. Kerr

aminjoharinia, 2017-10-25:
i got this error even my code is 
```javascript
console.log("hi"):
```