espruino/Espruino #1263: Problem when creating a new String child

faghazi, 2017-10-23:
When using the jsvSetNamedChild from jsvar.c with a String child, the end result is not a JsVar.
Here is the exemple I used :
-----------------------------------------------------------------------------------------------
JsVar* myCfg = jsvNewObject();
jsvCreateNewChild(myCfg, jsvNewFromInteger(1), jsvNewFromString("region"));
JsvSetNamedChild(myCfg, jsvNewFromString("NA"),"region") ;
-----------------------------------------------------------------------------------------------

And Here are the expected and the actual result I got :
-----------------------------------------------------------------------------------------------
Expected : {"region": "NA"}
Actual :  }"region": "NA"
-----------------------------------------------------------------------------------------------

This behaviour is also seen when creating Child Array:
-----------------------------------------------------------------------------------------------
Expected :  {"chanList": [ 1, 2, 3 ]}
Actual : }"chanList": [ 1, 2, 3 ]
-----------------------------------------------------------------------------------------------

A possible origin of this bug may be some of the special symbols in a string or an Array ( ", \0, etc)

gfwilliams, 2017-10-23:
I'm afraid I don't understand - how are you outputting the object such that you get `}"region": "NA"` displayed? You could also use `jsvTrace` to get more details about the internal structure of the object.

In your code:
* You're creating a numeric child named `1` with value `region`
* Then creating a new child called `region` that has value `NA`
* you're not unlocking any of the unused memory, so every time it runs, memory will be irrevocably lost.

Personally I'd just do:

```
JsVar* myCfg = jsvNewObject();
jsvObjectSetChildAndUnLock(myCfg, "region", jsvNewFromString("NA"));
```

gfwilliams, 2017-10-23:
Just tested the code you posted:

```
  JsVar* myCfg = jsvNewObject();
  jsvCreateNewChild(myCfg, jsvNewFromInteger(1), jsvNewFromString("region"));
  jsvSetNamedChild(myCfg, jsvNewFromString("NA"),"region") ;
```

and it seems to work fine (apart from the memory leaks). For whatever reason, `jsvCreateNewChild(myCfg, jsvNewFromInteger(1), jsvNewFromString("region"));` doesn't seem to do anything.

I'd look at other bits of code in Espruino to see how this is done though.

faghazi, 2017-10-23:
I get the output result by 2 different way : 
- I either simply put a return myCfg. (which return the shown value)
- Or get the return value and inject it in a function that takes a JsVar* as an argument. (which returns an error cause the input is not a JsVar).

It is important to precise that i used "jsvSetNamedChild" and also what you have just suggested ("jsvObjectSetChildAndUnLock") by replacing jsvNewFromString("NA") by jsvNewFromInteger(1) and the result is perfectly fine.
The problem on my end seems to occur only with Strings and arrays.

But since everything works fine on your end, I may be missing some unknown behaviour of my embedded chip.

gfwilliams, 2017-10-23:
So `jsvObjectSetChildAndUnLock(myCfg, "region", jsvNewFromString("NA"));` fails, but `jsvObjectSetChildAndUnLock(myCfg, "region", jsvNewFromInteger(1));` works?

Which chip are you using then? It's possible that you have the number of JsVars so high there is no room for stack, and the stack is overwriting the JS variables.

I'd try compiling locally on Linux with `DEBUG=1 make` and see if you have problems first - that's what I try and do as much as possible as developing/debugging on Linux is much easier.

faghazi, 2017-10-23:
To answer your question : yes, exactly.
The chip i am using is the STM32L476
Unfortunately, it is a keil project so it's windows only. However i'm going to compile Espruino on my linux and add a simple example just to verify wether or not it is a stack problem.
I will post the results as soon as i get it done.

gfwilliams, 2017-10-23:
Ahh, it's unlikely to be a chip issue with an ARM in there, but it *could* be a Keil issue?

Also, make sure that in your `BOARD.py` file you lower the `variables` count significantly - as you may just have too many variables allocated.

faghazi, 2017-10-23:
I just tested it under my linux mavhine and everything is indeed working fine.
So it it most definetly a full stack issue.
Thank you for your support.
Issue closed.

closed 2017-10-23 by faghazi