espruino/Espruino #1271: dgram.createSocket('udp4').send() signature to support offset and length

opichals, 2017-11-05:
As there is bunch of code probably going to be ported from node.js it would be very nice to conform to its complete [send() signature](https://nodejs.org/api/dgram.html#dgram_socket_send_msg_offset_length_port_address_callback).

```js
socket.send(msg, [offset, length,] port [, address] [, callback])
```

I would have created a PR but I struggle to find a way to express all this in the C comments-based binding language properly (the /**JSON... */).

opichals, 2017-12-14:
With [these changes](https://github.com/espruino/Espruino/compare/master...opichals:dgramsocket-5args-send?expand=1) it appears that the 5th (address) argument is not passed in:

```sh
$ ./espruino  ./tests/test_dgram_multi.js 

41234,"localhost", 41234 "localhost"
3,2, 51234 Segmentation fault: 11
```

Perhaps due to the amount of argument info that can fit into JsnArgumentType's size as noted at https://github.com/espruino/Espruino/blob/master/src/jsnative.c#L106?

Any hints on how to approach this?

gfwilliams, 2017-12-14:
Sure - just put one argument as `JsVarArray` (grep the source to see how it's done). You can then access that as a standard array, and can have however many args you want.

To be honest the build *should* really fail if something like that happens. I'll see what I can do about that.

opichals, 2017-12-15:
@gfwilliams Thank you, PR submitted.

The callback argument support would need a bit more work there so I left it unsupported for now.