espruino/Espruino #1273: Bluetooth LE speed negotiation

gfwilliams, 2017-11-06:
We should look into whether we can lower the speed of polling when in a connection dynamically - it'd lower battery usage loads, and then we could go up to maximum when there was a lot of data to transfer

gfwilliams, 2017-12-18:
Seems to be doable with https://infocenter.nordicsemi.com/index.jsp?topic=%2Fcom.nordic.infocenter.sdk52.v0.9.1%2Fgroup__ble__sdk__lib__conn__params.html, and the library is already used in Espruino.

It should be as easy as using `ble_conn_params_change_conn_params`, so then we just need a function that lets people call it... Perhaps with an `"auto"`option where the connection interval is dropped if there are no tx/rx/writes/reads in 60 seconds.

wilberforce, 2018-12-20:
@gfwilliams 
Haven't you implemented this already?

gfwilliams, 2018-12-20:
Yes, thanks!

gfwilliams, 2018-12-20:
It can be tweaked with `NRF.setConnectionInterval(...)`

closed 2018-12-20 by gfwilliams