espruino/Espruino #1275: socketSendData() cutting UDP messages

opichals, 2017-11-09:
The `socketSendData()` allocates only the `net->chunkSize` and relies on the fact that additional data will be send during a another call (TCP). That is however not true for UDP where the message is cut (only `net->chunkSize` bytes is sent).

gfwilliams, 2017-11-10:
Ahh - interesting - thanks! I saw someone complaining about a ~500 byte limit for UDP send - I guess this is it?

Potentially you could use aalloc to allocate a larger buffer on the stack if it was needed (and there was space!). I could be wrong but I think all that socket processing happens on idle, which means that actually you do have a whole lot of stack available. There might be a few other areas where we could make use of that.

opichals, 2017-11-10:
The limit is around 260 chars.

I propose to allocate whatever the message size is on the stack with board specific max message limit to compensate for stack size limits. Perhaps we should add a net->maxUdpPacketSize so that any specific network implementation can have its own?

gfwilliams, 2017-11-14:
That sounds sensible... Actually maybe don't bother with a board-specific size just yet.

In a few places I do stuff like this:

```
if ((unsigned int)nBytes+256 > jsuGetFreeStack()) {
    jsExceptionHere(JSET_ERROR, "Not enough free stack to receive this amount of data");
    return 0;
  }
  unsigned char *buf = (unsigned char *)alloca((size_t)nBytes);
```

Maybe you could just do the same, so it just allocates the space it needs as long as there's enough space for the data and the `send` call to complete?

opichals, 2017-11-14:
This sounds good enough for now. Thanks for the hint!

BTW: Where does the 256 (in the nBytes+256) come from? Empirical value?

gfwilliams, 2017-11-15:
Yes, totally :) I would be tempted to leave a little more (1024?) In your case, since that send call may go into JS on some devices (when using GSM/WiFi via serial)

vaxus1, 2017-11-15:
Hello.
I m the one who reported the problem at first place, thanks to Opichals to take over.
My Amazon Alexa integration is on hold because of this problem :(

Any solution already? So I can reload Espruino night run and try it again.

Reading here I see there are standard methods that if implemented would help error handling.
getSendBufferSize()
setSendBufferSize()

source from: http://download.java.net/jdk7/archive/b123/docs/api/java/net/DatagramSocket.html

Let me know please.... v.denegin at gmail dot com

PR #1278 (2017-11-17): socketSendData() to use dataLen buffer for UDP
Fixes #1275

Note that this only changes the `dgramSocket.send()` method buffer handling where the available stack-1024 bytes is the limit for the data size.

The receiving side is still hardcoded board/net implementation specific (the `net->chunkSize` sized buffer is used).
closed 2017-11-17 by gfwilliams