espruino/Espruino #1290: digitalPulse timings are wrong on the ESP8266

calebbrewer, 2017-12-18:
Firmware Version: 1v95.4

I am trying to use digitalPulse to pulse an IR LED...

```
var buttonPulses = [9.05699999998,4.44600000014,0.59899999996,0.53699999989,0.59499999997,0.51100000018,0.62200000002,0.51199999984,0.59499999997,0.53600000001,0.59599999985,0.51100000018,0.62200000002,0.51,0.59499999997,0.53699999989,0.59600000008,0.51099999973,0.62200000024,1.63799999995,0.59699999997,1.63900000006,0.59600000008,1.63899999984,0.62299999990,1.63900000006,0.59800000008,1.63799999995,0.59600000008,1.64,0.62099999991,1.64,0.59600000008,1.63800000018,0.59799999985,1.63799999995,0.62300000013,1.63699999984,0.59799999985,1.64000000017,0.59599999985,1.64000000017,0.62200000002,0.51099999996,0.59600000008,0.53599999978,0.59600000031,0.50999999984,0.62299999990,0.51,0.59499999997,0.538,0.59600000008,0.51,0.62199999979,0.51099999996,0.59499999997,0.53700000012,0.59600000008,1.63799999972,0.59700000019,1.63899999984,0.62300000013,1.63799999995,0.59699999997,1.64000000017,0.59499999997,39.73900000005,9.04999999988,2.21499999997,0.59800000008];

digitalPulse(5, 1, buttonPulses);
```

When I do this and capture it with an IR receiver I get something like this...

```
[0.21500000001,1.14,0.48,6.54700000001,0.49399999998,3.51700000001,0.178,5.27,0.536,2.786,0.266,1.33200000001,0.211,9.974,0.49399999998,901.176,0.216]
```
The timings digitalPulse produces vary by a few millseconds sometimes. Is it just not able to be that exact on the 8266?

gfwilliams, 2017-12-18:
I believe what you're doing won't work because you need to send series of 40kHz pulses to most IR transmitters, not just a simple long square wave. There's info for the Espruino Pico here: http://www.espruino.com/Pico+Infrared

In that case we output a 40kHz square wave on one pin, and the pulses on the other. But you *can't do that on ESP8266* because it doesn't have hardware PWM.

Realistically I think you need to connect the ESP8266 directly to something that can measure the pulse lengths - but I think you'll find that if you do it's fine.

calebbrewer, 2017-12-19:
Thanks for the response. Ahh so that means that the IR receiver is doing the decoding. I didn't know that at first.

Is there a reason I can't use soft PWM on the 8266 to do this?

gfwilliams, 2017-12-19:
Ok, closing this then.

Soft PWM is unlikely to do up to 40kHz reliably.

What you could do is come up with a Float32Array of pulse timings, and actually issue explicit pulses at 40kHz during the periods that the IR is meant to be on.

... of course the easy solution is to support Espruino and use an official board ;)

closed 2017-12-19 by gfwilliams

calebbrewer, 2017-12-19:
Thank you! I started down that train of thought last night. Oh and I think I will be supporting the project. This is my first time playing with Espruino. Thank you so much for the work you have done.