espruino/Espruino #1294: unstable version without build option RELEASE

MaBecker, 2018-01-02:
building Espruino for BOARD=ESP8266_4MB without RELEASE=1 causes a endless loop after reboot

```
>require('ESP8266').reboot();
=undefined
>
 ets Jan  8 2013,rst cause:2, boot mode:(3,6)
load 0x40100000, len 2408, room 16
tail 8
chksum 0xe5
load 0x3ffe8000, len 776, room 0
tail 8
chksum 0x84
load 0x3ffe8310, len 632, room 0
tail 8
chksum 0xd8
csum 0xd8
2nd boot version : 1.6
  SPI Speed      : 80MHz
  SPI Mode       : QIO
  SPI Flash Size & Map: 32Mbit(1024KB+1024KB)
jump to run user1 @ 1000
Fatal exception (3):
epc1=0x401000b8, epc2=0x00000000, epc3=0x00000000, excvaddr=0x401000df, depc=0x00000000
Fatal exception (3):
epc1=0x401000b8, epc2=0x00000000, epc3=0x00000000, excvaddr=0x401000df, depc=0x00000000
```

Stange, a build with `BOARD=ESP8266_4MB  RELEASE=1 make` has no problems with reboot on the same device ....

MaBecker, 2018-01-03:
http://0x04.net/~mwk/doc/xtensa.pdf Table 4–64. Exception Causes

````
3 LoadStoreErrorCause   Processor internal physical address or data error      YES
                        during load or store [Exception Option]
````

Check with esptool.py  verify_flash .....

````
Verifying 0xf10 (3856) bytes @ 0x00000000 in flash against /esptools/esp_iot_sdk/sdk_20/bin/boot_v1.6.bin...
-- verify OK (digest matched)
Verifying 0x834a4 (537764) bytes @ 0x00001000 in flash against espruino_esp8266_user1.bin...
-- verify FAILED (digest mismatch)
Verifying 0x80 (128) bytes @ 0x003fc000 in flash against /esptools/esp_iot_sdk/sdk_20/bin/esp_init_data_default.bin...
-- verify OK (digest matched)
Verifying 0x1000 (4096) bytes @ 0x003fe000 in flash against /esptools/esp_iot_sdk/sdk_20/bin/blank.bin...
-- verify OK (digest matched)
Verifying 0x834a4 (537764) bytes @ 0x00101000 in flash against espruino_esp8266_user1.bin...
-- verify OK (digest matched)
````

MaBecker, 2018-01-03:
Looks like a flash speed problem

MaBecker, 2018-01-03:
Tried some different builds based on Espruino master 1v95

At the end it seems to be a problem of bin file  size - as far as I understand.

**Any advice /suggestion?**  

Because during development and testing  it is very helpful to have  a stable version with os_prints (require('ESP8266').setLog(2); )

crashing build

    make flash_erase && make clean && ESP8266_4MB=1 make flash
    user1.bin uses  537364 bytes of 962560 available

working builds

#default 

      make flash_erase && make clean && ESP8266_4MB=1 RELEASE=1 make flash
      user1.bin uses  509204 bytes of 962560 available

#build without 'GRAPHICS'

    make flash_erase && make clean && ESP8266_4MB=1 make flash
    user1.bin uses  513380 bytes of 962560 available

#build without CFLAGS += -mforce-l32

    make flash_erase && make clean && ESP8266_4MB=1 make flash
    user1.bin uses  501092 bytes of 962560 available

MaBecker, 2018-01-05:
full output of build for that crashing bin file: 

    make clean && ESP8266_4MB=1 make

    Sections:
    Idx Name          Size      VMA       LMA       File off  Algn
      3 .text         00007cdc  40100000  40100000  00003230  2**2
                      CONTENTS, ALLOC, LOAD, READONLY, CODE
      4 .irom0.text   000784a4  40201010  40201010  0000af10  2**4
                      CONTENTS, ALLOC, LOAD, READONLY, CODE
    To disassemble: xtensa-lx106-elf-objdump -d -l -x espruino_esp8266_user1.elf
    xtensa-lx106-elf-objdump -d -l -x espruino_esp8266_user1.elf >espruino_esp8266_user1.lst
      8 -rwxr-xr-x  1 markbecker  staff    2220  5 Jan 11:32 eagle.app.v6.data.bin
    968 -rwxr-xr-x  1 markbecker  staff  492708  5 Jan 11:32 eagle.app.v6.irom0text.bin 
     24 -rwxr-xr-x  1 markbecker  staff   10400  5 Jan 11:32 eagle.app.v6.rodata.bin
     64 -rwxr-xr-x  1 markbecker  staff   31964  5 Jan 11:32 eagle.app.v6.text.bin
    ** user1.bin uses  537364 bytes of 962560 available

So this is a ticking time bomb for the RELEASE=1 build.

MaBecker, 2018-01-05:
Hi @gfwilliams any idea from your side?

MaBecker, 2018-01-05:
>  just disable libraries I'm not using in the Python board file

That is my  workaround as well.

Using flash map 6 (4MB:1024/1024)  allows to use larger files to flash [ESP_FLASH_MAX=962560 ](https://github.com/espruino/Espruino/blob/master/boards/ESP8266_4MB.py#L36)

[image]

so check.sh is not complaining 

`PASS - size of 537828 is under 962560 bytes`

So there must be something that is missing....

netico14, 2018-01-07:
hello, I have a problem with my esp-01 or esp8266-01, sorry write here but i dont understand github, I was programming with arduino ide and it worked fine, but I could not receive a data from socket.io so I liked espruino but I can not connect, does not recognize my esp-01, I have already tried with another one that I have but it does not work, delete the firmware, update it with the files of the updated espruino page, probe with the ones that you uploaded some time ago but nothing, I made the update by esptool and by nodemcu firmware programmer but nothing works for me, I ask for your help, thank you very much

MaBecker, 2018-01-07:
Hi @netico14,

please post you ESP - Espruino problem [here](http://forum.espruino.com/microcosms/925/).

wilberforce, 2018-02-22:
@MaBecker 
Isn't new this related to the flashing and needing the  `flash_size 4MB-c1` To select the 1024/1024 map?

MaBecker, 2018-02-22:
When   `ESP8266_4MB=1` is set than `make flash` is using `flash_size 4MB-c1`

The question is : Why can't a build use more than 512 KB whe using `4MB-c1`like shown in picture  above?

wilberforce, 2018-02-22:
Perhaps it is the linking script - the .ld file -  the existing might have a 512K limitation.

MaBecker, 2018-02-22:
https://github.com/espruino/Espruino/blob/master/targets/esp8266/eagle.app.v6.new.2048.ld#L3-L9

MEMORY
{
  dport0_0_seg :                        org = 0x3FF00000, len = 0x10
  dram0_0_seg :                         org = 0x3FFE8000, len = 0x14000
  iram1_0_seg :                         org = 0x40100000, len = 0x8000
  irom0_0_seg :                         org = 0x40201010, len = 0xE0000
}

commit 240984bf (2018-05-10): Merge pull request #1411 from MaBecker/wifi

rearange rf_cal_sector (fix #1294)
commit 6eba1766 (2018-05-10): rearange rf_cal_sector (fix #1294)

* sector map for last 5 sectors of flash: ABCCC
   A : rf cal
   B : rf init data
   C : sdk parameters
* change last FlashArea
* update ESP8266_BOARD
closed 2018-05-10 by gfwilliams