espruino/Espruino #1300: preprocessor question

MaBecker, 2018-01-07:
Hi @gfwilliams ,

not sure how to pass `WIFI_DBG=2` to get verbose debug output?

````
// disable debug for RELEASE
#ifdef RELEASE
#define WIFI_DBG 0
//#else
//#define WIFI_DBG 1
#endif
// Set WIFI_DBG to 0 to disable debug printf's, to 1 for important printf's, to 2 for verbose
// Normal debug
#if WIFI_DBG > 0
#define DBG(format, ...) os_printf(format, ## __VA_ARGS__)
#else
#define DBG(format, ...) do { } while(0)
#endif
// Verbose debug
#if WIFI_DBG > 1
#define DBGV(format, ...) os_printf(format, ## __VA_ARGS__)
#else
#define DBGV(format, ...) do { } while(0)
#endif
````

As workaround I changed the #ifdef statement to have normal debug plus verbose debug :-(

gfwilliams, 2018-01-08:
Not sure what you're asking? `#if WIFI_DBG > 1` should work fine.

If you want to pass it from the Make command you could do `DEFINES="WIFI_DBG=2" make` as long as you're not overwriting it in your code above (you could use `ifndef` to check)

MaBecker, 2018-01-08:
Yes, that is exactly what I was looking for - Thanks!

closed 2018-01-08 by MaBecker