espruino/Espruino #1302: ES5/ES6 wishlist

gfwilliams, 2018-01-09:
ES5/ES6 features that aren't in Espruino but that would be useful... I'll do one post per feature, and please use the `add reaction` button to vote on each one.

gfwilliams, 2018-01-09:
- [ ] `String.startsWith/endsWith/includes` - would be trivial to add

gfwilliams, 2018-01-09:
- [ ] Proper `let` implementation - it's just treated as `var` at the moment, but nobody seems to actually care

gfwilliams, 2018-01-09:
- [ ] Proper `const` implementation - it's just treated as `var` at the moment

gfwilliams, 2018-01-09:
- [ ] `Array.find/findIndex` - find by executing a function for comparison

gfwilliams, 2018-01-09:
- [ ] JS classes - probably not too painful, and seems like it'd tidy up a lot of code

gfwilliams, 2018-01-09:
- [ ] Default argument values

gfwilliams, 2018-01-09:
- [ ] `Array.includes` - easy, just `indexOf()>=0`

gfwilliams, 2018-01-09:
- [ ] `Object.entries` and `Object.values`

gfwilliams, 2018-01-09:
- [ ]  `String.padStart/padEnd`

gfwilliams, 2018-01-09:
- [ ] `async/await` - although this it likely too difficult to implement in Espruino as it is

gfwilliams, 2018-01-09:
- [ ] getters and setters

FlowerOfLife, 2018-02-20:
Arrow functions to bind "this" PLEASE

gfwilliams, 2018-02-20:
- [ ] [Destructuring](https://github.com/lukehoban/es6features#destructuring)

gfwilliams, 2018-02-20:
- [ ] [Rest and Spread](https://github.com/lukehoban/es6features#default--rest--spread)

jjok, 2018-02-26:
`import` and `export`?

vshymanskyy, 2018-02-28:
Ok, I was actually able to get `async/await` working on Espruino (using Babel transpiler and some hacking).
**Here is a full example:  https://github.com/vshymanskyy/espruino-await**
Looks pretty usable, but a native implementation would be so much better ;)
What do you think?

louisvangeldrop, 2018-03-24:
Webassembly support?

andrewwakeling, 2018-04-03:
I did a similar exercise to @vshymanskyy and got various modern JS features working using Webpack/Babel. See: https://github.com/andrewwakeling/espruino-webpack-babel-sample

Although supporting modern JS features (spread, destructuring) out-of-the-box would be convenient, these things are somewhat trivial to support if you compile the code.
 
I would love to see fixes/improvement to ensure that more modern JS features work correctly and more reliably. It would also be great to see improvements that would reduce the size of the compiled code. e.g. Transpiling async/await increases the bundle size a fair amount. I understand that supporting async/await isn't trivial, so I'm not pushing for this.

gfwilliams, 2018-05-18:
[ ] [Map](https://developer.mozilla.org/en-US/docs/Web/JavaScript/Reference/Global_Objects/Map) and [Set](https://developer.mozilla.org/en-US/docs/Web/JavaScript/Reference/Global_Objects/Set)

opichals, 2018-06-06:
- [ ]  [Proxy](https://developer.mozilla.org/en-US/docs/Web/JavaScript/Reference/Global_Objects/Proxy)

This would be nice way to deal with stuff like [case-insensitive Object key lookup](https://github.com/espruino/Espruino/issues/1458#issuecomment-394998831) in a standard way.

loganpowell, 2019-01-03:
Hi, sorry to chime in as a complete stranger, but given the whole Babel ecosystem, wouldn't it be more prudent to get complete ES5 feature coverage and then let users compile their ES6+ code down to ES5 before running through Espruino? I'm sorry if this is a naive question that has a straight answer somewhere obvious.

gfwilliams, 2019-01-07:
Hey, no problem. It's actually pretty trivial to set up your own Babel + Espruino CLI toolchain right now if you want to - just not inside the IDE.

The problem with Babel on Espruino  is really:

*if you write code on the left-hand side of the IDE it goes straight to Espruino, so you don't get to use babel - in fact babelified code will probably be harder to interact with.
* Without  a lot of work it'd also mess up the debugging functionality
* In fact in quite a few cases babel will significantly increase the amount of code used on the micro itself

It's not a reason not to do it, but I think there's still a place for implementing this stuff natively, especially where doing so reduces the amount of code storage needed.

loganpowell, 2019-01-07:
That makes sense. Thank you for the clarification.

jjok, 2019-01-07:
@loganpowell Is there anything in particular from ES5 that you've noticed is missing? Maybe open a separate feature request, if there is something.

loganpowell, 2019-01-07:
After reading @gfwilliams reply, I believe my point was made moot. I was thinking that developers could more easily use NPM libraries with Espruino if it had an ES5 complete API, but I would imagine the bloat that Babel compilation would add to the program, which could be a big performance issue.

MaBecker, 2019-11-07:
What about
```
> var text = 'uololooo';
undefined
> 
> // With ES6
undefined
> [...text].forEach(c => console.log(c))
u
o
l
o
l
o
o
o
undefined
> 
```

nebbishhacker, 2020-07-08:
[Nullish coalescing operator (??)](https://developer.mozilla.org/en-US/docs/Web/JavaScript/Reference/Operators/Nullish_coalescing_operator)

It would be nice to be able to write `options.foo ?? 42` rather than `(options.foo != null) ? options.foo : 42`

nebbishhacker, 2020-07-08:
Similar to the above: [Optional chaining (?.)](https://developer.mozilla.org/en-US/docs/Web/JavaScript/Reference/Operators/Optional_chaining)

Allows `foo?.bar?.baz()` instead of `if (foo && foo.bar && foo.bar.baz) foo.bar.baz()`

Dabolus, 2023-02-03:
- [ ] [BigInt](https://developer.mozilla.org/en-US/docs/Web/JavaScript/Reference/Global_Objects/BigInt) and the related typed arrays ([BigUint64Array](https://developer.mozilla.org/en-US/docs/Web/JavaScript/Reference/Global_Objects/BigUint64Array) & [BigInt64Array](https://developer.mozilla.org/en-US/docs/Web/JavaScript/Reference/Global_Objects/BigInt64Array))

This would allow working with values that do not fit into a standard JS number. In the past, 64 bit numbers were supported, but they were (rightfully) removed to make Espruino more JS-compatible. This is the current status:

```js
const testNumber = 18446744073709551615; // 2^64 - 1, max value for a uint64
console.log(testNumber); // Logs -1 instead of 18446744073709551615

const testBigInt = 18446744073709551615n; // Syntax error
```

Note that (for obvious reasons) this feature cannot be transpiled using babel/typescript/esbuild/whatever.

gfwilliams, 2023-02-03:
Thanks - but I think BigInt probably won't make it in, as it would involve quite a lot of extra code to handle the arbitrary precision maths needed (I believe implementing just 64 bit maths isn't good enough?). I guess there is a possibility that Espruino could be extended with some non-standard operator overloading function such that someone could then make a BigInt class in JS though - that could be a possibility.

Dabolus, 2023-02-13:
- [ ] [`Object.fromEntries`](https://developer.mozilla.org/en-US/docs/Web/JavaScript/Reference/Global_Objects/Object/fromEntries) from ES2019

Looks like it could be an easy win. I tried opening a draft PR [here](https://github.com/espruino/Espruino/pull/2327), but I didn't test it yet.

Dabolus, 2023-02-15:
New array static methods:

- [ ] [`Array.of`](https://developer.mozilla.org/en-US/docs/Web/JavaScript/Reference/Global_Objects/Array/of)
- [ ] [`Array.from`](https://developer.mozilla.org/en-US/docs/Web/JavaScript/Reference/Global_Objects/Array/from)

Both for standard arrays and typed arrays (`Uint8Array.of`, `Uint8Array.from`, etc.)

`Array.of` consists in simply creating and returning an array from the provided variadic arguments, while `Array.from` consists in creating an array from an iterable or an array-like, optionally providing a map function as second argument.

Dabolus, 2023-02-15:
- [ ] [`Object.getOwnPropertyDescriptors`](https://developer.mozilla.org/en-US/docs/Web/JavaScript/Reference/Global_Objects/Object/getOwnPropertyDescriptors)

I noticed that many modern transpilers/bundlers use it and take its existence for granted, so right now some transpiled code doesn't work unless you add additional quirks and polyfills to make this function available before the transpiled code generated by the transpiler/bundler. Given that `Object.getOwnPropertyDescriptor` already exists, it should be trivial to also implement this one. Example shim:

```js
Object.getOwnPropertyDescriptors = function getOwnPropertyDescriptors (obj) {
  if (obj === null || obj === undefined) {
    throw new TypeError('Cannot convert undefined or null to object');
  }

  const protoPropDescriptor = Object.getOwnPropertyDescriptor(obj, '__proto__');
  const descriptors = protoPropDescriptor ? { ['__proto__']: protoPropDescriptor } : {};

  for (const name of Object.getOwnPropertyNames(obj)) {
    descriptors[name] = Object.getOwnPropertyDescriptor(obj, name);
  }

  return descriptors;
}
```

Dabolus, 2023-02-15:
- [ ] [`TextEncoder`](https://developer.mozilla.org/en-US/docs/Web/API/TextEncoder) and [`TextDecoder`](https://developer.mozilla.org/en-US/docs/Web/API/TextDecoder)

This should also be something easy to implement in its simplest form, as something similar already exists on Espruino:

- `new TextDecoder().decode(uint8Array)` is functionally equivalent to `E.toString(uint8Array)`
- `new TextEncoder().encode(str)` is functionally equivalent to `new Uint8Array(E.toArrayBuffer(str))`

Sorry for the noise by the way; I'm trying to run an SDK built for the web on Espruino, so I'm writing down all the things I discover while doing it that might be beneficial to implement in the Espruino core to increase compatibility with the web.

gfwilliams, 2023-02-20:
Thanks! These look like good ideas. `getOwnPropertyDescriptors` seems easy and a good one to add if transpilers expect it. `Array.of` looks neat too - interestingly we have `E.toUint8Array`/etc which does similar things, so the code for that could be re-used.

It's a shame there isn't `String.of` as that would be a very neat fix for issues we are currently having with `E.toString` doing a bit more than you might expect from the name.

TextEncoder/Decoder might be a bit trickier to do in a spec compliant way - I think the closest might be `E.decodeUTF8`

Dabolus, 2023-03-02:
- [ ] New (web-compatible) Event API (see [EventTarget](https://developer.mozilla.org/en-US/docs/Web/API/EventTarget))

Since version 15, Node.js fully supports it too.

Considering that every object in Espruino is an `EventEmitter`, probably the easiest way to (kinda) support this would be to simply expose the same `emit`, `on`, and `removeListener` functions just with a different name (respectively `dispatchEvent`, `addEventListener`, and `removeEventListener`).

I also noticed that Espruino actually already has an `addEventListener` function implemented [here](https://github.com/espruino/Espruino/blob/b458383a245f8e2d71e13903aa6f33ffa8dece14/src/jswrap_object.c#L779-L785), but it isn't exposed as of today.

This is the shim I'm currently using to use a library that makes use of `EventTarget` + `CustomEvent`, in case someone else needs it (non-spec compliant, as it doesn't accept options such as `passive`, `once`, etc., but does its work):

```js
class Event {
  constructor(type, options) {
    this.type = type;
    this.bubbles = !!(options && options.bubbles);
    this.cancelable = !!(options && options.cancelable);
    this.composed = !!(options && options.composed);
  }
}

class CustomEvent extends Event {
  constructor(type, options) {
    super(type, options);
    this.detail = options && options.detail;
  }
}

// Every object is Espruino is a Node.js EventEmitter
// See: https://www.espruino.com/Reference#l_Object_on
class EventTarget {
  dispatchEvent(event) {
    return this.emit(event.type, event);
  }

  addEventListener(type, listener) {
    this.on(type, listener);
  }

  removeEventListener(type, listener) {
    this.removeListener(type, listener);
  }
}

// Then, use it like this
class Test extends EventTarget {
  test() {
    this.dispatchEvent(new CustomEvent('test', { detail: { a: 'b' }}));
  }
}

const t = new Test();

t.addEventListener('test', e => console.log('received test event:', e));
t.test();

// As soon as the event is emitted, this is logged in the console:
// received test event: CustomEvent: {
//   "type": "test",
//   "bubbles": false, "cancelable": false, "composed": false,
//   "detail": {
//     "a": "b"
//    }
//  }
```

gfwilliams, 2023-03-03:
Thanks - yes, I guess that could be an idea, although it's not actually adding any functionality that's not there already so I feel like this is low priority.

Especially on the more constrained devices like micro:bit 1 where we're having to cut features out to fit, it seems a step back to have to remove existing features just so we can duplicate some functions but with longer names.

> already has an addEventListener function implemented [here](https://github.com/espruino/Espruino/blob/b458383a245f8e2d71e13903aa6f33ffa8dece14/src/jswrap_object.c#L779-L785)

That one is for use internally (it uses `char*` as an argument) so it can't actually be exposed to JS - it just happens to have the same name :)

atjn, 2023-12-29:
- [ ] [`console.debug`](https://developer.mozilla.org/en-US/docs/Web/API/console/debug_static), [`console.info`](https://developer.mozilla.org/en-US/docs/Web/API/console/info_static), [`console.warn`](https://developer.mozilla.org/en-US/docs/Web/API/console/warn_static), [`console.error`](https://developer.mozilla.org/en-US/docs/Web/API/console/error_static)

These are just different ways of doing a [`console.log`](https://developer.mozilla.org/en-US/docs/Web/API/console/log_static), which is already supported. At a minimum, it would be nice to have them implemented as an alias of `console.log`, as many programs use them, and will currently fail to run because of that.

<details>
<summary>If you want to do it properly</summary>
You should implement some icons and color styling for the different log types. It would also be ideal if  warn and error have some sort of trace. If you implement the trace, you might as well also implement `console.trace`.

Here is how Firefox does it. (in the order: debug, info, warn, error, trace)
[image]

Here is how Chrome does it:
[image]

</details>

Any other methods from the [`console`](https://developer.mozilla.org/en-US/docs/Web/API/console) would also be cool, but they might be harder to implement.

gfwilliams, 2024-01-02:
Thanks - just done. Aliasing doesn't add a bunch of extra memory use. Stack traces are much harder to accomplish, and while we could send extra formatting chars and the IDE will parse them, I guess if someone is connected to the Bangle via something that's not the IDE, it could cause problems

mariusGundersen, 2024-01-02:
Could console.warn and console.error set the background and text color using ansi codes?

atjn, 2024-01-02:
I believe that is what gfwilliams is addressing at the end of their comment. Maybe some programs will not be able to handle ANSI codes, so better to not use them.

gfwilliams, 2024-01-02:
Exactly, yes. Potentially if you really wanted this, it's pretty easy to add a file (maybe to `.boot1` which should work on all Espruino devices) that overrides `console.warn`/etc with one that adds the relevant escape sequences.

atjn, 2024-02-26:
- [ ] [`globalThis`](https://developer.mozilla.org/en-US/docs/Web/JavaScript/Reference/Global_Objects/globalThis)

It can be polyfilled/implemented in Espruino with the single line `global.globalThis = global`. (it can be overwritten by design)
This is very useful for programs because it is guaranteed to exist in all JS contexts, whereas `global` is non-standard and not supported in all JS engines, and `this` is not always supported or does not point to the global object.

thyttan, 2024-04-02:
[ ] `String.replaceAll`

I've tried to use it maybe three times now over the course of a couple of months. The functionality is already achievable e.g. with `String.replace` using a replace function as second argument - but I have not managed to learn that yet...

- https://developer.mozilla.org/en-US/docs/Web/JavaScript/Reference/Global_Objects/String/replaceAll
- https://developer.mozilla.org/en-US/docs/Web/JavaScript/Reference/Global_Objects/String/replace#parameters

gfwilliams, 2024-04-03:
Thanks - that looks like a good one to have. If you're using a regex I believe the `g` postfix works fine already:  `"stringing".replace(/i/g,"_")`

But it's nice not to have to rely on regex

Lemenxzy, 2024-06-11:
I strongly hope that you can support async and await functions. Thank you

Lemenxzy, 2024-06-11:
I would like to kindly request, if it's not too much trouble, the support for async and await functions. Your assistance would be greatly appreciated. Thank you ever so much

Lemenxzy, 2024-06-11:
- [ async   await ] 
I would like to kindly request, if it's not too much trouble, the support for async and await functions. Your assistance would be greatly appreciated. Thank you ever so much

Lemenxzy, 2024-06-11:
- [ ] async await

I would like to kindly request, if it's not too much trouble, the support for async and await functions. Your assistance would be greatly appreciated. Thank you ever so much

gfwilliams, 2024-06-12:
There's already an item for this above if you look - but as mentioned there it's extremely difficult to implement in Espruino due to the fact it doesn't work by compiling down to bytecode. I don't think realistically there's a way to make this work except in very basic cases (`await` used in the main basic block of the function). If you want async/await it's probably best to look at transpiling

thyttan, 2024-12-30:
Today I wanted to use 
- [ ] Array.findLastIndex

Not very important though.