espruino/Espruino #1306: WIZnet W5100 Ethernet support

gfwilliams, 2018-01-12:
It's old, but there are a bunch of cheap Arduino footprint modules out there with the chip on them - so it'd make sense to allow it to work with those.

opichals, 2018-01-21:
I gave this a shot in https://github.com/espruino/Espruino/pull/1316 but its not really ready (see the PR comments).

opichals, 2018-02-15:
Can be closed?

Or do you want to debug the [timeout](http://forum.espruino.com/conversations/259497/#14082434) issue first? @gfwilliams

gfwilliams, 2018-02-16:
Did you have the timeout issue as well? If so I'd like to try and get that fixed - it feels like it's not really that usable at the moment with it.

opichals, 2018-02-16:
If you could share the particular test code it would be handy. I can do more testing on that later.

gfwilliams, 2018-02-16:
I just used this:

```
SPI1.setup({ mosi:D11, miso:D12, sck:D13 });
var eth = require("WIZnet").connect(SPI1, D10);

eth.setIP({
ip:"192.168.1.123",
subnet:"255.255.255.0",
gateway:"192.168.1.1",
dns:"8.8.8.8",
mac:"00:08:dc:ab:cd:ef"})

function get(url) {
  var http = require("http");
  http.get(url, function(res) {
    res.on('data', function(data) {
      console.log(data);
    });
    res.on('close', function(data) {
      console.log("Closed");
    });
  }).on('error',function() {
    console.log("Error getting "+url);
  });
}
get("http://www.pur3.co.uk/hello.txt");
```

It was running on a prototype board, but the same should be fine. It basically just became unresponsive after the `Hello World` webpage was received. If left for a while (~10sec? can't remember) then it sprung back to life.

It was definitely the while loop in the network_wiznet `close` function, so I guess it was just checking on the wrong flags - or maybe it should never have tried the 'soft' close at all and should have just 'hard' closed the socket.

opichals, 2018-02-18:
I can confirm the timeout issue.

Using
```js
function get(url) {
  var http = require("http");
  console.log(new Date().toString() + " get " + url);
  http.get(url, function(res) {
    res.on('data', function(data) {
      console.log(new Date().toString() + "\n" + data);
    });
    res.on('close', function() {
      console.log(new Date().toString() + " Closed");
    });
  }).on('error',function(e) {
    console.log(new Date().toString() + " Error", e);
  });
}
```

I get

```js
>get("http://www.pur3.co.uk/hello.txt");
Thu Jan 1 1970 00:11:01 GMT+0000 get http://www.pur3.co.uk/hello.txt
=undefined
Thu Jan 1 1970 00:11:01 GMT+0000
Hello World!
Thu Jan 1 1970 00:11:33 GMT+0000 Error { "code": -7,
  "message": "max sockets"
 }
Thu Jan 1 1970 00:11:33 GMT+0000 Closed
```

gfwilliams, 2018-02-19:
Interesting - thanks! I never noticed the error message.

commit 2f512205 (2018-09-27): Pixl.js: Fix 30s pause when closing sockets on WIZnet W5100 (fix #1306)
closed 2018-09-27 by gfwilliams