espruino/Espruino #1307: Pico: Uncaught InternalError: Timeout on I2C Write Transmit Mode 2

MaBecker, 2018-01-12:
Coding

````
I2C1.setup({sda:B7,scl:B6});

var nfc = require("PN532").connect(I2C1);

setTimeout(function(){
  print('starting');
  print(nfc.getVersion());

  nfc.SAMConfig(); // start listening
  setInterval(function() {
    nfc.findCards(function(card) {
      print("Found card "+card);
    });
  }, 1000);
},1000);
````

Output

````
starting
Uncaught InternalError: Timeout on I2C Write Transmit Mode 2
 at line 1 col 172
...sh(0);this.i2c.writeTo(36,b);b=this.i2c.readFrom(36,6);b[4]&...
                              ^
in function "cmd" called from line 1 col 22
var a=this.cmd([2],12);return{chip:"PN5"+a[2].toString(16),f...
                     ^
in function "getVersion" called from line 2 col 24
  print(nfc.getVersion());
                       ^
in function called from system
````

Wiring

[image]

Process.env

````
>process.env
={
  "VERSION": "1v95",
  "BUILD_DATE": "Dec 15 2017",
  "BUILD_TIME": "17:07:02",
  "GIT_COMMIT": "7dbffcf5da09570bd95714b2386a12ab25ab1d50",
  "BOARD": "PICO_R1_3",
  "CHIP": "STM32F401CDU6",
  "CHIP_FAMILY": "STM32F4",
  "FLASH": 393216, "RAM": 98304,
  "SERIAL": "5a006000-13513335-32373134",
  "CONSOLE": "USB",
  "EXPORTS": { "jsvLock": 292125, "jsvLockAgainSafe": 292111, "jsvUnLock": 292085, "jsvSkipName": 108241,
    "jsvMathsOp": 76683, "jsvMathsOpSkipNames": 76733, "jsvNewFromFloat": 292465, "jsvNewFromInteger": 292501, "jsvNewFromString": 294893,
    "jsvNewFromBool": 292485, "jsvGetFloat": 108609, "jsvGetInteger": 104585, "jsvGetBool": 109601, "jspeiFindInScopes": 80629,
    "jspReplaceWith": 83769, "jspeFunctionCall": 86257, "jspGetNamedVariable": 165573, "jspGetNamedField": 167221, "jspGetVarNamedField": 166817,
    "jsvNewWithFlags": 292281 }
 }

````

MaBecker, 2018-01-12:
A little code change, as you recommended and it works.

````
var i2c = new I2C();
i2c.setup({sda:B7,scl:B6});

var nfc = require("PN532").connect(i2c);

setTimeout(function(){
  print('starting');
  print(nfc.getVersion());

  nfc.SAMConfig(); // start listening
  setInterval(function() {
    nfc.findCards(function(card) {
      print("Found card "+card);
    });
  }, 1000);
},1000);
````

Output

````
starting
{
  "chip": "PN532",
  "firmware": "1.6",
  "support": 7 }
Found card 4,158,86,27,0,191,0
````

gfwilliams, 2018-01-13:
The write error happens when there's a problem with I2C communications. It shouldn't really be an 'internal error'.

Do you have I2C pullup resistors?

MaBecker, 2018-01-13:
> Do you have I2C pullup resistors?

You mean additional to the "NFC Module V3" onboard 4k7 pullups? NO

Added external 4k7 pullups and removed the NFC board.

[image]

Coding
```
SDA=B7;SCL=B6;

function readFrom(i2c,id,c) {
  try {
    return i2c.readFrom(id,c);
  }
  catch(e) {
    return 'no device ['+e+']';
  }
}

console.log('\I2C Soft and Hard Test');

// I2C-Soft
i2c = new I2C();
i2c.setup({sda:SDA,scl:SCL});
rc = 
console.log('I2C-Soft readFrom(0,1):', readFrom(i2c,0,1));

// I2C-Soft
I2C1.setup({sda:SDA,scl:SCL});
console.log('I2C-Hard readFrom(0,1):', readFrom(I2C1,0,1));
```

Output

```
I2C Soft and Hard Test
I2C-Soft readFrom(0,1): new Uint8Array([255])
I2C-Hard readFrom(0,1): no device [Error: I2C device not responding]
=undefined
```

gfwilliams, 2022-06-08:
Closing this....

I don't think I've seen anyone else reporting similar (and this is 4 years old) so I guess this was a hardware issue?

closed 2022-06-08 by gfwilliams