espruino/Espruino #1309: Wifi.save does not work for espruino_1v95_esp8266_4mb_combined_4096.bin

MaBecker, 2018-01-16:
have to investigate

Workaround: erase_flash and flash_write files from folder espruino_1v95_esp8266_4mb

mbence, 2018-01-16:
I can confirm this on Wemos D1 mini v3.0.0 (original). Everything seems to work fine until wifi.save(). After reboot the board is stuck in an endless loop of exception(3).

The solution (or workaround) is to use `--flash_size 4MB-c1` for flashing, as you wrote at http://forum.espruino.com/conversations/314732/#14021236
Maybe worth updating the README_flash.txt there?

But thanks for the solution anyway!

MaBecker, 2018-01-18:
Yep - it's already done :)

https://github.com/espruino/Espruino/blob/master/targets/esp8266/README_flash.txt#L18-L22

@gfwilliams can you please replace the README_flash.txt in the espruino 1v95 zip file?

Flash is just  one point, using the combined 4MB file does not reconnect to AP on my side.

**still troubleshooting**

MaBecker, 2018-01-21:
There are some issues about "auth_expire" 

https://github.com/search?q=auth_expire&type=Issues&utf8=%E2%9C%93

wilberforce, 2018-04-08:
@MaBecker 

I don't think this is still an issue - I think it can be closed.

closed 2018-04-08 by MaBecker