espruino/Espruino #1312: EspruinoWifi MQTT subscription

mrwhy-orig, 2018-01-17:
I'm trying to subscribe to a MQTT broker by using the espruino MQTT Module, but as I do, the MQTT Client gets disconnected. The broker I'm using is Moquette. I tried other clients which are working just fine.

`
function mqttConnect() {
  mqtt = require("MQTT").create(MQTT_HOST, mqttOptions);
  mqtt.connect();
  mqtt.on('connected', function() {
    console.log("MQTT connected");
    // subscribe to wildcard for our name
    mqtt.subscribe("/test/#");
  });
  mqtt.on('publish', mqttMessage);
  mqtt.on('disconnected', function() {
    console.log("MQTT disconnected... reconnecting.");
    setTimeout(function() {
      mqtt.connect();
    }, 1000);
  });
}
`

gfwilliams, 2018-01-17:
That all looks fine, and works for me (without any mqttOptions, since obviously I don't know what those were).

However if you supply an invalid hostname you'll get something like this:

```
MQTT disconnected... reconnecting.
Uncaught Error: This socket is closed.
```

Note that you never received a `MQTT connected` event - the disconnected event just gets fired automatically in the case of a connection that fails, so you can handle it and try and reconnect. That's more that likely your problem?

Please ask any questions like this in the forum (http://forum.espruino.com/) in the future though - you'll generally get much more helpful and speedy advice there.

gfwilliams, 2018-01-18:
Yeah - problem is the MQTT Library (which might be at issue here) isn't part of this GitHub project - it's in EspruinoDocs :)

But anyway, let's try and sort this out in here. Can you try setting `mqttOptions` to `undefined`? It's possible you're specifying an MQTT version that the library isn't conforming to.

Also, you really need to try and get some logs from Moquette. Chances are it'll be dropping the connection because it thinks there's some error, so knowing what that is would be a big help.

mrwhy-orig, 2018-01-18:
Sorry @gfwilliams it is a bit confusing to find the modules in the EspruinoDocs project. 

So i set the options to undefined still the same issue. Then I compiled Moquette by myself, and debugged to the message handler. It seems to be an issue with the message Id. Next test I'm going to do is to check the subscription packages with wireshark.

MaBecker, 2018-01-18:
Herr you find that module
https://github.com/espruino/EspruinoDocs/blob/master/modules/MQTT.js

mrwhy-orig, 2018-01-18:
@MaBecker thx ;)

@gfwilliams the messageid is allways 0 in subscriptions but with other clients messageid > 0. At the moment i don't know if it is an issue in this module or at Moquette.

gfwilliams, 2018-01-19:
Try uploading now - looks like there was a refactoring error which meant that `mqttPid` referenced `this`, which was undefined - and so caused the packet ID to always be 0.

However this *might* mean that other commands (like publish) that are sent with a QoS of 0 (because TCP/IP) cause problems, because they then have a nonzero message ID. Let me know how you get on.