espruino/Espruino #1320: NRF5x SDK 14.2 & NFC

drandreas, 2018-01-25:
Hi Gordon

I've noticed you started preparing for an bump of Nordics SDK to 14.x. Some of my NFC customisations do conflict with Nordics changes others are rendered obsolete by Nordics
own implementation.

Would it help if I would port my NFC extensions to SDK 14.2?

gfwilliams, 2018-01-26:
That would be a huge help, yes! Thanks!

I had a quick look at the diffs, but it looked like a reasonably big chunk of work for me given I don't really know what goes on internally :) To be SDK14 building all you should have to do is copy in the SDK, then swap over the commented sections at the top of NRF52.make

drandreas, 2018-02-04:
With 14 Nordic added almost all features I had to implement on top of 13. I anticipate a significantly cleaner look in 14. Is it sufficient if I complete a stable version by the end of February 2018?

gfwilliams, 2018-02-05:
Yes - that'd be great! Honestly I'm just really happy that you're willing to do it at all!

drandreas, 2018-02-18:
I've reworked my NFC patches so they apply on top of SDK 14.2.0.

One tiny incompatibility remains Nordic defined with SDK14 `HAL_NFC_PARAM_ID_NFCID1` my previous patches defined `HAL_NFC_PARAM_ID_UID` either of the two constants has to be renamed.

[0001-NFC-Implemented-tx-of-4-bit-frames.patch.gz](https://github.com/espruino/Espruino/files/1734924/0001-NFC-Implemented-tx-of-4-bit-frames.patch.gz)
[0002-NFC-Whitelist-all-commands.patch.gz](https://github.com/espruino/Espruino/files/1734925/0002-NFC-Whitelist-all-commands.patch.gz)
[0003-NFC-Use-relax-delay-in-JS-callback.patch.gz](https://github.com/espruino/Espruino/files/1734926/0003-NFC-Use-relax-delay-in-JS-callback.patch.gz)
[0004-NFC-Fixed-integer-over-underflow.patch.gz](https://github.com/espruino/Espruino/files/1734927/0004-NFC-Fixed-integer-over-underflow.patch.gz)
[0005-NFC-Further-driver-hardening.patch.gz](https://github.com/espruino/Espruino/files/1734928/0005-NFC-Further-driver-hardening.patch.gz)
[0006-NFC-Removed-superfluous-dynamic-output-buffer.patch.gz](https://github.com/espruino/Espruino/files/1734929/0006-NFC-Removed-superfluous-dynamic-output-buffer.patch.gz)

Sorry about the gzips but GitHub doesn't let me upload patches.

gfwilliams, 2018-02-19:
That's great - thanks! Really handy to have a patch set too so hopefully it'll be a bit easier to merge in to newer SDKs.

I guess from the point of view of making it easier to move to new SDKs, we could rename your HAL_NFC_PARAM_ID_UID to HAL_NFC_PARAM_ID_NFCID1? But I can do that later.

gfwilliams, 2018-05-24:
Thanks! Not sure how I missed this when I was looking :(

commit 7e394e7b (2018-05-24): adding NFC support for SDK14 courtesy of @drandreas (fix #1320)
closed 2018-05-24 by gfwilliams