espruino/Espruino #1322: Starting to include JS in build

MaBecker, 2018-01-27:
## First of all "This is a great feature"  - Thanks 

And pretty easy to use if you have a build environment :)

tried to add two modules - result 50:50 

- [x] SSD1306_I2C.min.js (just I2C no SPI), works like designed.
- [ ] [tinyMQTT.min.js ](https://github.com/MaBecker/tinyMQTT/blob/master/tinyMQTT.min.js) does not connect to mqtt server

info on mqtt server side:
> 1517065191: New connection from 192.168.194.121 on port 1883.
1517065191: Socket error on client <unknown>, disconnecting.

tinyMQTT.js  works when downloaded.

Used espruini -m tinyMQTT.js -o tinyMQTT.min.js to get min file.

* Are there any code limits on JS code? 

* How to tell Espruino WEB IDE (Chrome App 0V67) that JS modules  are integrated? 
    generated[ json board file](https://raw.githubusercontent.com/MaBecker/esp8266/master/Espruino/json/ESP8266_4MB.json) and added url to IDE - still transfering included module.

MaBecker, 2018-01-29:
Can confirm that 'Wifi' and 'http' works, as uses in [jstools.min.js ](https://github.com/MaBecker/Espruino/blob/jsmodules/libs/js/jstools.min.js)

Why is `espruino -m jstools.js -o jstools.min.js` adding `Modules.removeAllCached(),`to this file?

gfwilliams, 2018-01-29:
Honestly, you'll have to try and debug that yourself. It may also be related to how ESP8266 executes from flash memory, so other platforms *might* be fine.

> Are there any code limits on JS code?

65k - the max size of a 'flat string' in Espruino

> How to tell Espruino WEB IDE (Chrome App 0V67) that JS modules are integrated?

Exactly as you did, with the JSON. Perhaps the IDE didn't manage to find out the board type when it connected and so didn't load the new JSON file? Chrome tends to cache files too, so you'd need to put the no-cache header on your server.

> `Why is espruino -m jstools.js -o jstools.min.js adding Modules.removeAllCached(),to this file?`

Because it's not just minifying a module. That is done with this: https://github.com/espruino/EspruinoDocs/blob/master/bin/minify.js

It's actually creating JS to upload, and the `removeAllCached` is to remove any pre-existing modules if `reset()` hadn't been used first. There's definitely an argument that it could be removed though...

Note though: it won't remove any of the modules that are included in the build.

MaBecker, 2018-01-29:
Thanks for your comments and hints.

Now using minify.js - wow this works nice.

MaBecker, 2018-01-29:
- [x] tinyMQTT works now including 256 byte 

next is captivePortal.min.js

gfwilliams, 2018-01-29:
Can we close this then? You already have a bug for the captive portal.

And honestly, just because you *can* stick whatever bits of JS you want in the firmware doesn't mean your *should*. It's a great personal project, but it's Espruino's main issue tracker isn't a good place for it - maybe the Espruino forum?

MaBecker, 2018-01-29:
sure

closed 2018-01-29 by MaBecker