espruino/Espruino #1325: Uncaught SyntaxError: Got catch expected EOF

jjok, 2018-02-01:
Hi. This seems like a weird one, but I don't think JS should be sensitive to whitespace like this. I assume it's an issue with Espruino, rather than the IDE.

This works fine.
```
try {
  // ...
} catch(e) {
  // ...
}
```
This causes the above error:
```
try {
  // ...
}
catch(e) {
  // ...
}
```
Using the Chrome app with an Espruino WiFi running 1v95. Thanks.

gfwilliams, 2018-02-02:
Thanks - this is actually due to the way Espruino parses code on the command-line. It counts brackets when it gets a newline to see if it should execute or not, so if all brackets are finished it'll execute. It's more obvious if you paste on the left-hand side.

The error won't occur inside a function or if you choose 'save on send' in the IDE, but there are actually a bunch of hacks in the IDE to work around these kind of things at the root scope. It looks like there just isn't one for `try...catch` - so I've just added one. When I release a new version of the Web IDE, it won't have this problem.

Just a note though - if you're doing something big at root scope that can throw an exception, it might not be doing what you expect. It's going to execute *at upload time*, not when the board powers on. See http://www.espruino.com/Saving for more info.

closed 2018-02-02 by gfwilliams

jjok, 2018-02-02:
Thanks. I hadn't seen that page about saving.