espruino/Espruino #1328: [ESP8266] Wifi won't connect using numeric password

martinlechner1, 2018-02-06:
I'm using this on an ESP8266 board, but I assume this will also happen with Espruino ESP8266 module.
```
var wifi = require("Wifi");
wifi.connect("myap", {password: "123"});
```
won't connect.
Changing PW on access point to: `123!` fixes this.

https://github.com/espruino/Espruino/blob/bfeb3cc6290d04c7df00fb7810d702b574ecbe76/libs/network/esp8266/jswrap_esp8266_network.c#L336
seems to be the issue. Maybe the JSON deserializer converts the password to a number when only given numeric characters in the string? Unfortunately I'm not a C developer -> I can't help here much on the issue.

gfwilliams, 2018-02-06:
What error message do you get? I imagine it's not `Expecting options.password to be a string but got ...`.

This could be to do with the rules for WPA2 passwords... I thought they had to be 8 chars or more, but maybe there's a rule for padding them?

martinlechner1, 2018-02-06:
Oh the password was longer than eight characters. The wifi got stuck in "connecting" state, but no error message / connect callback call at all. (Using Espruino Web IDE)

MaBecker, 2018-02-18:
Well mine does:

```
var Wifi = require('Wifi');

var SSID = "iP6PMB";
var password = '0123456789';

function onInit(){
  console.log('try to connect to',SSID,'with password',password);
  Wifi.connect(SSID, {password:password}, function(e){
      console.log(e);
      console.log(Wifi.getIP());
  });
}
setTimeout(onInit, 1000);
```

> try to connect to iP6PMB with password 0123456789
null
{
  "ip": "172.20.10.12",
  "netmask": "255.255.255.240",
  "gw": "172.20.10.1",
  "mac": "5c:cf:7f:c1:12:0b"
 }
>

gfwilliams, 2018-02-18:
Just in case try 12345678.

Starting with 0 with a 9 in there could *maybe* fall into some JavaScript
crazy place because parseInt(X).toString()!=x

MaBecker, 2018-02-18:
same result 

```
var Wifi = require('Wifi');

var SSID = "iP6PMB";
var password = '12345678';

function onInit(){
  console.log('try to connect to',SSID,'with password',password);
  Wifi.connect(SSID, {password:password}, function(e){
    console.log('duration:',((ts.stop-ts.start)/1000).toFixed(2));
    if ( e !== null ) {
      console.log(e);
    } else {
      console.log(Wifi.getIP());
    }
  });
}

setTimeout(onInit, 1000);
```

>try to connect to iP6PMB with password 12345678
{
  "ip": "172.20.10.12",
  "netmask": "255.255.255.240",
  "gw": "172.20.10.1",
  "mac": "5c:cf:7f:c1:12:0b"
 }
>

gfwilliams, 2018-02-19:
Thanks!

@martinlechner1 unless you have a specific example with a password that really does fail with up to date firmware we'll have to close this. It could just have been a compatibility issue with your access point.