espruino/Espruino #1335: Enhance memoryArea() fit for use with EPSs

MaBecker, 2018-02-12:
What do you think? 

add  to all ESP`<boards>`.py

```
'makefile' : [
    .....
     'MEMBASE=0x40200000',
     'MAXMEM=0x100000',
    .....
]
```
and enhance jswrap_espruino.c
```
JsVar *jswrap_espruino_memoryArea(int addr, int len) {
  if (len<0) return 0;
  if (len>65535) {
    jsExceptionHere(JSET_ERROR, "Memory area too long! Max is 65535 bytes\n");
    return 0;
  } 
#ifdef MEMBASE
  if ( addr  > $(MAXMEM)) {
    jsExceptionHere(JSET_ERROR, "This addr is not memory mapped \n");
    return 0;
  } else {
    return jsvNewNativeString((char*)(size_t)addr+$(MEMBASE), (size_t)len);
  }
#else
  return jsvNewNativeString((char*)(size_t)addr, (size_t)len);
#endif
}
```

MaBecker, 2018-02-12:
but this works:

add DEFINE entries to ESP family/ESP`<boards>`.make

```
DEFINES += -DMAXMEM=0x100000
DEFINES += -DMEMBASE=0x40200000
```

and enhance jswrap_espruino.c this way
```
JsVar *jswrap_espruino_memoryArea(int addr, int len) {
  if (len<0) return 0;
  if (len>65535) {
    jsExceptionHere(JSET_ERROR, "Memory area too long! Max is 65535 bytes\n");
    return 0;
  } 
#ifdef MEMBASE
  jsiConsolePrintf("Have MEMBASE\n");
  if ( addr  > MAXMEM) {
    jsExceptionHere(JSET_ERROR, "This addr is not memory mapped \n");
    return 0;
  } else {
    return jsvNewNativeString((char*)(size_t)addr+MEMBASE, (size_t)len);
  }
#else
  jsiConsolePrintf("NO MEMBASE\n");
  return jsvNewNativeString((char*)(size_t)addr, (size_t)len);
#endif
}
```

and you get
```
JsVar *jswrap_espruino_memoryArea(int addr, int len) {
  if (len<0) return 0;
  if (len>65535) {
    do { static const char flash_str[] __attribute__((section(".irom.literal"))) __attribute__((aligned(4))) = "Memory area too long! Max is 65535 bytes\n"; jsExceptionHere_flash(JSET_ERROR, flash_str); } while(0);
    return 0;
  }
  do { static const char flash_str[] __attribute__((section(".irom.literal"))) __attribute__((aligned(4))) = "Have MEMBASE\n"; jsiConsolePrintf_int(flash_str); } while(0);
  if ( addr > 0x100000) {
    do { static const char flash_str[] __attribute__((section(".irom.literal"))) __attribute__((aligned(4))) = "This addr is not memory mapped \n"; jsExceptionHere_flash(JSET_ERROR, flash_str); } while(0);
    return 0;
  } else {
    return jsvNewNativeString((char*)(size_t)addr+0x40200000, (size_t)len);
  }
}
```

wilberforce, 2018-02-12:
Hi,

The ESP32 does not used a fixed offset as per the ESP8266. It is set at runtime and saved into `romdata_jscode`

`err=esp_partition_mmap(p, 0, p->size, SPI_FLASH_MMAP_DATA, (const void**)&romdata_jscode, &hrom);`

https://github.com/espruino/Espruino/blob/9e6fc0aa28ac9e2b5fc286dc9eccf558e7dcfb98/targets/esp32/main.c#L74

This is how it is handled in the Flash code:

https://github.com/espruino/Espruino/blob/bfeb3cc6290d04c7df00fb7810d702b574ecbe76/src/jswrap_flash.c#L441-L457

So this can't be done with a #define MEMBASE constant directly in the ESP32 case.

MaBecker, 2018-02-13:
Thanks for sharing this.

MaBecker, 2018-02-13:
Well, MEMBASE can be defined for ESP32 like this:

```
DEFINES += -DMEMBASE='(extern char* romdata_jscode)'
```

not testes only checked gcc -E output

```
JsVar *jswrap_espruino_memoryArea(int addr, int len) {
  if (len<0) return 0;
  if (len>65535) {
    do { static const char flash_str[] __attribute__((section(".irom.literal"))) __attribute__((aligned(4))) = "Memory area too long! Max is 65535 bytes\n"; jsExceptionHere_flash(JSET_ERROR, flash_str); } while(0);
    return 0;
  }
  if ( addr > 0x100000) {
    do { static const char flash_str[] __attribute__((section(".irom.literal"))) __attribute__((aligned(4))) = "This addr is not memory mapped \n"; jsExceptionHere_flash(JSET_ERROR, flash_str); } while(0);
    return 0;
  } else {
    return jsvNewNativeString((char*)(size_t)addr+(extern char* romdata_jscode), (size_t)len);
  }
}
```

gfwilliams, 2018-02-13:
You can add `"DEFINES+=-DMAXMEM=0x100000"` to the `BOARD.py` file - it's done in a few other files if you look, so that's a nice easy way of getting it in there without changing the Makefile.

Your other option, which I guess might be more sensible given that ESP8266/ESP32 are different and there's already code in `jswrap_flash.c`, is to define a function `jshFlashGetMemMapAddress` in `jshardware.c` that maps a flash address to a memory mapped address (or 0 if it's not available) and then just make it pass through the address it's given in all the other platforms, but do fancy stuff in ESP8266/ESP32. Then you could use that in jswrap_flash *and* memoryArea, peek, and wherever else it's needed.

**It should also work for addresses in RAM** - I think your proposed changes to memoryArea would break if you try and access RAM with them? They'd also break if you used a pre-memory-mapped address.

I'm trying to tweak the flash stuff at the moment so I'll add it to my branch and update it.

gfwilliams, 2018-02-13:
Ok, it's here: https://github.com/espruino/Espruino/tree/flash_storage

It should:

* Fix your memoryArea issues - but you'd need to test
* Add `require(`Flash`).readFile/writeFile/listFiles/etc`
* Use the storage for saving code - so you can now do `setBootCode` without blowing away code from `save()`. With a bit of tweaking the IDE should also be able to create multiple writeFile calls instead of `E.setBootCode`, which will allow you to really max out the saved code (on the 4MB ESP8266 that means filling all 16 x 4096 byte pages with JS code).

gfwilliams, 2018-02-13:
Ideally the compaction and block iteration code would be a little better so that it could clean up some flash pages even if it doesn't have enough RAM to load all the saved boot code in - but for the moment it's reasonably promising.

gfwilliams, 2018-02-13:
@MaBecker I just merged this into master. Please can you check if memoryArea works ok for you, and close if so?

MaBecker, 2018-02-13:
Yes of course, thanks a lot!

gfwilliams, 2018-02-13:
Great! `require("Storage")` seems to work on ESP8266 as well - although you may need `require("Storage").eraseAll()` first to clear it out (I just committed a refactor).

I discovered an issue that might have been causing instability on ESP8266 too - ESP8266 can't deal with flash writes **from** non-aligned addresses, it just rebooted. I think it could have been pretty easy to get that using FlashWrite from flat strings

MaBecker, 2018-02-13:
- [x]  f.write expects addr be aligned by 4 

```
>f.write('Espruino rocks !',1003520);
=undefined
>f.write('Espruino rocks !',1003521);
Uncaught Error: Data and address must be multiples of 4
 at line 1 col 35
f.write('Espruino rocks !',1003521);
                                  ^
>f.write('Espruino rocks !',1003522);
Uncaught Error: Data and address must be multiples of 4
 at line 1 col 35
f.write('Espruino rocks !',1003522);
                                  ^
>f.write('Espruino rocks !',1003523);
Uncaught Error: Data and address must be multiples of 4
 at line 1 col 35
f.write('Espruino rocks !',1003523);
                                  ^
>f.write('Espruino rocks !',1003524);
=undefined
````

- [x] E.memoryArea() can handle all combinations
```
var len = 16;
var addr = 1003520;
var f = require('Flash');
f.erasePage(addr);
f.write('Espruino rocks !',1003520);

// fixed addr, variable length
for ( i = 0 ;i <= len ; i++) {
  console.log(E.memoryArea(1003520,i));
}
/*
E
Es
Esp
Espr
Espru
Esprui
Espruin
Espruino
Espruino
Espruino r
Espruino ro
Espruino roc
Espruino rock
Espruino rocks
Espruino rocks
Espruino rocks !
*/

// variable  addr, fixed length
for ( i = 0 ;i < len ; i++) {
  console.log(E.memoryArea(1003520+i,1));
}
/*
E
s
p
r
u
i
n
o
 
r
o
c
k
s
 
!

*/

// variable  addr, variable lenght
for ( i = 0 ;i < len ; i++) {
  console.log(E.memoryArea(1003520+i,len-i));
}
/* output
Espruino rocks !
spruino rocks !
pruino rocks !
ruino rocks !
uino rocks !
ino rocks !
no rocks !
o rocks !
 rocks !
rocks !
ocks !
cks !
ks !
s !
 !
!
*/
```
- [x] new module Storage

```
st = require('Storage');

st.<tab>
ompact             debug
erase               eraseAll
list                read
write               apply
bind                call
replaceWith         clone
emit                hasOwnProperty
length              on
removeAllListeners  removeListener
toString            valueOf

st.eraseAll()

>st.write('var1','value1');
[Flash] CreateFile
[Flash] CreateFile new 0x000e3010
=true
>st.write('var2','value2');
[Flash] CreateFile
[Flash] CreateFile new 0x000e3028
=true
>st.write('var3','value3');
[Flash] CreateFile
[Flash] CreateFile new 0x000e3040
=true
>st.list()
=[
  "var1",
  "var2",
  "var3"
 ]

>require('ESP8266').reboot();

>st = require('Storage');

>st.debug();

DEBUG FILES 66 live
PAGE 0x000e3000 (4096 bytes) - 66 live 4024 free
0x000e3010	var1	(6 bytes)
0x000e3028	var2	(6 bytes)
0x000e3040	var3	(6 bytes)
=undefined

>st.list()
=[
  "var1",
  "var2",
  "var3"
 ]
> 
>st.read('var1')
="value1"
>st.read('var2')
="value2"
>st.read('var3')
="value3"
> 

>save()
=undefined
Compacting Flash...
[Flash] Compacting
[Flash] Compacting - all data fits in RAM
[Flash] Compacting - copy data to swap
[Flash] EraseAll
[Flash] Compacting - copy data from swap
[Flash] CreateFile
[Flash] CreateFile new 0x000e3010
[Flash] CreateFile
[Flash] CreateFile new 0x000e3028
[Flash] CreateFile
[Flash] CreateFile new 0x000e3040
[Flash] Compaction Complete
Calculating Size...
[Flash] CreateFile
[Flash] CreateFile new 0x000e3058
Writing...
> 

>st.read('var3')
="value3"
>st.read('var2')
="valu\xFF\xFF"      **<- should be value2**
>st.read('var1')
="value1"
>st.read('var2')
="valu\xFF\xFF"
>st.debug()
DEBUG FILES 1232 live
PAGE 0x000e3000 (4096 bytes) - 1232 live 2856 free
0x000e3010	var1	(6 bytes)
0x000e3028	var2	(6 bytes)
0x000e3040	var3	(6 bytes)
0x000e3058	varimage	(1150 bytes)
=undefined
>E.memoryArea(0x000e3028,6)
="valu\xFF\xFF"    **<- should be value 2**

```

Amazing! Thank you very much for this cool buildin module!

So lets tripple  save pages for ESP8266_4MB  wich makes 192KB for 
E.setBootCode(),  save() and Store (PR will follow). 

@wilberforce or @jumjum123  can someone please check this on EPS32?

wilberforce, 2018-02-14:
Seems to write - however read is crashing....

```
 1v95.213 (c) 2017 G.Williams
Espruino is Open Source. Our work is supported
only by sales of official boards and donations:
http://espruino.com/Donate
>process.env
={
  "VERSION": "1v95.213",
  "BUILD_DATE": "Feb 14 2018 12:05:06",
  "GIT_COMMIT": "626ba6677427fe0eebceba3ec766344cb49749bc",
  "BOARD": "ESP32",
  "FLASH": 0, "RAM": 524288,
  "SERIAL": "240ac400-2f90",
  "CONSOLE": "Serial1",
  "MODULES": "Flash,Storage,fs," ... "r,crypto,neopixel",
  "EXPORTS": { "jsvLock": 1074618384, "jsvLockAgainSafe": 1074618528, "jsvUnLock": 1074620144, "jsvSkipName": 1074636128,
    "jsvMathsOp": 1074645544, "jsvMathsOpSkipNames": 1074647544, "jsvNewWithFlags": 1074630756, "jsvNewFromFloat": 1074631664, "jsvNewFromInteger": 1074631576,
    "jsvNewFromString": 1074631188, "jsvNewFromBool": 1074631636, "jsvGetFloat": 1074640464, "jsvGetInteger": 1074634760, "jsvGetBool": 1074640868,
    "jspeiFindInScopes": 1074660492, "jspReplaceWith": 1074660228, "jspeFunctionCall": 1074670648, "jspGetNamedVariable": 1074661984, "jspGetNamedField": 1074662372,
    "jspGetVarNamedField": 1074662484 },
  "EXPTR": 1073417248 }
>st = require('Storage');
=function () { [native code] }
>st.write('var1','value1');
Uncaught Error: File already written with different data
 at line 1 col 25
st.write('var1','value1');
                        ^
>require("Storage").eraseAll()
=undefined
>st.write('var1','value1');
=true
>st.list();
=[
  "var1"
 ]
>st.read('var3')
=undefined
>st.read('var1')
=" 
```

Stopped outputing after `="`

I compliled with RELEASE=1

perhaps this is why:

```
>st = require('Storage');
=function () { [native code] }
>st.debug()
Uncaught Error: Function "debug" not found!
 at line 1 col 4
st.debug()
   ^
>st.
compact             erase
eraseAll            list
read                write
apply               bind
call                replaceWith
clone               emit
hasOwnProperty      length
on                  removeAllListeners
removeListener      toString
valueOf
>st. 
```

wilberforce, 2018-02-14:
Looks like the old code save area needs to be blanked?

```
esptool.py  --chip esp32 --port COM3 --baud 921600 --after soft_reset write_flash -z --flash_mode dio --flash_freq 40m --flash_size detect 0x100000 blank.bin
```

However it does not look like it is writing correctly:

```
>st.debug();
DEBUG FILES 3236 live
PAGE 0x00100000 (4096 bytes) - 3236 live 856 free
0x00100010	var1	(6 bytes)
0x00100028	ÿÿÿÿÿÿÿÿ	(0 bytes)
0x00100038	varimage	(3182 bytes)
=undefined
>st.erase();
=undefined
>st.debug();
DEBUG FILES 3236 live
PAGE 0x00100000 (4096 bytes) - 3236 live 856 free
0x00100010	var1	(6 bytes)
0x00100028	ÿÿÿÿÿÿÿÿ	(0 bytes)
0x00100038	varimage	(3182 bytes)
=undefined
```

MaBecker, 2018-02-14:
`reset(true)` is the quickest way to delete save code pages

wilberforce, 2018-02-14:
recompiled with `make clean && BOARD=ESP32 DEBUG=1 make`

>st.debug();
DEBUG FILES 22 live
PAGE 0x00100000 (4096 bytes) - 22 live 4072 free
0x00100010      var1    (6 bytes)
=undefined
>st.eras

erase               eraseAll

>st.eraseAll()
[Flash] EraseAll
=undefined
>st.debug();
DEBUG FILES 0 live
=undefined

=undefined
>st.write('var1','test');
[Flash] CreateFile
[Flash] CreateFile new 0x00100010
=true

wilberforce, 2018-02-14:
> reset(true) is the quickest way to delete save code pages

```
>reset(true);
Erasing saved code.[Flash] EraseFile 0x00100038
Done!
=undefined
 ____                 _
|  __|___ ___ ___ _ _|_|___ ___
|  __|_ -| . |  _| | | |   | . |
|____|___|  _|_| |___|_|_|_|___|
         |_| espruino.com
 1v95.213 (c) 2017 G.Williams
Espruino is Open Source. Our work is supported
only by sales of official boards and donations:
http://espruino.com/Donate
>
=undefined
>
=undefined
>var st=require("Storage");
=function () { [native code] }
>st.debug();
DEBUG FILES 38 live
PAGE 0x00100000 (4096 bytes) - 38 live 856 free
0x00100010	var1	(6 bytes)
0x00100028	ÿÿÿÿÿÿÿÿ	(0 bytes)
0x00100038	varimage	(3182 bytes)	 DELETED
=undefined
```

So the erase is not working in the ESP32 case

MaBecker, 2018-02-14:
Yes - reset(true) is not wiping Storage pages, so have to use `st.eraseAll()`

MaBecker, 2018-02-14:
Will run further test with hashCode  to verify  vars values

gfwilliams, 2018-02-14:
Yeah, `reset(true)` won't wipe everything (only saved code). I guess it could do - but I imagine if you've got wifi details and stuff you probably want them though?

Erase seems to work as it's supposed to @wilberforce? `st.eraseAll()` in your post above looks good.

However *it shouldn't crash if the memory is corrupted* - it'll just refuse to write new files. If you can break it, could you try and give me an image of your memory area and I'll try and make it work ok?

The crash on read seems odd though. Could it be that ESP32 suffers from the same crash on unaligned accesses that ESP8266 had? What if you do debug and then try and view the area directly with `E.memoryArea`? 

@MaBecker I believe I just fixed the issue you were having with compaction knocking the final few bytes off one of the values.

gfwilliams, 2018-02-14:
Closing this now - lets use https://github.com/espruino/Espruino/issues/1337 for ESP32 issues

MaBecker, 2018-02-14:
now on  >process.env.GIT_COMMIT
="eba16e43a675ee2dcdbc95cfd5e506271fa03c11"

missing some functions

st = require('Storage');
=function () { [native code] }
>st.<tab>
apply               bind
call                replaceWith
clone               emit
hasOwnProperty      length
on                  removeAllListeners
removeListener      toString
valueOf
>

closed 2018-02-14 by MaBecker