espruino/Espruino #1336: commit adding #if instead of if in jswrap_wifi.c remove all wifi commands from ESP32

ticpu, 2018-02-14:
Since commit 7c2ba180d113710f7a324b95efe07f30128ea69b, there are no ESP32 specific wifi commands available.

Removing the pound sign allows for the command to appear and work.

```
>process.env
={
  "VERSION": "1v95.213",
  "BUILD_DATE": "Feb 13 2018 22:47:55",
  "GIT_COMMIT": "626ba6677427fe0eebceba3ec766344cb49749bc",
  "BOARD": "ESP32",
  "FLASH": 0, "RAM": 524288,
  "SERIAL": "30aea43b-7998",
  "CONSOLE": "Serial1",
  "MODULES": "Flash,Storage,fs," ... "r,crypto,neopixel",
  "EXPORTS": { "jsvLock": 1074616988, "jsvLockAgainSafe": 1075180156, "jsvUnLock": 1074618000, "jsvSkipName": 1074628996,
    "jsvMathsOp": 1074635852, "jsvMathsOpSkipNames": 1074637856, "jsvNewWithFlags": 1074624980, "jsvNewFromFloat": 1074625676, "jsvNewFromInteger": 1074625588,
    "jsvNewFromString": 1074625300, "jsvNewFromBool": 1074625648, "jsvGetFloat": 1074631992, "jsvGetInteger": 1074628168, "jsvGetBool": 1074632372,
    "jspeiFindInScopes": 1074648484, "jspReplaceWith": 1074648236, "jspeFunctionCall": 1074657828, "jspGetNamedVariable": 1074649976, "jspGetNamedField": 1074650364,
    "jspGetVarNamedField": 1074650476 },
  "EXPTR": 1073417248 }
>wifi = require("Wifi")
=function () { [native code] }
>wifi.
connect             disconnect
getAPIP             getIP
scan                startAP
stopAP              apply
bind                call
replaceWith         clone
emit                hasOwnProperty
length              on
removeAllListeners  removeListener
toString            valueOf
```

gfwilliams, 2018-02-14:
I just checked and before there was "connect\0disconnect\0getAPDetails\0getAPIP\0getDetails\0getHostByName\0getHostname\0getIP\0getStatus\0ping\0restore\0save\0scan\0setAPIP\0setConfig\0setHostname\0setIP\0startAP\0stopAP\0turbo\0" and there's now `"connect\0disconnect\0getAPIP\0getIP\0scan\0startAP\0stopAP\0"` according to the wrapper files.

Is that what you're reporting? I believe all of those functions that were removed just printed `not implemented` - I figured it was better if the docs accurately reflected what is in there and what isn't.

gfwilliams, 2018-02-14:
Ahh, ok - so stuff like `save` is missing. I'll look into this.

gfwilliams, 2018-02-14:
We now have:

```
connect
disconnect
getAPDetails
getAPIP
getDetails
getIP
getStatus
restore
save
scan
startAP
stopAP
```

closed 2018-02-14 by gfwilliams

MaBecker, 2018-02-14:
Missing setHostname and some other from esp8266 Wifi module

gfwilliams, 2018-02-14:
There's a new commit that fixes that now.

wilberforce, 2018-02-14:
@gfwilliams 

Working for ESP32 now

commit e0ba3ccb (2018-02-14): fix ESP32-specific commands - fix #1336