espruino/Espruino #1337: switch to new flash library causes save() to freeze on ESP32

ticpu, 2018-02-14:
From commit 2a4ebb251adfda823e89b00fb728893b00e763bb, using save(); on ESP32 causes the device to stop executing code and never returns.

From 2a4ebb251adfda823e89b00fb728893b00e763bb
```
>save();
=undefined
```

From master:
```
>save();
=undefined
Compacting Flash...
 ```

gfwilliams, 2018-02-14:
Maybe try `require("Storage").eraseAll()`?

If compacting fails it could be related to an issue with writes from unaligned memory.  I don't have an ESP32 here to check, but you could copy the changes from ESP8266's `jshFlashWrite` to ESP32's?

gfwilliams, 2018-02-14:
I just made some changes which may help with this - however if that's the case, the root cause would still be `jshFlashWrite` and it should probably be fixed as I'm pretty sure it's possible to feed it with unaligned data.

wilberforce, 2018-02-14:
@gfwilliams 

I've documented some of the output here, @MaBecker  asked me to test...

https://github.com/espruino/Espruino/issues/1335

It seems the indexes for  Storage are working, but the content is not saving.

gfwilliams, 2018-02-14:
Please could you try again with a new build and the suggestion there of using `E.memoryArea` to see if the file really is there?

It seems from what you posted that the writing is all working, it's the `read` that seems to crash it. Specifically, if it's printed `="` that means the whole file has been read, linked into a string, and it's crashing when trying to access the string.

gfwilliams, 2018-02-14:
Does peek32 at the address work?

I notice that Flash.read uses `spi_flash_read` - it doesn't use memory mapping at all... So I wonder if `memoryArea`/similar has ever worked?

gfwilliams, 2018-02-14:
Is there a way to just memory map the whole of SPI memory? I think the problem is that every other device uses a flat memory model, and for ESP32 we've got this thing where `jshFlashWrite`/etc use one set of addresses (actual SPI memory addresses), but then when you use memoryArea, the addresses are totally different.

There's RAM, which is somewhere random, but then low addresses are assumed to be offsets in the js_code partition, which is definitely at a nonzero location in flash.

gfwilliams, 2018-02-14:
According to https://www.espressif.com/sites/default/files/documentation/esp32_technical_reference_manual_en.pdf it looks like we might just be able to add `0x3F400000` to the memory address to offset into Flash memory? You could have a go with `peek32` and see what happens?

In which case we could drop the whole `romdata_jscode` thing - or move it into `jshFlashGetFree`.

wilberforce, 2018-02-14:
Agreed. It is the memory mapping that will be the issue. I was trying to look in the earlier code/ branches to see how the js_code was used. 
Sleep time now!

gfwilliams, 2018-02-14:
Ok, it's broken anyway so I'll commit some changes, and maybe someone with an ESP32 can check.

In the mean time, I should probably order one :)

gfwilliams, 2018-02-14:
It's done, with 71d19a91cca1b1051b9726acd44dabde295ecf8e - so lets see if that works or not.

If it does, we'll want to tidy up the partitions thing I guess - especially as `jshFlashGetFree` reports the area of memory that is used for saved code as 'free' - so stuff like FlashEEPROM would presumably delete any saved code :)

ticpu, 2018-02-14:
Oh!! However, using erase all first works!
```
>require("Storage").eraseAll()
=undefined
>
 ____                 _
|  __|___ ___ ___ _ _|_|___ ___
|  __|_ -| . |  _| | | |   | . |
|____|___|  _|_| |___|_|_|_|___|
         |_| espruino.com
 1v95.221 (c) 2017 G.Williams
Espruino is Open Source. Our work is supported
only by sales of official boards and donations:
http://espruino.com/Donate
>
=undefined
>process.env
={
  "VERSION": "1v95.221",
  "BUILD_DATE": "Feb 14 2018 09:42:20",
  "GIT_COMMIT": "71d19a91cca1b1051b9726acd44dabde295ecf8e",
  "BOARD": "ESP32",
  "FLASH": 0, "RAM": 524288,
  "SERIAL": "30aea43b-7998",
  "CONSOLE": "Serial1",
  "MODULES": "Flash,Storage,fs," ... "r,crypto,neopixel",
  "EXPORTS": { "jsvLock": 1074618444, "jsvLockAgainSafe": 1074618588, "jsvUnLock": 1074620204, "jsvSkipName": 1074636188,
    "jsvMathsOp": 1074645604, "jsvMathsOpSkipNames": 1074647604, "jsvNewWithFlags": 1074630816, "jsvNewFromFloat": 1074631724, "jsvNewFromInteger": 1074631636,
    "jsvNewFromString": 1074631248, "jsvNewFromBool": 1074631696, "jsvGetFloat": 1074640524, "jsvGetInteger": 1074634820, "jsvGetBool": 1074640928,
    "jspeiFindInScopes": 1074660552, "jspReplaceWith": 1074660288, "jspeFunctionCall": 1074670708, "jspGetNamedVariable": 1074662044, "jspGetNamedField": 1074662432,
    "jspGetVarNamedField": 1074662544 },
  "EXPTR": 1073417248 }
>save();
=undefined
Compacting Flash...
Calculating Size...
Writing...........
Compressed 80000 bytes to 9472
Running onInit()...
WARNING: jshI2CSetup: driver installed, sda: 5 sdl: 4 freq: 50000,
> 
```

gfwilliams, 2018-02-14:
Great! But with the old data in there it would just crash?

ticpu, 2018-02-14:
I try again and it didn't freeze, however, I had a crash but this happened before too. So it seems to work even without eraseAll() now, I don't know what eraseAll erases, but would it be possible there was some lingering stuff it got rid off?

Here is for the crash, this may be worth opening another issue.
```
>save();
=undefined
Compacting Flash...
Calculating Size...
Writing...........
Compressed 80000 bytes to 9349
Running onInit()...
WARNING: jshI2CSetup: driver installed, sda: 5 sdl: 4 freq: 50000,
Guru Meditation Error of type StoreProhibited occurred on core  0. Exception was unhandled.
Register dump:
PC      : 0x4000c2e4  PS      : 0x00060433  A0      : 0x80084e22  A1      : 0x3ffdd840
A2      : 0x00000000  A3      : 0x3ffafaa8  A4      : 0x3ffafae8  A5      : 0x00000000
A6      : 0xffffffff  A7      : 0x3ffafaa8  A8      : 0x00000000  A9      : 0x3ffb0590
A10     : 0x3ffb0b0c  A11     : 0x00000001  A12     : 0x0000000e  A13     : 0x000002c2
A14     : 0x3ffb6928  A15     : 0x3ffb7e48  SAR     : 0x00000008  EXCCAUSE: 0x0000001d
EXCVADDR: 0x00000000  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0x03ffafad
Backtrace: 0x4000c2e4:0x3ffdd840 0x40084e1f:0x3ffdd850 0x400852e8:0x3ffdd870 0x40087269:0x3ffdd8b0 0x400874f6:0x3ffdd8e0 0x40127c7f:0x3ffdd900 0x400f30a7:0x3ffdd950 0x4010385a:0x3ffdd970
================= CORE DUMP START =================
Lots of base64...
================= CORE DUMP END =================
Rebooting...
ets Jun  8 2016 00:22:57
rst:0xc (SW_CPU_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
```

wilberforce, 2018-02-14:
please see:

https://github.com/espruino/Espruino/pull/1341

gfwilliams, 2018-02-19:
Now that's merged, can we close this?

ticpu, 2018-02-19:
I'm fine with closing this!

closed 2018-02-19 by ticpu