espruino/Espruino #1340: Storage module missing  functions

MaBecker, 2018-02-14:
now on >process.env.GIT_COMMIT
="eba16e43a675ee2dcdbc95cfd5e506271fa03c11"

missing some functions

st = require('Storage');
=function () { [native code] }

>st. TAB
apply bind
call replaceWith
clone emit
hasOwnProperty length
on removeAllListeners
removeListener toString
valueOf

wilberforce, 2018-02-14:
works for me on ESP32 - might be only ESP8266?

```
>var st=require("Storage");
=function () { [native code] }
>st.
compact             debug
erase               eraseAll
list                read
write               apply
bind                call
replaceWith         clone
emit                hasOwnProperty
length              on
removeAllListeners  removeListener
toString            valueOf
>process.env.GIT_COMMIT
="eba16e43a675ee2dcdbc95cfd5e506271fa03c11"
```

MaBecker, 2018-02-14:
now mine works too :)
```
>st.
compact             debug
erase               eraseAll
list                read
write               apply
bind                call
replaceWith         clone
emit                hasOwnProperty
length              on
removeAllListeners  removeListener
toString            valueOf
>process.env.GIT_COMMIT
="eba16e43a675ee2dcdbc95cfd5e506271fa03c11"
> 
```

closed 2018-02-14 by MaBecker

wilberforce, 2018-02-15:
@MaBecker 

Please cab you check:

```
>process.env.MODULES.split(',');
=[
  "Flash",
  "Storage",
  "fs",
  "net",
  "dgram",
  "tls",
  "http",
  "NetworkJS",
  "Wifi",
  "TelnetServer",
  "crypto",
  "neopixel"
 ]
```

Was not expecting `NetworkJS`

MaBecker, 2018-02-15:
can confirm, this was merged long time ago check [ChangeLog](https://github.com/espruino/Espruino/blob/master/ChangeLog#L583).

gfwilliams, 2018-02-15:
Yes, NetworkJS is just part of the network lib as it takes very little space - it enables using custom network hardware with JavaScript.

Potentially it could be taken out, but actually with very little effort things could be tweaked to allow switchover from ESP32 networking such that you could talk to devices like GSM/LTE modems from ESP32.