espruino/Espruino #1342: brand new module Storage

MaBecker, 2018-02-15:
Wow: This is a birlliant solution to use save pages as file storage!

have you thought about

 * to return addr and size to use with E.memoryArea() , or  directly  return .memoryArea('file')

or

 * to make a 'file' visible as global var  like this[ hack](http://forum.espruino.com/comments/12965854/) does for a function

wilberforce, 2018-02-15:
My two cents worth.

a. I believe the term `file` might be a bit confusing with the FS lib. They are sort of analogous to the snippets in the web ide. If not sure if this is a better term - but it might be less confusioning.

b. How about accessing via [] or perhaps a prop method?

given:
```
st=require("Storage");
>st.write("data", '2000');
[Flash] CreateFile
[Flash] CreateFile new 0x00100024
=true
>st.read("data");

```

`console.log( st["data"] );`
or
`console.log( st.prop.data] );`

The low level `read` and `write` would still be available.

c. We could then allow setters:

`st['new']='new stuff';`

or
`st.prop.new='new stuff';`

c. How about more complex types?

```
var config = { wifi: { sid: 'mysid', password: 'mypass' }, relay_pin: 10 };
={
  "wifi": {
    "sid": "mysid",
    "password": "mypass"
   },
  "relay_pin": 10 }

st.write('config',JSON.stringify(config)); 
```

On startup, we can then do:

```
var start_config=JSON.parse(st.read('config'));
={
  "wifi": {
    "sid": "mysid",
    "password": "mypass"
   },
  "relay_pin": 10 }

```

d. What would be really cool is to allow objects and do the `JSON.stringify` and `JSON.parse` transparently, by looking at the parameter.

We could then just have:

`st['config']=config;`

and

`var startup_config=st['config'];`

e. extending the complex types further, would be to include a compress option that uses heatshrink, so that compressing and decompression is done silently as per the code compression. This might need another flag to store the nature of the content.

f. combinging d and e, if the string was over a certain length threshold, siliently:
`object->JSON -> compress`
`decompress -> Json  -> object`

What do you think?

gfwilliams, 2018-02-16:
> return addr and size to use with E.memoryArea() , or directly return .memoryArea('file')

That's what it does already when you read? https://github.com/espruino/Espruino/blob/master/src/jswrap_storage.c#L80

Or am I missing something?

> make a 'file' visible as global var

If you just do `eval(require("Storage").read("myfunc"))` then if that's a function it'll get loaded in, with the code still stored in flash. I'd have thought that was simple enough as-is?

> I believe the term file might be a bit confusing 

True - I deliberately changed `readFile` to `read`/etc and moved it into the `storage` lib. What are you proposing - just to change the documentation not to mention 'file' at all?

Thing is while they're not proper files, they're still more file-like than variables. The names could be changed to `get` and `set` which might help?

> accessing via [] or perhaps a prop method?

Sadly that's not something Espruino can do at the moment - it'd require quite a bit of hacking around in the interpreter.

>  How about more complex types?

Yes - that's how I saw it being used.

> allow objects and do the JSON.stringify and JSON.parse transparently

That would be really neat. I think automatically stringifying objects would be good.

I'm unsure about reconstructing them though - I mean, it means that you could have code that worked fine with some data in storage, but that failed with a type error if the data written was a string.

What about just having a `readJSON` utility function that saved some typing?

> heatshrink

That would be cool - however it does mean that the file could no longer be served up as a `memoryArea` and would have to be decompressed into RAM. That's not a big deal, but I wonder if we want to do it automatically?

I think that's something to look at adding later, but we could add a third `options` parameter to `write` that let you specify compression.

wilberforce, 2018-02-16:
> just to change the documentation not to mention 'file' at all?
Yes - call it something else... data is a bit odd? Not sure what?

Is there any room in the indexing framework to store simple flags?

Then on write your can store the type of the data at the time, if it is a simple type, json, compressed, binary, etc. Then on read the flags can be read and decoded appropriately based on the flags?

What I'm suggesting  is before this gets into the wild, prevent any future changes to the binary format before anyone starts using it!

MaBecker, 2018-02-16:
@gfwilliams

> Or am I missing something?
No i did :)

opichals, 2018-02-16:
> **forcing** people to **readJSON is** probably **more readable**

👍

MaBecker, 2018-02-16:
readJS() and writeJS() 

```
st.write('writeJS',"writeJS=function(n,o){require('Storage').write(n,JSON.stringify(o));};");
st.write('readJS',"readJS=function(e){return JSON.parse(require('Storage').read(e));};");
eval(st.read('readJS'));
eval(st.read('writeJS'));
global
={
  "readJS": function (e) { ... },
  "writeJS": function (n,o) { ... }
 }

var config = { wifi: { sid: 'mysid', password: 'mypass' }, relay_pin: 10 };
={
  "wifi": {
    "sid": "mysid",
    "password": "mypass"
   },
  "relay_pin": 10 }

writeJS('config',config);
=undefined

delete config;
=true

readJS('config').wifi.sid;
="mysid"

```

commit ae7e9814 (2018-02-19): Add ability to store flags in files, add readJSON/readArrayBuffer convenience fns, auto-stringify JSON - fix #1342
closed 2018-02-19 by gfwilliams