espruino/Espruino #1344: broken scripts/build_docs.py

MaBecker, 2018-02-16:
@gfwilliams Should this work or are you still working on things that breaks this?

>scripts/build_docs.py
Script location /Volumes/Espruino/repos/Espruino/scripts
Scanning for jswrap.c files
Scanning ./libs/bluetooth/jswrap_bluetooth.c
......
Scanning ./targets/esp32/jswrap_rtos.c
FOUND 'DO_NOT_INCLUDE_IN_DOCS' IN FILE ./targets/esp32/jswrap_rtos.c
Scanning ./targets/esp8266/jswrap_esp8266.c
Scanning ./targets/esp8266/jswrap_nodemcu.c
Scanning ./targets/nucleo/jswrap_nucleo.c
Scanning finished.
Traceback (most recent call last):
  File "scripts/build_docs.py", line 339, in <module>
    html_description(desc, jsondata["name"])
  File "scripts/build_docs.py", line 86, in html_description
    d = htmlify(d)
  File "scripts/build_docs.py", line 72, in htmlify
    d = markdown.markdown(d, extensions=['urlize'])
  File "/Library/Python/2.7/site-packages/markdown/__init__.py", line 493, in markdown
    md = Markdown(*args, **kwargs)
  File "/Library/Python/2.7/site-packages/markdown/__init__.py", line 159, in __init__
    configs=kwargs.get('extension_configs', {}))
  File "/Library/Python/2.7/site-packages/markdown/__init__.py", line 185, in registerExtensions
    ext = self.build_extension(ext, configs.get(ext, {}))
  File "/Library/Python/2.7/site-packages/markdown/__init__.py", line 264, in build_extension
    module = importlib.import_module(module_name_old_style)
  File "/System/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/importlib/__init__.py", line 37, in import_module
    __import__(name)
ImportError: Failed loading extension 'urlize' from 'urlize', 'markdown.extensions.urlize' or 'mdx_urlize'

gfwilliams, 2018-02-17:
No, it should work.

Most likely it's just some issue with you installing the correct thing from pip - `urlize`?

MaBecker, 2018-02-17:
tried to find the missing piece without succes, as this works on youre side - close it

closed 2018-02-17 by MaBecker

MaBecker, 2019-04-10:
this what fixed this issue for Mac OS X 

```
def htmlify(d,current):
-  d = markdown.markdown(d, extensions=['urlize'])
+  d = markdown.markdown(d, extensions=['markdown_include.include'])
  # replace <code> with newlines with pre
  idx = d.find("<code>")
```

gfwilliams, 2019-04-11:
can't you just `pip install urlize` or something like that? Seems like removing `urlize` probably changes what comes out the end of build_docs

MaBecker, 2019-04-11:
Thanks , please do not take care about this, it is just my local work around to be able to check my local  changes before creating PRs .

This is what I did, but it did not fix it.

# Needs:
#    pip install markdown
#    pip install markdown-urlize