espruino/Espruino #1345: using !=  does not change false to true

MaBecker, 2018-02-17:
found this issue  as working with != 

```
>FLAG=false
=false

>FLAG != FLAG
=false                     <- expected true

>FLAG = !FLAG
=true                      <- as expected true 
```

gfwilliams, 2018-02-17:
But flag==flag is true, so the opposite should be false? Does this work if you do it in the console of a desktop browser? (I'm in a phone now)

gfwilliams, 2018-02-17:
Ahh. Do you assume it's like += ? It's actually the opposite of ==

MaBecker, 2018-02-17:
Yes, like to use `flag != flag` to toggle from false to true and back (console)

with node it is the same
```
$ node
> FLAG=false
false
> FLAG != FLAG
false
> FLAG == FLAG
true
> FLAG == FLAG
true
```

Hmm, is this a bug ...

So close this and use `FLAG = !FLAG` to toggle :)

closed 2018-02-17 by MaBecker