espruino/Espruino #1346: ESP32: setWatch and digitalWrite Assert failure - DHT11 module

wilberforce, 2018-02-19:
Reported as  issues with the ESP32 and the DHT11 module:

http://forum.espruino.com/conversations/316936
http://forum.espruino.com/conversations/316733

To reproduce:

```
>setWatch(function(e) {console.log("Button pressed",e);}, D16, { repeat: true, edge: 'rising' });
=1
>digitalWrite(D16,0);
ASSERT(channel >= EV_EXTI0 && channel <= EV_EXTI_MAX) FAILED AT src/jsdevices.c:413
  #1[r2,l2] Object {
    #2[r1,l2] Name String [1 blocks] "\xFF"      #3[r1,l2] Object {
        #6[r1,l2] Name String [2 blocks] "timers"          #8[r2,l1] Array(0) [ ]
        #9[r1,l2] Name String [2 blocks] "watches"          #11[r2,l1] Array(2) [
            #72[r1,l2] Name Integer 1              #63[r1,l1] Object {
                #65[r1,l2] Name String [1 blocks] "pin"                  #64[r1,l1] Pin 16
                #67[r1,l2] Name String [2 blocks] "recur"                  #66[r1,l1] Bool true
                #70[r1,l2] Name String [1 blocks] "edge"= int 1
                #69[r1,l2] Name String [2 blocks] "callback"                  #46[r1,l1] Function {
                    #45[r1,l2] Name Param "\xFFe"                       undefined
                    #44[r1,l2] Name String [1 blocks] "\xFFcod"                      #57[r1,l1] FlatString [3 blocks] "console.log(\"Button pressed\",e);"
                  }
              }
          ]
        #13[r1,l2] Name String [1 blocks] "net"          #14[r1,l2] String [1 blocks] "\x05\x00\x00\x00\x00\xFF\xFF\xFF"
        #25[r1,l2] Name String [2 blocks] "modules"          #27[r1,l1] Object {
            #29[r1,l2] Name String [1 blocks] "Wifi"              #28[r1,l1] NativeFunction 0x400ea898 (0) { }
          }
        #47[r1,l2] Name String [2 blocks] "history"          #42[r1,l1] Array(1) [
            #61[r1,l2] Name Integer 0              #48[r1,l1] String [9 blocks] "setWatch(function(e) {console.log(\"Button pressed\",e);}, D16, { repeat: true, edge: 'rising' });"
          ]
      }
  }
HALTING.
```

`digitalWrite(D16,0)` on it own works ok.

To get the assert (rather than a core dump) - compile:
`make clean && BOARD=ESP32 DEBUG=1 make`

https://github.com/espruino/Espruino/blo­b/master/src/jsdevices.c#L706
```
void jshSetEventCallback(
    IOEventFlags channel,             //!< The event that fires the callback.
    JshEventCallbackCallback callback //!< The callback to be invoked.
  ) {
  // Save the callback function for this event channel.
  assert(channel>=EV_EXTI0 && channel<=EV_EXTI_MAX);
  jshEventCallbacks[channel-EV_EXTI0] = callback;
}
```

EV_EXTI_MAX is defined as: https://github.com/espruino/Espruino/blo­b/e609b94f83f665c26cf4f86acac068c396427e­99/src/jsdevices.h#L54

```
EV_EXTI15, // External Interrupt 15
  EV_EXTI_MAX = EV_EXTI15,
```

@jumjum123 did the setwatch code for ESP32 - so I'm unsure how pin numbers (D22 in this case) is mapping to a channel?

@gfwilliams 

It seems the code is built for board with pins 0-15? We are getting a failure when a pin > 16 is used.

Sorry - without having an understanding of the  code I'm note sure how to fix or the relationship between `setWatch()` and `digitalWrite()`  and channels and pins?

gfwilliams, 2018-02-19:
On all the other platforms there is only a relatively small amount of hardware that can do watches, so EV_EXTIx corresponds to that and there's a mapping, eg: https://github.com/espruino/Espruino/blob/master/targets/nrf5x/jshardware.c#L867

I'm not sure what the case is with ESP32 but if it is then it's hidden behind their API.

Given the competition for space in that enum (otherwise we'd have to double the size of it, which wastes ~128 bytes of input buffer on most platforms), I'd rather not increase that size across the board. We could perhaps define EV_EXTI_COUNT in platform_config.h though and use that?

jumjum123, 2018-02-19:
> nrf5x/jshardware.c#L867

Pin 16 and 17 are used/reserved for PSRAM.
I would recommend, not to use them.

wilberforce, 2018-02-19:
Thanks @gfwilliams and @jumjum123 
Even thought there are gpio0-gpio39' some are reserved and can't be watched, like 6-11 are reserved for the flash chip. So I guess a mapping table of useable pins that can be watched, mapping to a channel would be the real solution.

I'm not sure how to cleanly do the changes to jsdevice.h without a conditional `#ifdef ESP32`.

Currently there is no tie up with the generated platform.h, so within the enum defines are made for non-existent serial ports for example, that would free up some of the numbers, (or would that be used for software serial?)

gfwilliams, 2018-02-19:
Yes, I was wondering about the unused Serial/I2C/etc too. The tie-up could have to be done with build_platform_config based on chip family, but it would be nice improvement I think.

Having said that, how many ESP32 pins *can't* be used? If there are only 16 usable then it totally makes sense to use the 16 EXTI that we have.

Just a note, but if jshPinWatch returns EV_NONE, I believe you get a nice `can't watch this pin` error message, so it'd be good to add for 6-11 in any case.

wilberforce, 2018-02-19:
Thanks. I'm trying to look at the ESP-idf docs to see if there is a function to check watchability of pins like in the nfc sample you pointed me too. I see that checks in a loop - wondering if the speed of that has any bearing?

And do you know a cunning way of checking that enum does not exceed 64 at compile time?

wilberforce, 2018-02-19:
According this there appears to be at least 19-20 pins:

https://github.com/espruino/Espruino/blob/028b59265e2a04e7bb2c5a7f7e68b16a00e4f481/targets/esp32/jshardware.c#L113-L120

gfwilliams, 2018-02-19:
Ok, one sec and I'll update `devices.h` - while adding 4 isn't a big deal it'd be good to have flexible numbers as on some other platforms I'm throwing some RAM away for them

wilberforce, 2018-02-19:
Thanks @gfwilliams 

This is now working on the ESP32:

```
setWatch(function(e) {console.log("Button pressed",e);}, D22, { repeat: true, edge: 'rising' });
```

closed 2018-02-19 by wilberforce