espruino/Espruino #1350: Cant save() latest build on ESP32

gurchk, 2018-02-21:
When trying to save() it gets stuck:
>save()
=undefined
Compacting Flash...
Calculating Size...

Tried flashing different firmware (v1.95) and then saving with the same code, no problem. 
Think it has something to do with the latest fix you did for the ESP32 Meditation error fix.

wilberforce, 2018-02-21:
Did you build yourself?
Oor get from here?  http://www.espruino.com/files/espruino_1v96.zip

The storage module http://www.espruino.com/Reference#Storage changes have changed the area used for code storage - so the issue might be old code being incompatible with the new code.

Please try:

```
require("Storage").eraseAll();
```

Then try re-saving your code.

If that still fails - How large is your code? is it possible to send it to test against? You can PM me if you prefer.

gurchk, 2018-02-22:
I did the build myself, might have messed up something. 
Downloaded the 1v96v you sent me, now it works. 
Thx.

closed 2018-02-22 by gurchk