espruino/Espruino #1358: Add missing option ssid_hidden to Wifi.startAP()

MaBecker, 2018-03-18:
Extend list of possible option with "hidden" to start a AP  visible or hidden.

AP is visible 

    "hidden" : 0          default value

AP is hidden

    "hidden" : 1         

Gordon where do I find the API reference for Wifi.startAP(), because  I like to add this to the reference too.

wilberforce, 2018-03-18:
@MaBecker 

https://github.com/espruino/Espruino/blob/0ad87619ddbbeefa91fc6e9ff712c34ed2e98c2a/libs/network/jswrap_wifi.c#L274-L276

MaBecker, 2018-03-18:
Thanks

what about ESP32 and hidden option?

MaBecker, 2018-03-18:
How do you create this nice looking source  code links?

wilberforce, 2018-03-18:
I did a search to find `startAP`
In the results page clicked on the line number 261 here libs/network/jswrap_wifi.c

Then clicked on line 272 and shift-clicked line 278. This updates the URL to have this line range.
Then cut and paste into the comment. I have added a single back tick to make this appear:

`https://github.com/espruino/Espruino/blob/0ad87619ddbbeefa91fc6e9ff712c34ed2e98c2a/libs/network/jswrap_wifi.c#L272-L278`

wilberforce, 2018-03-18:
> what about ESP32 and hidden option?

Here is an example
https://github.com/espruino/Espruino/blob/0ad87619ddbbeefa91fc6e9ff712c34ed2e98c2a/libs/network/jswrap_wifi.c#L291

Not sure you can use a condtition for part of the options list - you just say ESP32 specfic

gfwilliams, 2018-03-19:
Yes, there's no way to have one argument only for one platform. You'd just have to document it manually in the `/*JSON` comment

wilberforce, 2018-03-19:
Oh I think your asking if esp32 supports hidden  AP - yes!

Set as part of the struct here:

https://github.com/espruino/Espruino/blob/9ffc5162047e93200e1356c3334b05f5ab67db53/libs/network/esp32/jswrap_esp32_network.c#L851

MaBecker, 2018-03-19:
Cool Wilberforce

So adding option hidden has to take place in ESP8266,ESP32 and ESP8266WiFi.js - correct?

Just checked AT Instructions and figured out how to handle this

    AT+CWSAP=<ssid>,<pwd>,<chl>,<ecn>[,<max	conn>][,<ssid	hidden>]

    • [<ssid	hidden>]	(optional):
    ‣ 0: SSID is broadcasted. (the default setting)
    ‣ 1: SSID is not broadcasted.

gfwilliams, 2018-03-19:
The main JS one to add it to is `EspruinoWiFi.js`, but `ESP8266WiFi.js` and `ESP8266WiFi_0v25.js` would be the ones for Espruino & ESP8266.

The thing to watch out for is whether the ESP8266 firmware actually supports the AT command. Adding the extra arguments if it didn't would make the call fail :(

MaBecker, 2018-03-19:
So what to order to be able to test (got a pico and some ESP01)?

closed 2018-05-09 by MaBecker