espruino/Espruino #1359: pipe doesn't work 1v96

bikevit2008, 2018-03-22:
var chunksize = 2048;

try{
        var f = E.openFile(pathname, "r");
        var options = { chunkSize : int=chunksize };
        res.writeHead(200, {'Content-Type': contentType, 'Content-Encoding': 'gzip'});
        f.pipe(res, options); // streams the file to the HTTP response

    } catch (err) {
        // else not found - send a 404 message
        res.writeHead(404, {'Content-Type': 'text/plain'});
        res.end("404: Page "+pathname+" not found");
    }

This code work in 1v95 without changes, but in 1v96 i have error.

"Error {
  "msg": "Field or method \"pipe\" does not already exist, and can't create it on undefined",
  "type": "Error",
  "stack": " at line 56 col 10\n        f.pipe(res, options); // streams the file to the HTT...\n         ^\n"
 }
ERROR: Headers have already been sent"

Am i not right? Or is this error really?

gfwilliams, 2018-03-22:
Looks to be like `f` is undefined. That means `E.openFile(pathname, "r");` didn't open the file, so chances are you just requested a file that's not on the SD card?

bikevit2008, 2018-03-23:
I use internal fat storage (1Mb) for files, file exists exactly. File size ~133Kb. My browser starts download and can download only 8.6Kb, but after this, browser get error and file server on Esp32 give error to console. This problem not exists on 1v95.

gfwilliams, 2018-03-26:
Try adding a `console.log(pathname)` before `openFile` so you can see what it's trying to open. It might be requesting `favicon.ico` or something.

wilberforce, 2018-03-26:
Try chunk size 512 - 2048 it quite large, and it will get broken into smaller packets anyway.

And before f.pipe, check: 

```
If ( f === undefined ) 
    console.log( { missing: pathname } )
 else 
    f.pipe(res, options); // streams the file to the HTTP response
```

bikevit2008, 2018-03-26:
I tried, pathname to my file with react app is not undefined. When I used different chunk size, for example 1500, I got this trouble together, but browser can download not 8,6Kb, but ~119 bytes and I get error again.

gfwilliams, 2018-03-26:
You problem is that `E.openFile(pathname, "r");` returns `undefined`. Most likely because that file doesn't exist.

However we're going to struggle to help if you further if can't provide us with a complete example (code + files) that causes the problem.

wilberforce, 2018-03-26:
You are telling the browser that you are sending gzipped content
`Content-Encoding': 'gzip`

Are you sending compressed content?

bikevit2008, 2018-03-26:
I can give access to my code in the private repository. 
All files gzipped before upload in ESP32 is naturally.

gfwilliams, 2018-03-26:
Maybe you can invite @wilberforce if he's happy to help.

Giving you free support for your proprietary product that's built on free software isn't something I'm willing to do.

bikevit2008, 2018-03-26:
Actually, i don't need for support my software, i wanted notify for bug and maybe get info about bug. Why i think its really bug? It's simple, in the equals cases in 1v95 my code with files - works, in 1v96 my code with files - works incorrect. I checked choices in this topic before my first post here.

wilberforce, 2018-03-26:
I'm using f.pipe on 1.96 on the ESP32 and it is working fine. I have given suggestions on reducing the chunk size and asked about compression and you have not replied with responses to that. We have asked you to check to see if f === undefined and you respond with the file is there. How do you know?

closed 2018-04-09 by gfwilliams

barbiani, 2019-01-17:
Espruino 1.99

I am seeing the same here. Happening every 2 or 4 requests.