espruino/Espruino #1361: Load from flash makes the built-in module symbol tables not found

opichals, 2018-04-01:
In the current clean-built master whenever there is something stored in the flash (espruino.flash file on Linux) and gets loaded back upon startup the jwwrapper.c:jswGetSymbolListForObject() method cannot find the symbol table (guessing because they get loaded into another part of memory?).

See the following short flow for reproduction:

```
$ ./espruino 
Interactive mode.
Size of JsVar is now 32 bytes
Size of JsVarRef is now 4 bytes
Added SIGINT hook
Added SIGHUP hook
Added SIGTERM hook

 ____                 _ 
|  __|___ ___ ___ _ _|_|___ ___ 
|  __|_ -| . |  _| | | |   | . |
|____|___|  _|_| |___|_|_|_|___|
         |_| espruino.com
 1v96.3762 (c) 2018 G.Williams

Espruino is Open Source. Our work is supported
only by sales of official boards and donations:
http://espruino.com/Donate

>f = require('Flash');
=function () { [native code] }
>console.log( Object.getOwnPropertyNames(f) );
protoOwner Osymbols 59dca8d
[ 
  "erasePage", 
  "getFree", 
  "getPage", 
  "read", 
  "write"
 ]
=undefined
>save()
=undefined
Compacting Flash...
[Flash] Compacting
[Flash] Already fully compacted
Calculating Size...
[Flash] CreateFile (4248 bytes)
[Flash] CreateFile new 0x1003e010
[Flash] CreateFile write header
[Flash] CreateFile written header
Writing..
Compressed 131072 bytes to 4248
Press Ctrl-C again to exit

$ ./espruino 
Interactive mode.
Size of JsVar is now 32 bytes
Size of JsVarRef is now 4 bytes
Added SIGINT hook
Added SIGHUP hook
Added SIGTERM hook
Loading 4248 bytes from flash...

>dump()
=undefined
>f
=function () { [native code] }
>console.log( Object.getOwnPropertyNames(f) );
Uncaught Error: Function "log" not found!
 at line 1 col 9
console.log( Object.getOwnPropertyNames(f) );
        ^
```

gfwilliams, 2018-04-03:
And you're sure that's not after Espruino has been rebuilt?

What OS are you on? It works for me.

It's most likely that the physical address of the functions has changed. The `real_prototype_chain` branch wouldn't have this problem - so if I can get the few remaining issues with that hit on the head then that'd be the best solution for this.
 
... however it shouldn't be a problem with any embedded devices

opichals, 2018-04-03:
I am on OS X and I don't rebuild the binary in between.

Anyway I vote to go for the `real_prototype_chain` as that brings other benefits as well. How far is that from getting to master?

opichals, 2018-04-03:
The https://github.com/espruino/Espruino/issues/1288 tracks the address-based native function lookup challenges.

gfwilliams, 2018-04-04:
It shouldn't be too painful to move to master. The last commit or so on there was done by running a script and should be backed out, then merged, then the script run again and it should go relatively smoothly.

The reason it hasn't been merged yet is there are still some issues getting at the prototypes of built-in types (eg `[]` and `""`) and I need to get back into it a bit more to figure out why.

gfwilliams, 2018-05-22:
Fixed now - it'll refuse to load any saved code from something with a different commit hash

opichals, 2018-05-22:
@gfwilliams This issue was not about a different commit hash but rather about restarting the espruino.

gfwilliams, 2018-05-22:
Argh, thanks! :)