espruino/Espruino #1362: telnet.setConsole() calls UART setup & initializes but doesn't need to

jgtaylor, 2018-04-01:
### General Issue:

In file jsinteractive.c, line [176](https://github.com/espruino/Espruino/blob/3ad580107292c00a4ed680a7dc69b160aec4b18b/src/jsinteractive.c#L176), the if statement will always execute for telnet. This results in de-muxing and re-muxing any UART pins that are otherwise assigned (in my case, to I2C1), due to the jshUSARTSetup() call.

It seems that telnet doesn't have DEVICE_INITALIZED_FLAGS set, or is false. It may also be that the ESP32 code isn't setting it correctly, but I'm not well versed in C. 

Either way, I've worked around it by adding another if statement around the test for jshDeviceInitialised(device), with a simple:
``` if (device != EV_TELNET) { ... } ```
I expect this is a less than ideal solution, but it works (sort of) for now 😄 

### Random details

I posted on the [forum](http://forum.espruino.com/conversations/318858/) about this, and kept digging around until I found the work-around.

The work-around is not ideal. After connecting via telnet, you can not go back to USART without a reset() (or power off). In my case, this ok because I'm running the device via a wall-wart and not USB.

Also, this may be exclusive to ESP32, but I don't have anything else to verify the issue.

I also noticed that jshardware_common.c only has InitInfo for USART, SPI & I2C. Perhaps something about telnet should go in there? Perhaps not. I assume that if there was something there, the jshIsDeviceInitialised(device) call would return true. again, I'm probably way off.

jgtaylor, 2018-04-02:
So after thinking I'd found a work around because connecting via telnet didn't immediately result in:
```
ERROR: jshI2CWrite:, Operation timeout because the bus is busy.
ERROR: jshI2CRead:, Operation timeout because the bus is busy.
```
they showed back up after about 2h38m. Additionally, the console started outputing socket errors (which I wasn't able to capture, unfortunately - though I will try).

This is particularly confusing because power off/on, ESP32.reboot(), nor reset(1) followed by code upload, seems to clear the issue (even temporarily). This is happening with both I2C devices I have access to (BME280 & BH1750).

Now I'm wondering if there is some kind of USART polling that happens somewhere. Also suspect is network code that may be sneaking into the USART address space. I don't know how to even begin looking for the latter, but I'll try figuring something out with the former.

wilberforce, 2018-05-01:
@jumjum123 

This appears to be related to this line:

https://github.com/espruino/Espruino/blob/75f84021cf6cbeb79032b34ab047676d6f53fc62/targets/esp32/jshardwareUart.c#L31

Info  form the forum post- update: fails on D17/D16, works on scl: D18, sda: D19.

When the console is on telnet - I think that serial 3 might be assumed here....

https://github.com/espruino/Espruino/blob/75f84021cf6cbeb79032b34ab047676d6f53fc62/targets/esp32/jshardwareUart.c#L66

wilberforce, 2018-05-14:
@jumjum123 
While you are looking at the console switching code - would you be able to review this? When switching to telnet it looks like the uart is getting re-intialised?

gfwilliams, 2018-05-14:
Just tweaked this. Hopefully that'll sort it for you?

closed 2018-05-14 by gfwilliams

jgtaylor, 2018-05-15:
all sorted and works like a charm!

Thanks!

commit dfac6037 (2018-05-14): ESP32: Fix accidental initialisation of UART3 when switching to Telnet (fix #1362)