espruino/Espruino #1367: Not creating new variable scope in catch block

andrewwakeling, 2018-04-02:
The identifier (which holds the thrown value) [should only be available in the catch block.](https://developer.mozilla.org/en-US/docs/Web/JavaScript/Reference/Statements/try...catch)

```javascript
var e = 'hi';
try {
  throw new TypeError();
} catch(e) {}
console.log(e);
```

Expected: 'hi'
Actual: TypeError

andrewwakeling, 2018-04-02:
Apologies. I know this is really obscure. It caught me unawares when my code was minified.

gfwilliams, 2018-04-03:
Hi - thanks for reporting these. Out of interest, how did you find them? Did you run tests, or was it literally just some code you had that exhibited all these problems? It seems like you're particularly unlucky if so.

andrewwakeling, 2018-04-03:
I stumbled across this when minifying code from my workaround I posted in #1363.

`valueOrThenable` was renamed to `e` which caused my workaround to do strange things when my code was minified.

andrewwakeling, 2018-04-03:
@gfwilliams, this is what I've been working on https://github.com/andrewwakeling/espruino-webpack-babel-sample

commit 317cd029 (2018-04-09): try..catch now creates exception in its own scope (fix #1367)
closed 2018-04-09 by gfwilliams