espruino/Espruino #1372: Add ON Off - reasonable  or a waste of vars.

MaBecker, 2018-04-10:
Added ON OFF to my fork and asking myself if this would make sense to add this to the master.

src/jswrap_number.c:

```
/*JSON{
  "type" : "variable",
  "name" : "ON",
  "generate_full" : "1",
  "return" : ["int32","Logic 1 - this is the same as just typing `1`"]
}*/

/*JSON{
  "type" : "variable",
  "name" : "OFF",
  "generate_full" : "0",
  "return" : ["int32","Logic 0 - this is the same as just typing `0`"]
}*/
```

gfwilliams, 2018-04-10:
Honestly, probably not... We already have `HIGH` and `LOW` for Arduino compatibility - are ON and OFF used there as well, or it just seems nice to have?

MaBecker, 2018-04-10:
Nichtig have 😊

MaBecker, 2018-04-10:
Nice to have 😊

closed 2018-04-10 by MaBecker