espruino/Espruino #1373: webpack boilerplate can cause crash

andrewwakeling, 2018-04-12:
webpack seems to be generating some boilerplate code (when bundling modules from multiple files), which is causing Espruino to crash.

The below is my best attempt to reduce the code down to the most minimal necessary parts to reproduce the problem.

```javascript
(function() {
  var modules = [];
  function __webpack_require__(moduleId) {
    modules[moduleId].call(null, __webpack_require__);
  }
  modules[0] = function () {
    d();
  };
  modules[1] = function () {
    (function () {
      __webpack_require__(0);
    }).call(null);
  };
  modules[2] = function () {
    __webpack_require__(1);
  };
  function d() {
    console.log(modules.length);
  }
  __webpack_require__(2);
}());
```

I acknowledge that the way this code is invoked is fairly obscure. It would be great if this could be solved and it's probably beneficial to understand the root cause.

I am capable of reliably reproducing the crash with the above code on my d1 mini pro (ESP8266). This is the output that I see:

```
>3
1
=undefined
>
 ets Jan  8 2013,rst cause:2, boot mode:(3,6)
load 0x40100000, len 2408, room 16
tail 8
chksum 0xe5
load 0x3ffe8000, len 776, room 0
tail 8
chksum 0x84
load 0x3ffe8310, len 632, room 0
tail 8
chksum 0xd8
csum 0xd8
2nd boot version : 1.6
  SPI Speed      : 80MHz
  SPI Mode       : QIO
  SPI Flash Size & Map: 32Mbit(1024KB+1024KB)
jump to run user1 @ 1000
```

Any insight into what's happening would be great. Thanks.

gfwilliams, 2018-04-16:
That's an odd one - it works great on official Espruino boards by the look of it, so this seems to be ESP8266 specific.

wilberforce, 2018-05-15:
The code above works fine on the ESP32 and linux builds.

I get the same crash on a 4MB esp8266

```
reset causes:
    0: 
    1: normal boot
    2: reset pin
    3: software reset
    4: watchdog reset

boot device:
    0:
    1: ram
    3: flash
```

So seems to be a reset pin: `ets Jan  8 2013,rst cause:2, boot mode:(3,6)`

MaBecker, 2018-10-13:
this code is causing no problem:

```
  var modules = [];
  function __webpack_require__(moduleId) {
    modules[moduleId].call(null, __webpack_require__);
  }
  modules[0] = function () {
    d();
  };
  modules[1] = function () {
    (function () {
      __webpack_require__(0);
    }).call(null);
  };
  modules[2] = function () {
    __webpack_require__(1);
  };
  function d() {
    console.log(modules.length);
  }
  __webpack_require__(2);

```

@andrewwakeling what do you think?

MaBecker, 2018-10-16:
Tested again on GIT_COMMIT: "b70c2278" and compiled with `unset RELEASE`  to get some assert errors, but still reboots,  now with  cause:4 instead of cause:2

opichals, 2018-10-16:
Not sure the cause would be different but the code below (as above but extended with few more modules) fails on MacOS like as can be seen below. As soon as I change the `if (0)` to `if (1)` it doesn't crash.

The problem looks like too much recursion going on. @MaBecker Could you set the [jsutils.h:JSV_LOCK_MAX](https://github.com/espruino/Espruino/blob/master/src/jsutils.h#L307) to `31`, rebuild and re-test on the ESP8266 board?

```
5
ASSERT(jsvGetLocks(var) < 15) FAILED AT src/jsvar.c:656
  #1[r6,l2] Object { 
    #2[r1,l2] ASSERT(jsvGetLocks(var) < 15) FAILED AT src/jsvar.c:656
EXITING.
```

```
var modules = [];
function __webpack_require__(moduleId) {
    modules[moduleId].call(null, __webpack_require__);
}

modules[0] = function () {
    d();
};

modules[1] = function () {
    (function () {
        __webpack_require__(0);
    }).call(null);
};

modules[2] = function () {
    __webpack_require__(3);
};

if (0) {
  modules[3] = function () {
     __webpack_require__(1);
  };
} else {
  modules[3] = function () {
      __webpack_require__(4);
  };
  modules[4] = function () {
      __webpack_require__(1);
  };
}

function d() {
    console.log(modules.length);
}
__webpack_require__(2);
```

MaBecker, 2018-10-17:
Did, still reboots.

As Gordon mentioned above:

> That's an odd one - it works great on official Espruino boards by the look of it, so this seems to be ESP8266 specific.

MaBecker, 2018-10-23:
Just tested with 2v00

no more reboots on my side.

gfwilliams, 2018-10-24:
Let's close this. Original reporter couldn't even be bothered to reply to questions and it seems fixed anyway.

closed 2018-10-24 by gfwilliams